SvenVD/rpisurv #109: rotate screen - not cyling through streams

ozett, 2021-01-02:
hi,
i tested rpisurv and found it great. 

but as i have my screen "rotated" (not meaning the cycling, that you call "rotation" in the readme.md)
the rpisurv dont show a screen anymore. screen-resolution does not change with rotation.

so what to configure for a raspi-rotation in config.txt?

(latest raspian used, also rpiusrv3)

thx. 

```
sudo nano /boot/config.txt
Add one of the following lines to the bottom of the file :

display_rotate=0
display_rotate=1
display_rotate=2
display_rotate=3
0 is the normal configuration. 1 is 90 degrees. 2 is 180 degress. 3 is 270 degrees.
```

SvenVD, 2021-01-02:
Currently support questions should be asked on https://community.rpisurv.net/. Please redirect your support question there in the future. 

For vertical screen 1x3 example om 1080x1920 on Rpisurv 3 with a Rpi4, nothing should be changed in /boot/config.txt:

`minimal config (1x3 vertical):

    camera_streams:
        url: "put url here"
        force_coordinates: [0, 0, 1080, 640]
        freeform_advanced_vlc_options: " --mmal-vout-transform=270 --codec avcodec,none"
        url: "put url here"
        force_coordinates: [0, 640, 1080, 1280 ]
        freeform_advanced_vlc_options: " --mmal-vout-transform=270 --codec avcodec,none"
        url: "put url here"
        force_coordinates: [0, 1280, 1080, 1920]
        freeform_advanced_vlc_options: " --mmal-vout-transform=270 --codec avcodec,none"
        duration: 20000
`

ozett, 2021-01-02:
looks cool,
i will give it a try tonight.

sorry that i did not found it in the readme.md or other docu.. 😢 

big thanks

closed 2021-01-16 by SvenVD