SvenVD/rpisurv #110: Cannot overlay images with pngview on v3

glenne, 2021-01-04:
With V2 rpisurv I was able to overlay informational png images showing logo, tides and weather over the rpisurv generated content.  I am using pngview to target layer 2.    This worked nicely.

In V3 this no longer works.  I see in V3 that vlc is being used instead of omxplayer?  Is VLC somehow taking over all the layers which prevents this from working?  I haven't dug into this yet but thought it might be obvious to those who know what's changed in V3.  I can make my overlay system available on github if that would be useful but simply invoking 'pngview -l 2 -n image.png' with a transparent png should show the issue.

SvenVD, 2021-01-04:
Hi, it should still work, but you need to specify a higher layer. Rpisurv 3 is using the layers for caching. Try using layer 2100000000, this should display your transparant image on top of everything Rpisurv 3 does.

( As a side note, Rpisurv 3 also uses pngview for some things, so the binary is already included in /usr/local/bin/rpisurv/bin/pngview )

glenne, 2021-01-04:
Using layer 2100000000 worked!  Thanks.  Is there a good reference on layer numbers?  I've been using layer 1,2,3 with pngview but never considered something like 2100000000.

I ended up with a pull request for pngview to monitor the png file for changes and support a signal to refresh - https://github.com/glenne/raspidmx/blob/master/pngview/pngview.c.  Without use of the refresh modes, the overlay would briefly disappear and reappear when the png file needed to be updated.  I found this objectionable.

Here's the overlay with your layer suggestion:
[image]

SvenVD, 2021-01-04:
It is somewhat documented in the [code](https://github.com/SvenVD/rpisurv/blob/55630d1fbba168f147962321e87289a38c3caf61/surveillance/core/Screen.py#L19) itself.

closed 2021-01-04 by SvenVD