SvenVD/rpisurv #112: Blinking stream on 2x2

gabiX87, 2021-01-21:
My raspberryPI 4 running rpisurv v3_ľatest is experiencing ,on a 2x2 layout, a strange blinking issue. See here [https://youtu.be/8axsQqVENfc](https://youtu.be/8axsQqVENfc)

- camera_streams:
        - url: "rtsp://192.168.50.96:7447/5b23fae20cf2a2f4c6031f3e_2"
        - url: "rtsp://192.168.50.96:7447/5789f6650cf2169dc47bf524_2"
        - url: "rtsp://192.168.50.96:7447/5b6b4ab30cf2967703a3c1b6_2"
        - url: "rtsp://192.168.50.96:7447/5817b8980cf24dec5d4e5538_2"
   #  duration: 60
      nr_of_columns: 2

pulling the ( _2 ) low resolution RTSP streams from Unifi Video NVR + another 4 high quality streams on rotation.
I tried changing the caching but it did not help.

Also if I enabled audio on all 4 high res streams the audio is overlaping each other. Wouldn't it be possible to mute the audio until the stream is displayed on the screen ?

SvenVD, 2021-01-21:
This is most probably because you are requesting more from the raspberry gpu then it can handle. You should use low res streams as the gpu of the pi can not handle downscaling to the window size with that much high res streams.

For the audio we can theoretically disable  it for all streams when that screen is inactive. However that requires caching to be disabled too. Please create feature request on https://community.rpisurv.net for this. My time is limited so no guarantees this will ever be implemented. But when it is a feature request then at least it is on the radar if time becomes available.

gabiX87, 2021-01-21:
streams on the 2x2 are 640x360 @ 10FPS 

streams ending with _2 - low , _1 - medium , _0 - high

I will try to incrementaly disable _0 streams which are solo 1x1 on rotation after the 2x2

SvenVD, 2021-01-21:
What raspberry pi device do you have,

My guess is that you high res streams, are too high res, if you remove them, do you still experience the issue?

gabiX87, 2021-01-22:
Raspberry PI 4 model B - 2GB RAM
High res streams are 1920x1080 i.e FHD @ 25FPS

PI can handle 4K@60FPS  

4 times FHD = one 4K   PI should handle twice as many 1080 streams

SvenVD, 2021-01-22:
it also depends if the pi needs to do downscaling or not. What were the result of your tests? Does it still flickers with lower res streams? ( so no downscaling needs to happen?)

gabiX87, 2021-01-22:
yeah , you're right....

I lowered all stream to LOW _2  ... blinking stopped. I am going to write the feature request then to **"not load the stream in the background"** It would allow for many more high quality streams with and w/o the audio overlapping issue.

SvenVD, 2021-01-22:
That option is already available: in general.yml set enable_opportunistic_caching_next_screen: False . This will disable loading the next screen in background.

gabiX87, 2021-01-22:
Thank you. I completely missed it , not familiar with all the options yet. Tomorrow I'm going to test it.

closed 2021-03-11 by SvenVD