SvenVD/rpisurv #113: Updating VLC to "3.0.12 Vetinari" breaks functionality on 3.0.0-beta5 (v3_latest)

rummens1337, 2021-01-23:
### Environment
Raspberry Pi OS Desktop (32 bit) 2021-01-11
VLC 3.0.12
Branch v3_latest
Tag 3.0.0-beta5

### Issue

Seems to have something to do with the mmal-layer that is missing from the vlc executable output (```/usr/bin/vlc```).
Switching to vlc 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72) fixes the issue (removing the vlc install from install.sh was needed, otherwise it still updates to 3.0.12).

Screenshot of function that is failing
[image]

Output of ```sudo ./install.sh```
```bash
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
Aborting installation, upgrade to latest vlc player with mmal support
```

Output of ```vlc --version```
``` bash
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
VLC version 3.0.12 Vetinari (3.0.12-1-0-gd147bb5e7e)
Compiled by buildd on bm-wb-02 (Jan 18 2021 16:35:27)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

Output of ```/usr/bin/vlc```
```bash
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
[017bee80] main interface error: no suitable interface module
[017376d0] main libvlc error: interface "globalhotkeys,none" initialization failed
[017376d0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[017bee80] skins2 interface error: cannot initialize OSFactory
[017a3060] main playlist: playlist is empty
[017bee80] [cli] lua interface: Listening on host "*console".
VLC media player 3.0.12 Vetinari
Command Line Interface initialized. Type `help' for help.
>
```

### Possible steps to reproduce
Install Raspberry Pi OS on a Rpi 4 (4gb ram)
run ```sudo apt full-upgrade```
verify vlc version with ```vlc --version```
output should be similar to
```
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
VLC version 3.0.12 Vetinari (3.0.12-1-0-gd147bb5e7e)
Compiled by buildd on bm-wb-02 (Jan 18 2021 16:35:27)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

```git clone https://github.com/SvenVD/rpisurv.git```
```git checkout v3_latest```
```sudo ./install.sh```
(The following version will be installed: "3.0.0-beta5")

Loving the work you guys have done, keep it up!

SvenVD, 2021-01-23:
It seems they did not made the latest vlc 3.0.12 raspberry pi "aware" yet. https://github.com/RPi-Distro/vlc/commits/buster-rpt

I am curieus why it then still install it, from which repo did you fetch it? What is your apt sources.list

SvenVD, 2021-01-23:
Hi,

For me it is working with

root@raspberrypi:~# vlc --version
VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
VLC version 3.0.12 Vetinari (3.0.12-1-0-gd147bb5e7e)
Compiled by serge on arm-build.pitowers.org (Jan 18 2021 16:35:27)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public License;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.
root@raspberrypi:~# apt-cache madison vlc
       vlc | 3.0.12-0+deb10u1+rpt1 | http://archive.raspberrypi.org/debian buster/main armhf Packages
       vlc | 3.0.12-0+deb10u1 | http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages

Especially you need package  3.0.12-0+deb10u1+rpt1 and not the default 3.0.12-0+deb10u1 from (recompiled) debian, which does not include the mmal rpi specific patches.

rummens1337, 2021-01-24:
Weird that the vlc version without mmal rpi specific patches is installed when an update is executed.
The update was fetched on a cleanly installed rpi.
Anyway, kudo's for figuring it out :+1:

closed 2021-01-24 by rummens1337

sborn, 2021-10-06:
I know this is closed, but now I am having this exact problem. I have no idea what changed. Was there a new version this week? I was installing fresh installs last week and it was working and today I started over with fresh installs and they are all doing this (as is VLC). Could someone explain the problem above for someone a little less technical?