SvenVD/rpisurv #115: Keypresses not detected

ostermang, 2021-02-11:
Most keypresses are not being detected, so screens are not switching. Using the + or n keys nearly always works to advance one screen. However, using the number keys 0-3 on the keypad doesn't work very often, even when holding the key down. I have enabled debug logging and see the keypresses logged when they do work. However, most of the time when I press a key, the keypress is not logged and the screen does not change.

Running on a dedicated RPi 4 with 4GB RAM. CPUs all around 8-15% and only using about 125mb of memory.

Running a 2x2 screen and then three 1x1 screens to go full screen with the streams on the 2x2 screen. All streams are 640x480 h264. Opportunistic caching is turned off.

SvenVD, 2021-02-12:
Hi, if you rapidly press a different numbers after each other, and it is still in the midst of changing the screen, it wont register new presses for some very short time (depending how long the screen change takes).

If you wait for the screen to change before pressing a new key, does it work then?

ostermang, 2021-02-12:
Unfortunately, no. I have tried being very patient with it, pressing the button and then waiting for a bit. Every now and then it will pick up the key press after some delay, but usually it doesn't catch the key press at all. I've also watched the logs and as you mentioned, waited until it wasn't changing or reloading screens and it is still hit or miss if it catches the key press.

SvenVD, 2021-02-12:
> Unfortunately, no. I have tried being very patient with it, pressing the button and then waiting for a bit. Every now and then it will pick up the key press after some delay, but usually it doesn't [...]

Unfortunately, that is something I can not reproduce, if you wait for the screen to change then I see no occurrence of a missed keypress, do you also have the problem with the F functions keys?

ostermang, 2021-02-12:
Not sure if this has anything to do with it, but is it normal for the following lines in the log to be happening 1-2 times per second?

`2021/02/11 17:31:01 - l_default - DEBUG - total available malloc gpu mem value is 16M
2021/02/11 17:31:01 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:13M outputmem_bytes:13631488.0
2021/02/11 17:31:01 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:16M outputmem_bytes:16777216.0
2021/02/11 17:31:01 - l_default - DEBUG - free reloc gpu mem value is 448M
2021/02/11 17:31:01 - l_default - DEBUG - total available reloc gpu mem value is 492M
2021/02/11 17:31:01 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:448M outputmem_bytes:469762048.0
2021/02/11 17:31:01 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
2021/02/11 17:31:01 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
2021/02/11 17:31:02 - l_default - DEBUG - free malloc gpu mem value is 13M
2021/02/11 17:31:02 - l_default - DEBUG - total available malloc gpu mem value is 16M
2021/02/11 17:31:02 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:13M outputmem_bytes:13631488.0
2021/02/11 17:31:02 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:16M outputmem_bytes:16777216.0
2021/02/11 17:31:03 - l_default - DEBUG - free reloc gpu mem value is 448M
2021/02/11 17:31:03 - l_default - DEBUG - total available reloc gpu mem value is 492M
2021/02/11 17:31:03 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:448M outputmem_bytes:469762048.0
2021/02/11 17:31:03 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
2021/02/11 17:31:03 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
`

SvenVD, 2021-02-12:
Normally not, you could try to disable the memory checking

memory_usage_check: False in general.yml

ostermang, 2021-02-12:
Ok, I disabled the memory usage check which made the logs much easier to read. I still get the line below once per second and still have the issue with the keypresses not being picked up. Although it does seem to have improved a little bit.

`2021/02/11 18:40:56 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens`

ostermang, 2021-02-12:
One interesting thing that I noticed in the logs is that it typically catches keypresses about 10-15 seconds after the 19 second update. Out side of this window, they are missed.

SvenVD, 2021-02-12:
Maybe the scheduled health probing of your setup takes unusually long (what is your config), during which also no keypressed are registered. Does increasing interval_check_status: to something very high like 86400 solves something?

ostermang, 2021-02-22:
Sorry for the delayed response. I tried setting interval_check_status to 86400 and it does make it a little more responsive, but still not perfect. There are still times where pressing the number keys on the keypad just refuses to work, but pressing the + key works nearly instantly.

Here is my general.yml:
`advanced:
  update_stats: False
  enable_opportunistic_caching_next_screen: False
  memory_usage_check: False
  interval_check_status: 86400

fallbacks:
  displays:
    - hdmi: 0
      display_number: 2
      resolution:
        width: "1920"
        height: "1080"
      device_name: "fallback_device_0"`

And my display1.yml:
`essentials:
  disable_autorotation: true

  screens:
    - camera_streams:
        - url: "rtsp://redacted_1/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
        - url: "rtsp://redacted_2/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
        - url: "rtsp://redacted_3/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
        - url: "rtsp://redacted_4/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
      nr_of_columns: "2"
    - camera_streams:
        - url: "rtsp://redacted_1/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
    - camera_streams:
        - url: "rtsp://redacted_2/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
    - camera_streams:
        - url: "rtsp://redacted_3/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true
    - camera_streams:
        - url: "rtsp://redacted_4/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: true`

AadityaRathod97, 2021-04-01:
Hi, 
I want to be  able to change the stream with the help of GPIO input, is that possible 
if yes? where should I put my code and also how to stop the switching between streams  automatically

SvenVD, 2021-04-01:
> Hi,
> I want to be able to change the stream with the help of GPIO input, is that possible

This should be possible with, for example, an intermediate python script.
1. register you gpio input 
2. depending on gpio input, simulate the wanted standard keyboard keys presses( which keys rpisurv understand: https://github.com/SvenVD/rpisurv/tree/v3_latest#rpisurv-in-operation )

> if yes? where should I put my code and also how to stop the switching between streams automatically

disable_autorotation: False
https://github.com/SvenVD/rpisurv/blob/v3_latest/surveillance/conf/display1.yml

AadityaRathod97, 2021-04-02:
Hi,
Thanks For Your Help

i am fairly new to this 
so if I have wanted to add my gpio code where should i put my code
in which file should I add to?

AadityaRathod97, 2021-04-02:
HI,
ThankYou For Your Help

AadityaRathod97, 2021-04-02:
Hello again,
so i have the code which sends the key press ready and i am running it on autostart through rc.local but i think my code is not executing because i think rpisurv takes over
not sure 
how do i make sure that my code and rpi both are working together?

i am new to this.

SvenVD, 2021-04-03:
> Hello again,
> so i have the code which sends the key press ready and i am running it on autostart through rc.local but i think my code is not executing because i think rpisurv takes over
> not sure
> how do i make sure that my code and rpi both are working together?
> 
> i am new to this.

Maybe you can find some inspiration here: https://github.com/SvenVD/rpisurv-api

thezab, 2021-04-06:
Hi, 
So i wanted to know if it's possible to switch between streams under 3s? 
I tried interval_check_status and cache_next_screen but it still takes 9-11s to start the stream 
So i was wondering is there any way that can be improved? 

Thanks for your help:)

SvenVD, 2021-04-07:
Please do not use this issue for non-related questions.

Support forums are over here /https://community.rpisurv.net/

> Hi,
> So i wanted to know if it's possible to switch between streams under 3s?
> I tried interval_check_status and cache_next_screen but it still takes 9-11s to start the stream
> So i was wondering is there any way that can be improved?
> 
> Thanks for your help:)

You need to allow some time for next screens cache to warm up. Typically that takes longer then 3s. If a screen cycles and you wait for example 20 seconds, then cycling to the next screen will be seamless, if you cycle to fast you will see the streams still loading.

SvenVD, 2021-04-07:
> Sorry for the delayed response. I tried setting interval_check_status to 86400 and it does make it a little more responsive, but still not perfect. There are still times where pressing the number [...]

Just a guess, could you try to disable probing and see if that changes anything, for the keypresses delay you experience?

disable_autorotation: False
https://github.com/SvenVD/rpisurv/blob/v3_latest/surveillance/conf/display1.yml

I can not think of a reason why the + key would be registered better then another key. I still can not reproduce

on top of that rtsp_over_tcp is  generally not needed anymore in vlc player as it does not seem to have the smearing issues like omxplayer, at least that is what I experienced

SvenVD, 2021-06-13:
stale issue

closed 2021-06-13 by SvenVD

SvenVD, 2023-09-25:
Maybe keypresses of your numpad are registered differently, debug log of rpisurv could reveal what it detects