SvenVD/rpisurv #116: Touchscreen + hdmi resolution error

RinusHdev, 2021-03-15:
When the touchcreen is connected to the DSI interface, this display shows up as a non-hdmi device in the tvservice list.
This makes it so that when a 1920x1080 display is attached to one of the hdmi connections, the following line of code fails with a not-supported message.

`pygame.display.set_mode((resolution_width, resolution_height))`

SvenVD, 2021-03-15:
what is the exact error message?

RinusHdev, 2021-03-26:
Here's the output when connected to a different (4K) display:

`pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
2021/03/26 15:54:23 - l_default - INFO - Starting rpisurv 3.0.0-beta6
2021/03/26 15:54:23 - l_default - INFO - Auto detected display SNY-SONY_TV___00 at HDMI 0 with display number 2 3840 x 2160
2021/03/26 15:54:23 - l_default - INFO - Unique id of this installation is 4a972de61ab81fb765b4123aa499bf7d804d5966b14bdd7bba1f8cc7d0cc804f
Traceback (most recent call last):
  File "surveillance.py", line 188, in <module>
    screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
  File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 20, in __init__
    self._init_drawinstance()
  File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 214, in _init_drawinstance
    self.drawinstance = Draw([int(self.display["resolution"]["width"]), int(self.display["resolution"]["height"])], self.disable_pygame, self.name)
  File "/usr/local/bin/rpisurv/core/util/draw.py", line 52, in __init__
    self.surface = pygame.display.set_mode((self.resolution_width, self.resolution_height))
pygame.error: No video mode large enough for 3840x2160`

SvenVD, 2021-04-08:
So you have two devices connected,

One SNY-SONY_TV___00 at HDMI 0 with display number 2 3840 x 2160  and one touchscreen?

If you disconnect SNY-SONY_TV___00 at HDMI 0, then no hdmi will be found and fallback should be used. Does it work then? The reason why I am asking is that I am not sure mmal rpi customatisation to vlc can output to a DSI display, and then the fallback settings wont be of any help either.

SvenVD, 2021-06-13:
stale issue

closed 2021-06-13 by SvenVD