SvenVD/rpisurv #117: Smearing rearing its ugly head again

cqsimple, 2021-04-03:
I just found this and I believe it will fit right into my scheme of things for my cameras.  I have omxplayer version  Build date: Thu, 01 Aug 2019 12:50:43 +0000
        Version   : f543a0d [master]

I have 4 streams running and two things are happening.  1.  is the smearing  I understand that I should add trtsp_over_tcp: true o the rstp url but what is the syntax?
my url is : 
- camera_streams:
        - url: "rtsp://192.168.1.99:7447/bk9LbnfrOrwI3qX8" 
The other issue is that the system draws 4 boxes with the messages but the camera streams are off center to the left and to the top.  There is the bottom rpisurv.net messages at the bottom and on the right sde.  My TV is a 720 if that helps any

Thanks for any ideas to get this working!

SvenVD, 2021-04-03:
Hi,

Support forums are over here /https://community.rpisurv.net/

For all config option and syntax for Rpisurv 2 example see here, indentation is important and only use spaces: https://github.com/SvenVD/rpisurv/blob/v2_latest/surveillance/conf/surveillance.yml

omxplayer is depecrated, Rpisurv 3 uses vlc as backend and is now in beta.
https://github.com/SvenVD/rpisurv/tree/v3_latest
config options for Rpisurv 3 https://github.com/SvenVD/rpisurv/blob/v3_latest/surveillance/conf/display1.yml

For your streams that are off try https://github.com/SvenVD/rpisurv/tree/v3_latest#troubleshooting :
On a Raspberry Pi 3 and 4 it seems the default overscan settings are not good. If full screen is not used, if you have an unused bar in the bottom -> try to set disable_overscan=1 in /boot/config.txt

cqsimple, 2021-04-04:
Thank you Sven,

I reinstalled the version 3 rpisurv and after much trial and error I have a
screen working with 4 cameras.  When I setup a 2nd screen with two cameras
It flickered hen it switched to the second screen and continued to flicker
without rotating to the 1st screen.  Can I force the 2nd screen to be a 2x2
and show two unconnected feeds?  It would keep the camera aspects the same
and wouldn't get stretched out.

Thanks alot!

SvenVD, 2021-04-05:
Why would you want to show unconnectable feeds? If you really want you can

`#      #NOT recommendeded to enable disable_probing_for_all_streams, it disables the removal of dead feeds which also disables auto-rearranging the screen layout.
#      #Also if using multiple screens, enabling this will slow down rotation of one screen to the next if there are unconnectable streams
#      #Defaults to False
#      disable_probing_for_all_streams: False`

flickering can have different causes, most observed in the field is that you have to many (high cost) streams for the rpisurv gpu to handle at the same time. Caching caches the streams of the next screen so it will take up some gpu bandwidth for the benefit of seemless cycling to next screen.

You could disable caching with
enable_opportunistic_caching_next_screen: True

cqsimple, 2021-04-05:
OK message recieved.  I was able to clean up the streams by lowering the
resolution from the NVR to the lowest setting.

closed 2021-04-07 by SvenVD