SvenVD/rpisurv #118: Black Screen after Streams load

Duglz, 2021-04-13:
Got all of my streaming URLS setup at first I edited the demo config with my URLS, I'm just looking for a simple 2x3 grid. The streams will load one by one and once the last stream is present the screen goes black.

**display1.yml**

```
    - camera_streams:
        - url: "rtsp://192.168.1.11:81/BKYRD"
        - url: "rtsp://192.168.1.11:81/DWY"
        - url: "rtsp://192.168.1.11:81/DRBL"
        - url: "rtsp://192.168.1.11:81/LRM"
        - url: "rtsp://192.168.1.11:81/SDE"
        - url: "rtsp://192.168.1.11:81/SIDGATE"
```

**Errors in logs:**

```
2021/04/13 11:32:19 - l_default - ERROR - CameraStream: screen_manager_0_screen1_cam_stream1 rtsp://<hidden_username>:<hidden_password>@192.168.1.11:81/BKYRD Not Connectable (failed socket connect), configured timeout: 3 OSError(101, 'Network is unreachable')
2021/04/13 11:32:19 - l_default - ERROR - CameraStream: screen_manager_0_screen1_cam_stream2 rtsp://<hidden_username>:<hidden_password>@192.168.1.11:81/DWY Not Connectable (failed socket connect), configured timeout: 3 OSError(101, 'Network is unreachable')
2021/04/13 11:32:19 - l_default - ERROR - CameraStream: screen_manager_0_screen1_cam_stream3 rtsp://<hidden_username>:<hidden_password>@192.168.1.11:81/DRBL Not Connectable (failed socket connect), configured timeout: 3 OSError(101, 'Network is unreachable')
```

SvenVD, 2021-04-13:
What is the resolution of your streams? Could you try a lower resolution, a substream perhaps?

Duglz, 2021-04-14:
That did it, just adjusting the H in my url on 3 out of 6 cameras got all of them to show up. Great app! Threw a donation your way.

closed 2021-04-14 by Duglz