SvenVD/rpisurv #119: RTMP

Minglarn, 2021-04-17:
Hi! 
Trying to get FRIGATEs RTMP streams to work with RPISURV with no luck. 

Running you 3.0.7beta and get these in the logs: 
`ERROR - CameraStream: screen_manager_1_screen1_cam_stream1 rtmp://192.168.1.121/live/frigate_entren Not Connectable (ffprobe timed out, try increasing probe_timeout for this stream), configured timeout: 3`

It works well with VLC but not with rpisurv. 
Any hints?

http stream works but it puts some heavy load on my NUC so it's not an option.

akoryuerek, 2021-04-17:
Try to increase probe_timeout and it will work.

Minglarn, 2021-04-17:
It worked! Thanks... Must have missed that option... :)
probe_timeout is set to 20 and it works...

closed 2021-04-17 by SvenVD