SvenVD/rpisurv #121: Help a noob :)

MrPresident2, 2021-06-30:
Hey guys, I've been looking for a way to watch RTSP and HTTP streams and it seems like this is the only way and from what I see on other people it seems like it's working great. but I can make it work with my raspberry pi 4.
after going through the install guide when I reboot the pi I'm getting nothing just a black screen with what looks like hss screen but no stream, can anyone help me with that? maybe go with me on the installation process? I'm so lost ill even pay for someone's time to help me out.
Thanks

SvenVD, 2021-06-30:
Please provide output of

-  /usr/local/bin/rpisurv/logs/main.log
- /var/log/daemon.log
- you config file

so we can figure out what is wrong

remove password first

MrPresident2, 2021-06-30:
> * /usr/local/bin/rpisurv/logs/main.log

Thank you for your help,
I'm kind of a noob at all of this, how can I find those files? when I look at the sd card on my pc I don't see any of those files or directories.

MrPresident2, 2021-08-16:
no answer so ill close the post

closed 2021-08-16 by MrPresident2