SvenVD/rpisurv #123: Not Running

blackwellj, 2021-08-04:
Hi, I have followed the instructions on the GIT using a fresh install of Raspbian lite. 

When I run sudo systemctl status rpisurv

I get the following

● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2021-08-04 14:06:36 BST; 3s ago
  Process: 8694 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 8694 (code=exited, status=1/FAILURE)

Any help would be appreciated

SvenVD, 2021-08-04:
What is the output of /var/log/daemon.log?

si458, 2021-08-05:
Hi @SvenVD im having the same issue but i think mine is something to do with the display and the raspbian lite version
```
Aug  5 12:18:42 rpisurv rpisurv[547]: pygame 1.9.4.post1
Aug  5 12:18:42 rpisurv rpisurv[547]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug  5 12:18:42 rpisurv rpisurv[547]: 2021/08/05 12:18:42 - l_default - INFO - Starting rpisurv 3.0.0-beta7
Aug  5 12:18:42 rpisurv rpisurv[547]: 2021/08/05 12:18:42 - l_default - INFO - Auto detected display PHL-PHL_223V5 at HDMI 0 with display number 2 1920 x 1080
Aug  5 12:18:42 rpisurv rpisurv[547]: 2021/08/05 12:18:42 - l_default - INFO - Unique id of this installation is {LONGIDHERE}
Aug  5 12:18:45 rpisurv rpisurv[547]: Traceback (most recent call last):
Aug  5 12:18:45 rpisurv rpisurv[547]:   File "surveillance.py", line 188, in <module>
Aug  5 12:18:45 rpisurv rpisurv[547]:     screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
Aug  5 12:18:45 rpisurv rpisurv[547]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 20, in __init__
Aug  5 12:18:45 rpisurv rpisurv[547]:     self._init_drawinstance()
Aug  5 12:18:45 rpisurv rpisurv[547]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 214, in _init_drawinstance
Aug  5 12:18:45 rpisurv rpisurv[547]:     self.drawinstance = Draw([int(self.display["resolution"]["width"]), int(self.display["resolution"]["height"])], self.disable_pygame, self.name)
Aug  5 12:18:45 rpisurv rpisurv[547]:   File "/usr/local/bin/rpisurv/core/util/draw.py", line 52, in __init__
Aug  5 12:18:45 rpisurv rpisurv[547]:     self.surface = pygame.display.set_mode((self.resolution_width, self.resolution_height))
Aug  5 12:18:45 rpisurv rpisurv[547]: pygame.error: No video mode large enough for 1920x1080
Aug  5 12:18:46 rpisurv systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Aug  5 12:18:46 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
```

si458, 2021-08-05:
@SvenVD ok i tried using raspbian os 1.1gb version without the recommended software aand it works no problem
just seems to be the lite version that doesnt work for some reason?

SvenVD, 2021-08-05:
I have no idea for the lite version I did not test it

closed 2021-08-05 by SvenVD