SvenVD/rpisurv #125: Issues with Beta7

tubalainen, 2021-08-12:
Hi Sven.

Raspberry pi 3 b+
Raspberry OS Lite

Connected to a Philips full-HD TV via HDMI

Method of install
```
added gpu_mem=512 to /boot/config.txt
sudo apt get update ; sudo apt get upgrade
sudo apt get install git
git clone https://github.com/SvenVD/rpisurv
cd rpisurv
git checkout v3_latest
sudo ./install.sh
reboot
```

error from /var/syslog:
```
Aug 12 15:21:52 rpisurv systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 12 15:21:53 rpisurv rpisurv[682]: pygame 1.9.4.post1
Aug 12 15:21:53 rpisurv rpisurv[682]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug 12 15:21:53 rpisurv rpisurv[682]: 2021/08/12 15:21:53 - l_default - INFO - Starting rpisurv 3.0.0-beta7
Aug 12 15:21:53 rpisurv rpisurv[682]: 2021/08/12 15:21:53 - l_default - INFO - Auto detected display PHL-Philips_FTV at HDMI 0 with display number 2 1920 x 1080
Aug 12 15:21:53 rpisurv rpisurv[682]: 2021/08/12 15:21:53 - l_default - INFO - Unique id of this installation is 988ab0f34f17a1e68860f49abebad3f43d130fadd78d8d1597a5ba79f90e653a
Aug 12 15:21:55 rpisurv rpisurv[682]: Traceback (most recent call last):
Aug 12 15:21:55 rpisurv rpisurv[682]:   File "surveillance.py", line 188, in <module>
Aug 12 15:21:55 rpisurv rpisurv[682]:     screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
Aug 12 15:21:55 rpisurv rpisurv[682]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 20, in __init__
Aug 12 15:21:55 rpisurv rpisurv[682]:     self._init_drawinstance()
Aug 12 15:21:55 rpisurv rpisurv[682]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 214, in _init_drawinstance
Aug 12 15:21:55 rpisurv rpisurv[682]:     self.drawinstance = Draw([int(self.display["resolution"]["width"]), int(self.display["resolution"]["height"])], self.disable_pygame, self.name)
Aug 12 15:21:55 rpisurv rpisurv[682]:   File "/usr/local/bin/rpisurv/core/util/draw.py", line 52, in __init__
Aug 12 15:21:55 rpisurv rpisurv[682]:     self.surface = pygame.display.set_mode((self.resolution_width, self.resolution_height))
Aug 12 15:21:55 rpisurv rpisurv[682]: pygame.error: No video mode large enough for 1920x1080
Aug 12 15:21:56 rpisurv systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Aug 12 15:21:56 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
```
error from: cat /usr/local/bin/rpisurv/logs/main.log
```
2021/08/12 15:26:41 - l_default - INFO - Starting rpisurv 3.0.0-beta7
2021/08/12 15:26:41 - l_default - INFO - Auto detected display PHL-Philips_FTV at HDMI 0 with display number 2 1920 x 1080
2021/08/12 15:26:41 - l_default - INFO - Unique id of this installation is 988ab0f34f17a1e68860f49abebad3f43d130fadd78d8d1597a5ba79f90e653a

```

Config files:
Test 1: The default files provided as "showcase"
Test 2: Also tried with my own config with same error and results.

Please assist.

tubalainen, 2021-08-12:
More info here: https://www.tapatalk.com/groups/rpisurv/stuck-at-command-line-after-fresh-installs-t128-s10.html

SvenVD, 2021-08-13:
merged pull request

closed 2021-08-13 by SvenVD