SvenVD/rpisurv #127: V3-Beta7 - Issues with HikVision DS-2CD2T42WD

tubalainen, 2021-08-12:
Using the exact same configuration and settings the RTSP stream is very choppy in V3-Beta7 while it works great in V2-latest
The frames updates every 5-10 seconds.

SvenVD, 2021-08-13:
You can activate debug logging with  

`sed -i 's/level:.*/level: DEBUG/g' /usr/local/bin/rpisurv/conf/logging.yml; systemctl restart rpisurv`

However, errors are logged without debug also.

must be some resource constraint. You could start checking if it is the cpu, with `top`. Otherwise do you have the issue with one stream also?

tubalainen, 2021-08-13:
Well, this is the strange thing. It do idles around 0.4-0.5.
There is something fishy with the framerate and perhaps key frames coming from this cameras. 
Are you using any HW acceleration to offload the CPU? Such as `h264_mmal`?

The `camera_streams:` and config above do have other brands of cameras in them and they do work as expected.

I will get back to you with the debug logs.

tubalainen, 2021-08-13:
So nothing special showing in the debug logs as far as I can see. 

htop screenshot
[image]

```
# main.log
2021/08/13 08:51:08 - l_default - DEBUG - MAIN screen_manager_0: regular start update_active_screen (every 19 seconds since start of rpisurv)
2021/08/13 08:51:08 - l_default - DEBUG - screen_manager_0: update_active_screen screen_manager_0_screen10
2021/08/13 08:51:08 - l_default - DEBUG - Screen: Start polling connectivity for the camera_streams part of screen: screen_manager_0_screen10
2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream1 rtsp://<hidden_username>:<hidden_password>@192.168.21.38:554/Streaming/Channels/2 Connectable
2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream2 rtsp://<hidden_username>:<hidden_password>@192.168.21.32:554/Streaming/Channels/2 Connectable
2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream3 rtsp://<hidden_username>:<hidden_password>@192.168.21.30:554/Streaming/Channels/2 Connectable
2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream4 rtsp://<hidden_username>:<hidden_password>@192.168.21.31:554/Streaming/Channels/2 Connectable
2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream5 http://nvr:81/mjpg/cam6_hd Connectable
2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream6 rtsp://<hidden_username>:<hidden_password>@192.168.21.39:554/Streaming/Channels/2 Connectable
2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream7 http://192.168.21.207:5000/api/cam1_framsidan/person/best.jpg?h=800&bbox=1&motion=1 Connectable
2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream8 http://192.168.21.207:5000/api/cam3_framsidan/person/best.jpg?h=800&bbox=1&motion=1 Connectable
2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream9 http://192.168.21.207:5000/api/cam9_uppfart/person/best.jpg?h=800&bbox=1&motion=1 Connectable
2021/08/13 08:51:09 - l_default - DEBUG - screen_manager_0: update_connectable_camera_streams, disable_probing_for_all_streams is off for this screen, so using probes screen_manager_0_screen10
2021/08/13 08:51:09 - l_default - DEBUG - Screen: Connectable camera streams stayed the same, from 9 to 9, screen: screen_manager_0_screen10 does not need full redraw
2021/08/13 08:51:10 - l_default - DEBUG - free malloc gpu mem value is 8M
2021/08/13 08:51:10 - l_default - DEBUG - total available malloc gpu mem value is 12M
2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:8M outputmem_bytes:8388608.0
2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:12M outputmem_bytes:12582912.0
2021/08/13 08:51:10 - l_default - DEBUG - free reloc gpu mem value is 424M
2021/08/13 08:51:10 - l_default - DEBUG - total available reloc gpu mem value is 492M
2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:424M outputmem_bytes:444596224.0
2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
2021/08/13 08:51:10 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens

# screen_manager_0_screen10_cam_stream2.log
2021/08/13 08:44:22 - core.worker - DEBUG - Starting stream screen_manager_0_screen10_cam_stream2 with commandline ['/usr/bin/cvlc', '--aspect-ratio=640:360', '--vout', 'mmal_vout', '--network-caching', '500', '--no-video-title-show', '--mmal-display=hdmi-1', '--input-timeshift-granularity=0', '--repeat', '--mmal-vout-transparent', '--mmal-vout-window', '640x360+640+0', '--mmal-layer', '2000000000', '--no-audio', 'rtsp://username:superhemligt@192.168.21.32:554/Streaming/Channels/2']
2021/08/13 08:44:22 - core.worker - DEBUG - stopworker.value = screen_manager_0_screen10_cam_stream2 

```

For details on the behaviour, please checkout the videoclip here that I created: https://youtu.be/zo-q7u5Q-tA

SvenVD, 2021-08-21:
> Well, this is the strange thing. It do idles around 0.4-0.5.
> There is something fishy with detection of the framerate and perhaps key frames coming from this cameras.
> Are you using any HW acceleration to offload the CPU? Such as `h264_mmal`?
> 
> The `camera_streams:` and config above do have other brands of cameras in them and they do work as expected.
> 
> I will get back to you with the debug logs.

Yes hardware offloading is used with by the vlc instance which is spawned by Rpisurv

SvenVD, 2021-08-21:
> So nothing special showing in the debug logs as far as I can see.
> 
> htop screenshot
> [image]
> 
> ```
> # main.log
> 2021/08/13 08:51:08 - l_default - DEBUG - MAIN screen_manager_0: regular start update_active_screen (every 19 seconds since start of rpisurv)
> 2021/08/13 08:51:08 - l_default - DEBUG - screen_manager_0: update_active_screen screen_manager_0_screen10
> 2021/08/13 08:51:08 - l_default - DEBUG - Screen: Start polling connectivity for the camera_streams part of screen: screen_manager_0_screen10
> 2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream1 rtsp://<hidden_username>:<hidden_password>@192.168.21.38:554/Streaming/Channels/2 Connectable
> 2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream2 rtsp://<hidden_username>:<hidden_password>@192.168.21.32:554/Streaming/Channels/2 Connectable
> 2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream3 rtsp://<hidden_username>:<hidden_password>@192.168.21.30:554/Streaming/Channels/2 Connectable
> 2021/08/13 08:51:08 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream4 rtsp://<hidden_username>:<hidden_password>@192.168.21.31:554/Streaming/Channels/2 Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream5 http://nvr:81/mjpg/cam6_hd Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream6 rtsp://<hidden_username>:<hidden_password>@192.168.21.39:554/Streaming/Channels/2 Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream7 http://192.168.21.207:5000/api/cam1_framsidan/person/best.jpg?h=800&bbox=1&motion=1 Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream8 http://192.168.21.207:5000/api/cam3_framsidan/person/best.jpg?h=800&bbox=1&motion=1 Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - CameraStream: screen_manager_0_screen10_cam_stream9 http://192.168.21.207:5000/api/cam9_uppfart/person/best.jpg?h=800&bbox=1&motion=1 Connectable
> 2021/08/13 08:51:09 - l_default - DEBUG - screen_manager_0: update_connectable_camera_streams, disable_probing_for_all_streams is off for this screen, so using probes screen_manager_0_screen10
> 2021/08/13 08:51:09 - l_default - DEBUG - Screen: Connectable camera streams stayed the same, from 9 to 9, screen: screen_manager_0_screen10 does not need full redraw
> 2021/08/13 08:51:10 - l_default - DEBUG - free malloc gpu mem value is 8M
> 2021/08/13 08:51:10 - l_default - DEBUG - total available malloc gpu mem value is 12M
> 2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:8M outputmem_bytes:8388608.0
> 2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:12M outputmem_bytes:12582912.0
> 2021/08/13 08:51:10 - l_default - DEBUG - free reloc gpu mem value is 424M
> 2021/08/13 08:51:10 - l_default - DEBUG - total available reloc gpu mem value is 492M
> 2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:424M outputmem_bytes:444596224.0
> 2021/08/13 08:51:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
> 2021/08/13 08:51:10 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
> 
> # screen_manager_0_screen10_cam_stream2.log
> 2021/08/13 08:44:22 - core.worker - DEBUG - Starting stream screen_manager_0_screen10_cam_stream2 with commandline ['/usr/bin/cvlc', '--aspect-ratio=640:360', '--vout', 'mmal_vout', '--network-caching', '500', '--no-video-title-show', '--mmal-display=hdmi-1', '--input-timeshift-granularity=0', '--repeat', '--mmal-vout-transparent', '--mmal-vout-window', '640x360+640+0', '--mmal-layer', '2000000000', '--no-audio', 'rtsp://username:superhemligt@192.168.21.32:554/Streaming/Channels/2']
> 2021/08/13 08:44:22 - core.worker - DEBUG - stopworker.value = screen_manager_0_screen10_cam_stream2 
> ```
> 
> And for reference, ive tried the RTSP streams that are choppy in VLC on my windows machine and they all run great without any choppyness.
> [image]
> 
> For details on the behaviour, please checkout the videoclip here that I created: https://youtu.be/zo-q7u5Q-tA

Please note your password is in the debug logs.

Secondly I am a little bit confused by your logs which don't match your config provided earlier.

To rule out any raspberry pi resource constraints could you remove all config and only keep one stream which showed the choppiness ( for example with gras robot ). If this stream is still choppy then it must be 
- either the camera itself, which is overloaded by taking too many streams from other locations ( or even the cache of Rpisurv). Is this the case?
- some incompatibility between used codecs/brand/vlc. Which you could try to change codes on the camera itself and trial/error your way out.
- The network between pi and the choppy camera (low bandwitdh not stable). Try to reduce resolution or use a more compact codec like h265

tubalainen, 2021-09-01:
Waaaay too many parallel RTSP connections to one camera. Ive had like 5 connections per camera to different systems.
The older Hik cameras could not handle that.

I have now installed https://github.com/aler9/rtsp-simple-server as a redirect/proxy . And it has been working great! one connection to the camera <-> rtsp-simple-server <-> multiple services connects here.

One thing to add to the "notes" are the fact that different clients (players) are handling dropped frames differently. VLC on Windows for instance did not show the symptoms that rpisurv did....

"superhemligt" translate to "supersecret" - hope I did not expose my U/P.

closed 2021-09-01 by tubalainen