SvenVD/rpisurv #128: Unable to start rpisurv-Raspberry 4

marcusdevautomation, 2021-08-14:
Hi,
Thanks for the great work.
I am currently having trouble to start rpisurv on Raspberry 4.
I had completed install and configure,
but it seems the Pi would not load the Hikvision RTSP feed .
Viewing the RTSP feed via VLC no issue.
It now stuck on CLI only instead displaying GUI.

Checking on the logs only mentions starting the services repeatedly.
[image]
[image]
[image]
[image]

marcusdevautomation, 2021-08-14:
Problem solved after remove all the non related data & comments on the /etc/rpisurv file.
Now able to stream 1 channel from Hikvision IP Camera. Will try to stream more.
Thanks for the great from all.

SvenVD, 2021-08-14:
You are running v2. The default out of the box config for v2 is indeed not runnable and contains a lot of non-runnable examples.

However if you try Rpisurv v3 you will see that it has a demo config out of the box that is runnable right after installation.

closed 2021-08-14 by SvenVD

zeltak, 2021-08-17:
Hi having a similar issue i guess and confused by the documentation :)

i am in branch V3, using the default config but cant get the app to load after the reboot and only get the CLI

any clue?

sborn, 2021-09-12:
I'm in the same position...
> 
> 
> Hi having a similar issue i guess and confused by the documentation :)
> 
> i am in branch V3, using the default config but cant get the app to load after the reboot and only get the CLI
> 
> any clue?

marcusdevautomation, 2021-09-13:
I have not yet tested V3...but i had share the steps to getting started with V2.Link Youtube below,

https://youtu.be/M4Et7hpzsAg

One point to notice, make sure your camera & Wifi connected via same Switch, i had tested if you cascade switch within same vlan / subnet , it could not connect even though testing via VLC are OK.

SvenVD, 2021-09-13:
> I have not yet tested V3...but i had share the steps to getting started with V2.Link Youtube below,
>
> https://youtu.be/M4Et7hpzsAg
>
> One point to notice, make sure your camera & Wifi connected via [...]

Thanks for the video. 

The cameras should be connectable over layer 3 (ip) and with the ports of the respective streaming protocol. How  many switches are in between does not matter, even different subnets do not matter. Connecting to a camera on the other side of the world is also possible as long as the network connection (ip) between Rpisurv and the camera is OK

SvenVD, 2021-09-13:
> Hi having a similar issue i guess and confused by the documentation :)
> 
> i am in branch V3, using the default config but cant get the app to load after the reboot and only get the CLI
> 
> any clue?

You should provide /var/log/daemon.log to be able to troubleshoot

SvenVD, 2021-09-13:
> I'm in the same position...
> 
> > Hi having a similar issue i guess and confused by the documentation :)
> > i am in branch V3, using the default config but cant get the app to load after the reboot and only get the CLI
> > any clue?

You should provide /var/log/daemon.log to be able to troubleshoot, better open a new ticket for that

sborn, 2021-09-28:
Hi Sven. I did a fresh install, and expected a demo to pop up. Is that not the case?

The logs are cycling over and over with each restart of the command line. Here is a final chunk:

Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "surveillance.py", line 150, in <module>
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     screen_manager_main=ScreenManager('screen_manager_main', resolution, screens_cfg, fixed_width, fixed_height, $
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 23, in __init__
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     self._init_screens()
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 145, in _init_screens
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     self.all_screens.append(Screen("screen" + str(counter), screen_cfg, self.resolution, self.pygamescreen, self.$
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/local/bin/rpisurv/core/Screen.py", line 46, in __init__
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     self._init_camera_streams()
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/local/bin/rpisurv/core/Screen.py", line 55, in _init_camera_streams
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     cam_stream = CameraStream(cam_stream_name, camera_stream)
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/local/bin/rpisurv/core/CameraStream.py", line 44, in __init__
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     self.port = self.parsed.port
Sep 27 20:53:11 raspberrypi rpisurv[1281]:   File "/usr/lib/python2.7/urlparse.py", line 113, in port
Sep 27 20:53:11 raspberrypi rpisurv[1281]:     port = int(port, 10)
Sep 27 20:53:11 raspberrypi rpisurv[1281]: ValueError: invalid literal for int() with base 10: '<port>'
Sep 27 20:53:11 raspberrypi systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Sep 27 20:53:11 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Sep 27 20:53:32 raspberrypi systemd[1]: rpisurv.service: Service RestartSec=20s expired, scheduling restart.
Sep 27 20:53:32 raspberrypi systemd[1]: rpisurv.service: Scheduled restart job, restart counter is at 60.
Sep 27 20:53:32 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Sep 27 20:53:32 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Sep 27 20:53:32 raspberrypi rpisurv[1291]: No handlers could be found for logger "l_default"
Sep 27 20:53:32 raspberrypi rpisurv[1291]: pygame 1.9.4.post1
Sep 27 20:53:32 raspberrypi rpisurv[1291]: Hello from the pygame community. https://www.pygame.org/contribute.html
Sep 27 20:53:32 raspberrypi rpisurv[1291]: 2021/09/27 20:53:32 - l_default - INFO - Starting rpisurv 2.1.9
Sep 27 20:53:32 raspberrypi rpisurv[1291]: 2021/09/27 20:53:32 - l_default - INFO - Unique id of this installation is b55196229459b4b5e59dd939912879e67d539a$
Sep 27 20:53:35 raspberrypi rpisurv[1291]: Traceback (most recent call last):
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "surveillance.py", line 150, in <module>
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     screen_manager_main=ScreenManager('screen_manager_main', resolution, screens_cfg, fixed_width, fixed_height, $
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 23, in __init__
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     self._init_screens()
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 145, in _init_screens
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     self.all_screens.append(Screen("screen" + str(counter), screen_cfg, self.resolution, self.pygamescreen, self.$
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/local/bin/rpisurv/core/Screen.py", line 46, in __init__
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     self._init_camera_streams()
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/local/bin/rpisurv/core/Screen.py", line 55, in _init_camera_streams
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     cam_stream = CameraStream(cam_stream_name, camera_stream)
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/local/bin/rpisurv/core/CameraStream.py", line 44, in __init__
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     self.port = self.parsed.port
Sep 27 20:53:35 raspberrypi rpisurv[1291]:   File "/usr/lib/python2.7/urlparse.py", line 113, in port
Sep 27 20:53:35 raspberrypi rpisurv[1291]:     port = int(port, 10)
Sep 27 20:53:35 raspberrypi rpisurv[1291]: ValueError: invalid literal for int() with base 10: '<port>'