SvenVD/rpisurv #129: git clone https://github.com/SvenVD/rpisurv not working

MrPresident2, 2021-08-16:
can anyone help me with why git clone https://github.com/SvenVD/rpisurv won't work for me? I'm getting "-bash: git: command not found"

akoryuerek, 2021-08-16:
sudo apt install git

MrPresident2, 2021-08-16:
Thanks
What about the cd rpisurv? I'm getting -bash: cd: rpisurv/: No such file or directory

akoryuerek, 2021-08-16:
git clone https://github.com/SvenVD/rpisurv
cd rpisurv
git checkout v3_latest
sudo ./install.sh

you can read everything on https://github.com/SvenVD/rpisurv/tree/v3.0.0-beta7

MrPresident2, 2021-08-16:
yes I know I read it but git clone https://github.com/SvenVD/rpisurv got me an error 
same as cd rpisurv I'm getting -bash: cd: rpisurv: No such file or directory

MrPresident2, 2021-08-16:
sudo apt install git clone https://github.com/SvenVD/rpisurv
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package clone
E: Unable to locate package https://github.com/SvenVD
E: Couldn't find any package by glob 'https://github.com/SvenVD'
E: Couldn't find any package by regex 'https://github.com/SvenVD'

That's what I'm getting here

akoryuerek, 2021-08-16:
just copy & paste what I did poste before.

MrPresident2, 2021-08-16:
cool, I didn't thought about it let me write it again :)
git clone https://github.com/SvenVD/rpisurv 
Gives me:  "-bash: git: command not found"   
:)
can you help me with that?

akoryuerek, 2021-08-16:
sudo apt install git. please close :)

MrPresident2, 2021-08-16:
Thanks :) as I said before seems like you are mad or something but
sudo apt install git clone https://github.com/SvenVD/rpisurv
Got me this:
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package clone
E: Unable to locate package https://github.com/SvenVD
E: Couldn't find any package by glob 'https://github.com/SvenVD'
E: Couldn't find any package by regex 'https://github.com/SvenVD'

so it seems it's not working and after that cd rpisurv I'm getting -bash: cd: rpisurv: No such file or directory

maybe just try to help me and be nicer? I'm just trying to figure what's wrong sir
and I do appreciate your help

akoryuerek, 2021-08-16:
just type  git clone https://github.com/SvenVD/rpisurv and not sudo apt install git clone https://github.com/SvenVD/rpisurv

I´m really nice btw :D

MrPresident2, 2021-08-16:
Well, I reinstalled the os and now it's working, but I do have another problem. 
When I try to edit Display1.yml  /etc/rpisurv/display1.yml
I'm getting this: -bash: /etc/rpisurv/display1.yml: Permission denied
can you help me with that :)?

closed 2021-08-17 by MrPresident2