SvenVD/rpisurv #132: Config codec for input

Katsuryoku, 2021-08-20:
I'm on RPI4b 8Go.

I set a grid of 4 RTSP H264 stream (1920*1080) but only 3/4 works. I know that the last one can't work on gpu (memory limitation). 
Is-it possible to set last one on codec CPU directly ?

Secondly, h265 support is it possible on this project ?

SvenVD, 2021-08-21:
- First if your monitor is only 1920*1080 then it is more efficient to use lower resolution substreams so the pi does not need to spend resources to downscale them.
- Rpisurv v3 uses vlc player which can play h265.
- And lastly you can use `freeform_advanced_vlc_options` to passthrough any vlc option to vlc directly https://github.com/SvenVD/rpisurv/blob/v3_latest/surveillance/conf/display1.yml. So maybe there are options to force cpu only.

akoryuerek, 2021-08-30:
try --codec avcodec,none

Katsuryoku, 2021-08-31:
when I use `freeform_advanced_vlc_options` my service rpisurv doesn't start. And vlc can start with sudo and rpisurv work perfectly without `freeform_advanced_vlc_options`.

My display1.yml :

```
    - camera_streams:
         - url: "rtsp://user:password@IP:port/profile3/media.smp"
         - url: "rtsp://user:password@IP:port/profile3/media.smp"
         - url: "rtsp://user:password@IP:port/profile3/media.smp"
         - url: "rtsp://user:password@IP:port/profile3/media.smp"
          freeform_advanced_vlc_options: "--codec avcodec,none"
```

My systemctl status rpisurv at start :

```
rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-08-31 08:08:51 CEST; 3s ago
 Main PID: 1734 (rpisurv)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/rpisurv.service
           ├─1734 /bin/bash /usr/bin/rpisurv
           └─1735 python3 surveillance.py

août 31 08:08:51 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
août 31 08:08:51 raspberrypi rpisurv[1734]: /usr/local/bin/rpisurv/core/util/config.py:4: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read
août 31 08:08:51 raspberrypi rpisurv[1734]:   cfg = yaml.load(ymlfile)
août 31 08:08:51 raspberrypi rpisurv[1734]: /usr/local/bin/rpisurv/core/util/setuplogging.py:15: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Pleas
août 31 08:08:51 raspberrypi rpisurv[1734]:   logcfg = yaml.load(ymlfile)
août 31 08:08:51 raspberrypi rpisurv[1734]: pygame 1.9.4.post1
août 31 08:08:51 raspberrypi rpisurv[1734]: Hello from the pygame community. https://www.pygame.org/contribute.html
août 31 08:08:51 raspberrypi rpisurv[1734]: 2021/08/31 08:08:51 - l_default - INFO - Starting rpisurv 3.0.0-beta7
août 31 08:08:51 raspberrypi rpisurv[1734]: 2021/08/31 08:08:51 - l_default - INFO - Auto detected display SAM-SAMSUNG at HDMI 0 with display number 2 3840 x 2160
août 31 08:08:51 raspberrypi rpisurv[1734]: 2021/08/31 08:08:51 - l_default - INFO - Unique id of this installation is c39226555e3541fad35c37232a97b65fab050fef80d5fe40c3689deb604cdbbb
```
And my main log : 

```
2021/08/31 08:17:49 - l_default - INFO - Starting rpisurv 3.0.0-beta7
2021/08/31 08:17:49 - l_default - DEBUG - Make sure vlc binary can be executed as root
2021/08/31 08:17:49 - l_default - INFO - Auto detected display SAM-SAMSUNG at HDMI 0 with display number 2 3840 x 2160
2021/08/31 08:17:49 - l_default - INFO - Unique id of this installation is c39226555e3541fad35c37232a97b65fab050fef80d5fe40c3689deb604cdbbb
2021/08/31 08:17:49 - l_default - DEBUG - Start_time is 1630390669.1643097
2021/08/31 08:17:55 - l_default - DEBUG - screen_manager_0: Looking for config file conf/display1.yml
```

My systemctl status rpisurv after 7s :

```
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2021-08-31 08:08:58 CEST; 6s ago
  Process: 1734 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 1734 (code=exited, status=1/FAILURE)
```

SvenVD, 2021-09-04:
I suspect your indentation is not correct.
- use spaces not tabs
- make sure they are aligned correctly
- freeform_advanced_vlc_options should be on same level as url as it is a per url setting

closed 2021-11-11 by SvenVD