SvenVD/rpisurv #133: Can't get 2x2 - 1 camera always fails

geeklingo, 2021-09-06:
Just installed the latest v3 build, fairly simple setup using the 2x2 grid. One camera always fails to load though, all the streams are fine (confirmed via VLC) and if i move the camera order around it works but another fails. That said i notice that the ones that won't run together are the same model (Dahua PC-T5442TM-AS). The other 2 Dahua's stream fine together (IPC-HDBW4421E). I think H.265 might be the difference with the new ones (ones that fail).

dispay1.yml:
```
essentials:
  #Rpisurv will rotate over the following screens in the order they are configured here, if you only define one screen no rotation will happen

  #If you enable this option no auto rotation will happen, but you will still be able to select a screen by pressing and holding a Function Key (F1-F12) or numpad keys
  #Defaults to False
  #disable_autorotation: False

  screens:
    - camera_streams:
        - url: "rtsp://user:pwd@192.168.100.1:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://user:pwd@192.168.100.2:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://user:pwd@192.168.100.3:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://user:pwd@192.168.100.4:554/cam/realmonitor?channel=1&subtype=1" 
```
 
Status:
```
i@raspberrypi:~ $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-09-06 13:18:51 AEST; 6s ago
 Main PID: 22887 (rpisurv)
    Tasks: 88 (limit: 1023)
   CGroup: /system.slice/rpisurv.service
           ├─22887 /bin/bash /usr/bin/rpisurv
           ├─22888 python3 surveillance.py
           ├─22899 python3 surveillance.py
           ├─22900 python3 surveillance.py
           ├─22901 /usr/bin/vlc -I dummy --aspect-ratio=400:240 --vout mmal_vout --network-caching 500 --no-video-title-show --mmal-display=hdmi-1 --input-timeshift-granularity=0 --repeat --mmal-vout-transparent --mmal-vout-window 400x240+0+0 --mmal-layer 2000000000 --no-audio rtsp://user:pwd@192.168.100.1:554/cam/realmonitor?channel=1&subtype=1
           ├─22902 python3 surveillance.py
           ├─22903 /usr/bin/vlc -I dummy --aspect-ratio=400:240 --vout mmal_vout --network-caching 500 --no-video-title-show --mmal-display=hdmi-1 --input-timeshift-granularity=0 --repeat --mmal-vout-transparent --mmal-vout-window 400x240+400+0 --mmal-layer 2000000000 --no-audio rtsp://user:pwd@192.168.100.2:554/cam/realmonitor?channel=1&subtype=1
           ├─22904 python3 surveillance.py
           ├─22905 /usr/bin/vlc -I dummy --aspect-ratio=400:240 --vout mmal_vout --network-caching 500 --no-video-title-show --mmal-display=hdmi-1 --input-timeshift-granularity=0 --repeat --mmal-vout-transparent --mmal-vout-window 400x240+0+240 --mmal-layer 2000000000 --no-audio rtsp://user:pwd@192.168.100.3:554/cam/realmonitor?channel=1&subtype=1
           └─22907 /usr/bin/vlc -I dummy --aspect-ratio=400:240 --vout mmal_vout --network-caching 500 --no-video-title-show --mmal-display=hdmi-1 --input-timeshift-granularity=0 --repeat --mmal-vout-transparent --mmal-vout-window 400x240+400+240 --mmal-layer 2000000000 --no-audio rtsp://user:pwd@192.168.100.4:554/cam/realmonitor?channel=1&subtype=1

Sep 06 13:18:56 raspberrypi rpisurv[22887]: gpuserv: vc_gpuserv_init: starting initialisation
Sep 06 13:18:56 raspberrypi rpisurv[22887]: gpuserv: vc_gpuserv_init: could not open vchiq service: -1
Sep 06 13:18:56 raspberrypi rpisurv[22887]: [6850d240] mmal_codec decoder: VCSM init succeeded: Legacy
Sep 06 13:18:56 raspberrypi rpisurv[22887]: gpuserv: vc_gpuserv_execute_code: called without calling vc_gpuserv_init
Sep 06 13:18:56 raspberrypi rpisurv[22887]: gpuserv: vc_gpuserv_execute_code: vchiq service not initialised
Sep 06 13:18:56 raspberrypi rpisurv[22887]: error: XDG_RUNTIME_DIR not set in the environment.
Sep 06 13:18:56 raspberrypi rpisurv[22887]: error: XDG_RUNTIME_DIR not set in the environment.
Sep 06 13:18:56 raspberrypi rpisurv[22887]: error: XDG_RUNTIME_DIR not set in the environment.
Sep 06 13:18:56 raspberrypi rpisurv[22887]: error: XDG_RUNTIME_DIR not set in the environment.
Sep 06 13:18:56 raspberrypi rpisurv[22887]: [6850d240] main decoder error: buffer deadlock prevented
```

SvenVD, 2021-09-06:
Some things to try

- Can the NVR handle so many simultaneous streams?
- What if you lower the resolutions

geeklingo, 2021-09-07:
Hi, will check the other 2 points but the first is that NVR is pi 4.

closed 2021-11-29 by SvenVD