SvenVD/rpisurv #135: no /etc/rpisurv directory

sborn, 2021-09-30:
When trying to edit any of the yml files on a brand new install (it is hanging on the command prompt on boot), I get:

[ Path '/etc/rpisurv' is not a directory ]

When opening /etc/rpisurv as a file in nano (sudo nano /etc/rpisurv), it is the contents of what should be the display1.yml file:

> #THIS IS A YAML FILE, INDENTATION IS IMPORTANT. ALSO DO NOT USE TABS FOR INDENT$
> #THE FOLLOWING ARE EXAMPLES WITH ALL OPTIONS EXPLAINED, REMOVE OR ADAPT THEM WI$
> 
> essentials:
>   #rpisurv will rotate over the following screens in the order they are configu$

and so forth.

What happened? Can I build back the folder as it was intended?

sborn, 2021-09-30:
So, to be clear, the yml configuration files do not exist as far as I can tell. They would be in that directory, correct? They should all be in /etc/rpisurv?

akoryuerek, 2021-09-30:
Hi, the correct path for the config is...

/usr/local/bin/rpisurv/conf/display1.yml

sborn, 2021-09-30:
Oh, thank you. The documentation says /etc/rpisurv

I appreciate the help.

sborn, 2021-09-30:
The default files are not there. Should I upload them?

sborn, 2021-09-30:
pi@protect-pi-test-1:/usr/local/bin/rpisurv/conf $ dir
logging.yml  surveillance.yml
pi@protect-pi-test-1:/usr/local/bin/rpisurv/conf $

akoryuerek, 2021-09-30:
These files should be there if you made everything right.

pi@rasp-1:~ $ ls -la /usr/local/bin/rpisurv/conf/
total 24
drwxr-xr-x 2 pi pi 4096 Sep 28 15:22 .
drwxr-xr-x 8 pi pi 4096 Sep 15 00:47 ..
-rw-r--r-- 1 pi pi  643 Sep 30 18:54 display1.yml
-rw-r--r-- 1 pi pi 2934 Sep 14 18:47 display2.yml
-rw-r--r-- 1 pi pi 2290 Sep 14 18:47 general.yml
-rw-r--r-- 1 pi pi  832 Sep 14 18:46 logging.yml

sborn, 2021-09-30:
I just did a fresh install on a new install of PiOS. It didn't create those...

akoryuerek, 2021-09-30:
did you execute ./install.sh and confirmed everthing with "yes" during installation?

sborn, 2021-09-30:
Yes

akoryuerek, 2021-09-30:
I assume you didn´t install v3. For this you need to do "git checkout v3_latest" and the install.sh

sborn, 2021-09-30:
https://github.com/SvenVD/rpisurv doesn't pull the latest?

If I install this over the one I messed up will it fix it? Or should I start clean again 😞

SvenVD, 2021-09-30:
/etc/rpisurv is a symlink to /usr/local/bin/rpisurv/conf/ if the install went correctly.

rpisurv v3 is still in beta status, I did not have the time to make it production, that is why v2 is still the main version.

You could try the install.sh v3, I think it should work.

sborn, 2021-09-30:
I feel like a dope. Thanks guys. It's working great.

In case anyone stumbles in here, syntax is this: 

git clone -b v3_latest https://github.com/SvenVD/rpisurv

closed 2021-10-02 by sborn