SvenVD/rpisurv #136: VLC sabotage of fresh installs

sborn, 2021-10-06:
Earlier there was a report by rummens1337 about VLC: https://github.com/SvenVD/rpisurv/issues/113

I was doing installs from scratch last week and they were all working great. Fresh PiOS, then rpisurv as instructed. Each one worked to stream rtsp from unifi protect.

I had to put it on hold for a few days and got back to work with all new installs and now they are all doing what the rummens1337 post outlined. I do not have the old ones. Did VLC just upgrade?

I couldn't get VLC to work on more than one of my Win10 desktops (I have dozens), but if I downgraded on Win10 then VLC could stream rtsp without a problem on all of them. None of the pis can stream VLC today.

Can someone explain better the solution that Sven described?

akoryuerek, 2021-10-06:
What's your kernel version ?

sborn, 2021-10-06:
5.10
Everything is fresh out of the box and burned new.
I messed around on some other machines via LogMeIn last night and was getting really variable results. I'm going to try an ffmpeg test on them today when I get there (I can't test them from home because I'm not set up with Pis I can remote into via ssh). I'll report back.

akoryuerek, 2021-10-06:
I had also problems with 5.10.63-v7l+ by downgrading to 5.10.59-v7l+ everything will work again. Just execute  sudo rpi-update cb70bb6d7dd3327c539dd90090c1ec88006dbcef and reboot and please give a feedback.

sborn, 2021-10-06:
Wow. Worked perfectly! Will that be fixed in later upgrades? How do I avoid it from breaking again?

akoryuerek, 2021-10-06:
Yes you can avoid, don’t update :) I assume it´s not a problem with rpisurv and the new kernel it´s actually a problem with vlc  and the kernel.  I tried to explain this situation at the vlc forum with my broken English, but no one was interested or they also don’t update :)

sborn, 2021-10-06:
Yeah, I was asking if the fix would happen elsewhere. There is this from a while back: 
https://github.com/aler9/rtsp-simple-server/issues/223

sborn, 2021-10-08:
Which forum did you bring this up on? I have someone who wants to work on it.

kpfa, 2021-10-10:
What worked for us was to to downgrade (until a proper fix is released):
`sudo rpi-update cb70bb6d7dd3327c539dd90090c1ec88006dbcef` and reboot.

SvenVD, 2021-10-17:
For reference https://github.com/RPi-Distro/vlc/issues/48

daendaen, 2021-10-17:
> I had also problems with 5.10.63-v7l+ by downgrading to 5.10.59-v7l+ everything will work again. Just execute sudo rpi-update cb70bb6d7dd3327c539dd90090c1ec88006dbcef and reboot and please give a feedback.

you are my hero!!! Thank you!!!

nphawkins201, 2021-11-10:
Anyone have a fix to get this working on a fresh install of rasbian lite, it use to work by downgrading the rpi-update, but does not seem to work anymore, using version v3.0.0-beta7

SvenVD, 2021-11-10:
It seems to be fixed upstream https://github.com/RPi-Distro/vlc/issues/48.

nphawkins201, 2021-11-10:
Doing a fresh install with lite version released on October 30th, downgrade the rpi-update to ffdfc6e625eb38086b8c15eead2be90acab825c2, installed the v3.0.0-beta7 and unable to get the fresh install to display any of the demo items. running systemctl status rpisurv i get this:

pi@Cameraviewer:~ $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
     Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2021-11-09 18:37:06 MST; 494ms ago
    Process: 791 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
   Main PID: 791 (code=exited, status=1/FAILURE)
        CPU: 773ms

SvenVD, 2021-11-11:
What is the OS and vlc version you are running

nphawkins201, 2021-11-11:
`pi@Cameraviewer:~ $ vlc --version
VLC media player 3.0.16 Vetinari (revision 1.0.6-1680-ge976d3c3b)
VLC version 3.0.16 Vetinari (1.0.6-1680-ge976d3c3b)
Compiled by pi on serge-testpi (Oct 22 2021 14:49:51)
Compiler: gcc version 10.2.1 20210110 (Raspbian 10.2.1-6+rpi1)

pi@Cameraviewer:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@Cameraviewer:~ $

SvenVD, 2021-11-11:
There seems to be yet another and new issue with vlc on Bullseye
https://github.com/RPi-Distro/vlc/issues/49

nphawkins201, 2021-11-11:
I will give it a try, thanks.

SvenVD, 2021-11-11:
I opened new issue https://github.com/SvenVD/rpisurv/issues/141 for this to not keep this clean, this issue resolved

closed 2021-11-11 by SvenVD

JGVB, 2022-06-28:
I have the same problem on Bullseye with none of these fixes working