SvenVD/rpisurv #141: VLC issue on raspbian bullseye

SvenVD, 2021-11-11:


SvenVD, 2021-11-11:
https://github.com/RPi-Distro/vlc/issues/49

SvenVD, 2021-11-17:
https://github.com/RPi-Distro/vlc/issues/50

tmedina72, 2021-11-22:
I ran sudo rpi-update cb70bb6d7dd3327c539dd90090c1ec88006dbcef but still have bullseye after reboot, any suggestions?

SvenVD, 2021-11-28:
I think it is best to do a full reinstallation with a buster image

- https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Version_history
- https://downloads.raspberrypi.org/raspbian/images/

kpfa, 2021-11-30:
On a fresh install of bullseye (Raspbian lite) We loaded [the snap](https://snapcraft.io/install/vlc-pi/debian) and were then able to install rpisurv 3 from git.

`sudo apt update
sudo apt install snapd
sudo snap install core
sudo snap install vlc-pi --edge --devmode
sudo snap connect vlc-pi:hardware-observe`

However, another issue pops up 'tvservice is not supported when using the vc4-kms-v3d driver.'

`/usr/bin/rpisurv
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
/usr/local/bin/rpisurv/core/util/setuplogging.py:15: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  logcfg = yaml.load(ymlfile)
2021/11/30 10:57:10 - l_default - INFO - Starting rpisurv 3.0.0
/usr/bin/tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as modetest from libdrm-tests.
Traceback (most recent call last):
  File "/usr/local/bin/rpisurv/surveillance.py", line 170, in <module>
    displays=parse_tvservice()
  File "/usr/local/bin/rpisurv/surveillance.py", line 69, in parse_tvservice
    tvserviceresult_l = subprocess.check_output(['/usr/bin/tvservice', '-l'], text=True )
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/tvservice', '-l']' returned non-zero exit status 1.
root@raspberrypi:~/rpisurv# tvservice -l
tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as modetest from libdrm-tests.`

I'll start a new issue (if one doesn't already exist)

geeklingo, 2022-01-09:
FWIW, i resolved my issue yesterday by switching from H.265 to H.264 for my secondary stream (the one i used with rpisurv)

SvenVD, 2022-01-09:
> FWIW

on Bullseye?

SvenVD, 2022-01-09:
> FWIW, i resolved my issue yesterday by switching from H.265 to H.264 for my secondary stream (the one i used with rpisurv)

On Bullseye?

geeklingo, 2022-01-09:
> 

Yes, on the latest OS

SvenVD, 2022-01-09:
> > 
> 
> Yes, on the latest OS. And by switching, i mean on the cameras, didn't change anything server side.

Could you give some more details which version OS and which VLC version?

geeklingo, 2022-01-09:
My config details are in https://github.com/SvenVD/rpisurv/issues/147

Pi: 
```
Linux raspberrypi 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l
```

OS:
```
pi@raspberrypi:~ $ cat /etc/debian_version
10.11
```

VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)

SvenVD, 2022-01-09:
Yes Debian 10 is still Buster.

geeklingo, 2022-01-09:
Yeah ok, so the Kernel upgrade was what broke my install and gave this same error. Same kernel on Bullseye.

geeklingo, 2022-01-30:
scratch that, my fix seemed to only be temp, back to no feeds working.

CronoBandit, 2022-03-13:
rpisurv now works under Bullseye - you just need to switch to fake kms.
I just posted the work-around in ticket: https://github.com/SvenVD/rpisurv/issues/143
Can someone else confirm this works - thanks!

Jackfritt, 2022-06-02:
used the work-around on a fresh install of 2022-01-28-raspios-bullseye-armhf-lite.zip
on this Raspi
cat /proc/cpuinfo | grep Model
Model           : Raspberry Pi 2 Model B Rev 1.1

Its not in the v3_latest yet..

Works also with a big display.. 
Auto detected display DEL-Dell_U4919DW at HDMI 0

geeklingo, 2022-06-03:
Workaround didn't work for me @CronoBandit. I've just done a fresh bullseye install and still can't watch anything, keeps showing the connecting graphic but no issues with actual connection.

```
pi@PiNVR:/usr/local/bin/rpisurv/logs $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
     Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2022-06-03 13:47:20 AEST; 6min ago
   Main PID: 8082 (rpisurv)
      Tasks: 13 (limit: 1395)
        CPU: 54.000s
     CGroup: /system.slice/rpisurv.service
             ├─ 8082 /bin/bash /usr/bin/rpisurv
             ├─ 8083 python3 surveillance.py
             ├─ 8091 python3 surveillance.py
             ├─ 8092 python3 surveillance.py
             ├─ 8094 python3 surveillance.py
             ├─ 8096 python3 surveillance.py
             └─11247 /usr/bin/vcgencmd get_mem malloc_total

Jun 03 13:53:47 PiNVR rpisurv[8094]: 2022/06/03 13:53:47 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream3 attempts:38
Jun 03 13:53:47 PiNVR rpisurv[8096]: 2022/06/03 13:53:47 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream4 attempts:38
Jun 03 13:53:47 PiNVR rpisurv[11222]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun 03 13:53:47 PiNVR rpisurv[11222]: Try `vlc --help' for more information.
Jun 03 13:53:47 PiNVR rpisurv[11223]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun 03 13:53:47 PiNVR rpisurv[11223]: Try `vlc --help' for more information.
Jun 03 13:53:47 PiNVR rpisurv[11224]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun 03 13:53:47 PiNVR rpisurv[11224]: Try `vlc --help' for more information.
Jun 03 13:53:47 PiNVR rpisurv[11225]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun 03 13:53:47 PiNVR rpisurv[11225]: Try `vlc --help' for more information.
```
daemon.log
```
un  3 13:41:52 PiNVR rpisurv[4365]: 2022/06/03 13:41:52 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream1 attempts:14
Jun  3 13:41:52 PiNVR rpisurv[4367]: 2022/06/03 13:41:52 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream2 attempts:14
Jun  3 13:41:52 PiNVR rpisurv[4368]: 2022/06/03 13:41:52 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream3 attempts:14
Jun  3 13:41:52 PiNVR rpisurv[4370]: 2022/06/03 13:41:52 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream4 attempts:14
Jun  3 13:41:52 PiNVR rpisurv[5503]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:41:52 PiNVR rpisurv[5503]: Try `vlc --help' for more information.
Jun  3 13:41:52 PiNVR rpisurv[5504]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:41:52 PiNVR rpisurv[5504]: Try `vlc --help' for more information.
Jun  3 13:41:53 PiNVR rpisurv[5505]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:41:53 PiNVR rpisurv[5505]: Try `vlc --help' for more information.
Jun  3 13:41:53 PiNVR rpisurv[5506]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:41:53 PiNVR rpisurv[5506]: Try `vlc --help' for more information.
Jun  3 13:42:02 PiNVR rpisurv[4365]: 2022/06/03 13:42:02 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream1 attempts:15
Jun  3 13:42:02 PiNVR rpisurv[4367]: 2022/06/03 13:42:02 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream2 attempts:15
Jun  3 13:42:02 PiNVR rpisurv[4368]: 2022/06/03 13:42:02 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream3 attempts:15
Jun  3 13:42:03 PiNVR rpisurv[4370]: 2022/06/03 13:42:03 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream4 attempts:15
Jun  3 13:42:03 PiNVR rpisurv[5579]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:42:03 PiNVR rpisurv[5579]: Try `vlc --help' for more information.
Jun  3 13:42:03 PiNVR rpisurv[5580]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:42:03 PiNVR rpisurv[5580]: Try `vlc --help' for more information.
Jun  3 13:42:03 PiNVR rpisurv[5581]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:42:03 PiNVR rpisurv[5581]: Try `vlc --help' for more information.
Jun  3 13:42:03 PiNVR rpisurv[5583]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:42:03 PiNVR rpisurv[5583]: Try `vlc --help' for more information.
Jun  3 13:42:13 PiNVR rpisurv[4365]: 2022/06/03 13:42:13 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream1 attempts:16
Jun  3 13:42:13 PiNVR rpisurv[4367]: 2022/06/03 13:42:13 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream2 attempts:16
Jun  3 13:42:13 PiNVR rpisurv[4368]: 2022/06/03 13:42:13 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream3 attempts:16
Jun  3 13:42:13 PiNVR rpisurv[4370]: 2022/06/03 13:42:13 - core.worker - INFO - Trying to restart screen_manager_0_screen1_cam_stream4 attempts:16
Jun  3 13:42:13 PiNVR rpisurv[5663]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'
Jun  3 13:42:13 PiNVR rpisurv[5663]: Try `vlc --help' for more information.
[... cut ...]
#      # After installation you can start Rpisurv and the demo should be started
#      #######################################################################################################

    - camera_streams:
        - url: "rtsp://admin:mypwd@192.168.0.59:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://admin:mypwd@192.168.0.56:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://admin:mypwd@192.168.0.53:554/cam/realmonitor?channel=1&subtype=1"
        - url: "rtsp://admin:mypwd@192.168.0.51:554/cam/realmonitor?channel=1&subtype=1"

```

aifesteves, 2022-06-09:
AMD64 debian bullseye has the same problem.

CronoBandit, 2022-06-09:
Still working for me.
I’d like to help, but those of you saying the work-around doesn’t work, seem to be running different environments. 
Here’s mine:

Used rpi-imager to burn sdcard. Choose OS “Raspberry Pi OS (32-bit)” aka recommended.
Or, you can choose “Raspberry Pi OS Lite (32-bit)” or “Raspberry Pi OS Full (32-bit)” .
Note: I have **not** tested rpisurv nor this work-around under 64-bit.

uname -a
`Linux hostpi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`

Note: this is armv7l – not AMD, not aarch64. Only vlc on ARM environments will have the mmal libs compiled into vlc, which is why @geeklingo sees those  vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1

cat /proc/cpuinfo | grep Model
`Model		: Raspberry Pi 4 Model B Rev 1.5`

fkms -vs- kms (this is the work-around) :
grep vc4 /boot/config.txt 
```
#dtoverlay=vc4-kms-v3d
dtoverlay=vc4-fkms-v3d
```

Note: The main reason previously-working rpisurv systems broke when folks upgraded to Bullseye 32bit from Buster 32-bit, was the default config.txt in Bullseye switched over to kms api. If you go back and look at a typical Pi Buster install, you will see it loads fkms. I think the move to kms is going to be the way forward from what developer are saying, as it is the newest api. However, it is still very much “in-progress” and currently missing the mmal hooks needed to access the Pi’s GPU. Therefore, those that choose to run Bullseye, we must deliberately switch back to the fkms api until then.

vlc --version
```
VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
VLC version 3.0.16 Vetinari (1.0.6-1682-g88158c836)
Compiled by pi on serge-testpi (Nov 10 2021 15:40:45)
Compiler: gcc version 10.2.1 20210110 (Raspbian 10.2.1-6+rpi1)
```

apt list --installed | grep vlc
```
libvlc-bin/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
libvlc5/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
libvlccore9/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-bin/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-data/stable,now 3.0.16-1+rpi1+rpt2 all [installed,automatic]
vlc-l10n/stable,now 3.0.16-1+rpi1+rpt2 all [installed,automatic]
vlc-plugin-access-extra/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-base/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-notify/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-qt/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-samba/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-skins2/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-video-output/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-video-splitter/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-visualization/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed]
```

geeklingo, 2022-06-14:
Man, just did a clean build and same same.

uname -a
`Linux kitchennvr 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`

cat /proc/cpuinfo | grep Model
`Model           : Raspberry Pi 4 Model B Rev 1.1`

grep vc4 /boot/config.txt
`dtoverlay=vc4-fkms-v3d`

vlc --version
```
VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
VLC version 3.0.16 Vetinari (1.0.6-1682-g88158c836)
Compiled by pi on serge-testpi (Nov 10 2021 15:40:45)
Compiler: gcc version 10.2.1 20210110 (Raspbian 10.2.1-6+rpi1)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public Licence;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.
```

apt list --installed | grep vlc

```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libvlc-bin/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
libvlc5/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
libvlccore9/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-bin/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-data/stable,now 3.0.16-1+rpi1+rpt2 all [installed,automatic]
vlc-l10n/stable,now 3.0.16-1+rpi1+rpt2 all [installed,automatic]
vlc-plugin-access-extra/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-base/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-notify/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-qt/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-samba/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-skins2/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-video-output/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-video-splitter/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc-plugin-visualization/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed,automatic]
vlc/stable,now 3.0.16-1+rpi1+rpt2 armhf [installed]
```

main.log
```
2022/06/14 10:10:37 - l_default - INFO - Starting rpisurv 3.0.0
2022/06/14 10:10:37 - l_default - ERROR - Could not autodetect displays, use values from fallback config
2022/06/14 10:10:37 - l_default - ERROR - Using config display fallback_device_0 at HDMI 0 with display number 2 1920 x 1080
2022/06/14 10:10:37 - l_default - INFO - Unique id of this installation is ce24571e341b476d89a15a1a2f39ae83ceb95eec1f417053e4f687a04a508710
```

vlc
```
VLC media player 3.0.16 Vetinari (revision 1.0.6-1682-g88158c836)
[009c2020] main interface error: no suitable interface module
[0094fb98] main libvlc error: interface "globalhotkeys,none" initialization failed
[0094fb98] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[009c2020] skins2 interface error: cannot initialize OSFactory
[009ae878] main playlist: playlist is empty
[009c2020] [cli] lua interface: Listening on host "*console".
VLC media player 3.0.16 Vetinari
Command Line Interface initialized. Type `help' for help.
>
```

CronoBandit, 2022-06-14:
@geeklingo 
Good, your ENV looks identical to mine now. And it appears you are now past the lack of mmal support errors from vlc – cool.

However, your errors that stick-out to me are:

```
Jun 14 10:24:28 kitchennvr rpisurv[2202]: 2022/06/14 10:24:28 - l_default - ERROR - Could not autodetect displays, use values from fallback config
Jun 14 10:24:28 kitchennvr rpisurv[2202]: 2022/06/14 10:24:28 - l_default - ERROR - Using config display fallback_device_0 at HDMI 0 with display number 2 1920 x 1080
. . . 
Jun 14 10:24:29 kitchennvr rpisurv[2202]: pygame.error: ---> No video mode large enough for 1920x1080  <---
```

Wrong resolution -maybe ??
Almost like issues with your HDMI monitor. From what I hear, the Pi can be a bit finicky on some hardware. Try taking a look at this page for tips on troublesome HDMI displays:
https://learn.adafruit.com/using-weird-displays-with-raspberry-pi/everything-else
Its pretty in-depth with all sorts of things to try, even driving specific resolutions … please let the community know if any of that helped or not.

geeklingo, 2022-06-14:
@CronoBandit yeah i am using the Raspberry 7" touch screen. Where is it pulling that resolution config from?

CronoBandit, 2022-06-14:
@geeklingo 
Not sure. I would think that should have auto-configured. Perhaps poor ram-split is affecting yours.... have you played around with your gpu & cma memory-split settings? I noticed you have "gpu_mem" cranked all the way up to 512 ?? That's always given mine issues. It's going to eat into cma (auto-resize smaller). I remember reading somewhere that the Pi 4 B does not need gpu memory allocated that way anymore.... instead, keep the setting in, but drop it down and add a higher cma value in the overlay like this:

/boot/config.txt
```
dtoverlay=vc4-fkms-v3d-pi4,cma-512
gpu_mem=128
boot_delay=2
```

These settings works really nicely on mine, streaming 9 feeds on a 1920x1080 hdmi Asus monitor. Maybe a small boot delay might help with screen detect.

Also, I'll post this too for comparison...
/boot/cmdline.txt
`console=serial0,115200 console=tty1 root=PARTUUID=705a4f4a-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles`
I wanna say splash / no splash might help with auto-resolution.

geeklingo, 2022-06-14:
ahh ok will try a couple of things.

My cmdline.txt
`console=serial0,115200 console=tty1 root=PARTUUID=88e61625-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles`

si458, 2022-12-09:
my pi needed updating, so i updated all its packaged and now it runs vlc `v3.0.17.4-0+deb11u1+rpi1+rpt4`
but i keep getting the mmal is missing
`Dec 09 13:27:29 fthcctvpi rpisurv[706]: vlc: unknown option or missing mandatory argument `--mmal-display=hdmi-1'`

i was also running buster, then i stupid decided to try upgrading my pi to bullseye to see if the vlc was fixed there
now its running bullseye and vlc `3.0.18-0+deb11u1+rpi1` which is still producing the mmal is missing message

how can i now install an old version of vlc which has the mmal compiled?