SvenVD/rpisurv #143: Raspbian Bullseye - tvservice is not supported w/ vc4-kms-v3d driver

kpfa, 2021-11-30:
On a fresh install of bullseye (Raspbian lite) We loaded [the snap](https://snapcraft.io/install/vlc-pi/debian) and were then able to install rpisurv 3 from git.
```
sudo apt update
sudo apt install snapd
sudo snap install core
sudo snap install vlc-pi --edge --devmode
sudo snap connect vlc-pi:hardware-observe
```

However, another issue pops up 'tvservice is not supported when using the vc4-kms-v3d driver.'

```
/usr/bin/rpisurv
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
/usr/local/bin/rpisurv/core/util/setuplogging.py:15: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  logcfg = yaml.load(ymlfile)
2021/11/30 10:57:10 - l_default - INFO - Starting rpisurv 3.0.0
/usr/bin/tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as modetest from libdrm-tests.
Traceback (most recent call last):
  File "/usr/local/bin/rpisurv/surveillance.py", line 170, in <module>
    displays=parse_tvservice()
  File "/usr/local/bin/rpisurv/surveillance.py", line 69, in parse_tvservice
    tvserviceresult_l = subprocess.check_output(['/usr/bin/tvservice', '-l'], text=True )
  File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/tvservice', '-l']' returned non-zero exit status 1.

root@raspberrypi:~/rpisurv# tvservice -l
tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as modetest from libdrm-tests.
```

I'll start a new issue (if one doesn't already exist)

_Originally posted by @kpfa in https://github.com/SvenVD/rpisurv/issues/141#issuecomment-982927930_

SvenVD, 2021-12-04:
Thanks for reporting. However since VLC is not working anyways on bullseye (at the moment). I will put a fix for this on hold too. Recommend to use buster for the time being:  https://github.com/SvenVD/rpisurv/issues/141#issuecomment-981159241

CronoBandit, 2022-03-13:
@SvenVD 
Got rpisurv (the broken tvservice) to work on Bullseye with one simple edit :

Edit: /boot/config.txt

Change --> dtoverlay=vc4-kms-v3d
         To --> dtoverlay=vc4-fkms-v3d

After this change, & a reboot, rpisurv works fine again on pi 4 and 11.2 Debian :+1: 

I can't take credit for this ... saw it in another (unrelated) discussion recently. Figured I'd give it a go .... success !

Also, not sure what (if any) "trade-off" is for fkms -vs- kms. My weak understanding is maybe a kms-like passthru to the legacy hardware API. This might be the work-around needed until KMS matures fully.

felker, 2022-04-13:
@CronoBandit did you already have rpisurv installed before that change? If not, how were you able to install it without the following error:
> Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
Aborting installation, upgrade to latest vlc player with mmal support

What VLC version are you using? I have 
```
VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
```

felker, 2022-04-13:
@kpfa did you have to heavily modify `install.sh` to prevent it from re-installing `vlc` from `apt` and shadowing the snap-distributed `vlc-pi`?

CronoBandit, 2022-04-13:
@felker 
rpisurv is only for Raspberry Pi hardware. 
```
$ uname -a
Linux host1 5.10.103-v7l+ #1530 SMP Tue Mar 8 13:05:01 GMT 2022 **armv7l** GNU/Linux
$ sudo apt list --installed | grep vlc/
vlc/stable,now 3.0.16-1+rpi1+rpt2 **armhf** [installed]
```
So that we don't loose context here.... _The Raspbian OS base install includes vlc already installed_, and that vlc is already has mmal support which is specific to the GPU hardware on Pi. The purpose of this thread --> it seems rpisurv broke when the Raspbian OS migrated from Buster to Bullseye. My comment above simply offers a work-around to get it working again on Bullseye.

felker, 2022-04-13:
I am aware that this is for Raspberry Pi's--- trying to use this workaround on my 4B with a fresh Bullseye install. I am also using the 64-bit OS, which is another difference here. mmal does not seem to be supported on the official Raspberry Pi VLC (https://github.com/RPi-Distro/vlc/issues/49), so I thought that the alternative VLC suggested by @kpfa might be suitable, along with your FKMS suggestion. But you would need to modify `install.sh`, just not sure of the best way to do so.

CronoBandit, 2022-04-14:
My apologies ... it's got to be the 64-bit OS causing your added issues then.

To repeat my success:
_Burn the SD card via rpi-imager: choose the default image --> "Raspberry Pi OS (32bit) 0.8 GB" ( or 32-bit Full 2.2 GB )
Boot the Pi, go through the 1st time setup - reboot.
Edit: /boot/config.txt
     Change: dtoverlay=vc4-kms-v3d  --> To: dtoverlay=vc4-fkms-v3d
Reboot
Now install rpisurv via github, follow the install doc._

Hope that helps!

UntouchedWagons, 2022-06-30:
Yeah I still can't install rpisurv on Bullseye because the installer tries to install versions of VLC that don't exist

JGVB, 2022-07-24:
[image]

I am getting this error even on buster.  Any ideas?

si458, 2022-10-23:
not sure if it will help others but i used this comment and it worked a treat!
switch the GL driver to `legacy` mode
https://github.com/RPi-Distro/vlc/issues/50#issuecomment-1194943411

UntouchedWagons, 2022-11-30:
I tried the dtoverlay solution and I still can't install rpisurv on dietpi 8

gitterman, 2022-12-04:
I have a model 1 which serves as a doorbell monitor.
As such, it uses an USB camera and a small car monitor attached to the sdtv port.
The monitor is only switched on if someone is ringing the bell.
As the old system doesn't support the latest ssh anymore, I decided to upgrade to bullseye only to find that tvservice doesn't work with it.
I always get 'tvservice is not supported when using the vc4-kms-v3d driver'
I tried to change the driver in config.txt to vc4-fkms-v3d but no change.
I also used raspi-config to disable the GL driver but still no change.
Why the heck can't I disable the vc4-fkms-v3d once and for good?
I don't need HDMI at all.
What can I do to disable the HDMI driver and to switch the monitor on/and off ?

si458, 2022-12-04:
> I have a model 1 which serves as a doorbell monitor.
> As such, it uses an USB camera and a small car monitor attached to the sdtv port.
> The monitor is only switched on if someone is ringing the bell. [...]

Are you rebooting the pi after every change you are doing? You need to reboot the pi after changing graphics settings

gitterman, 2022-12-04:
Yes, I did reboot after the changes I did to config.txt.
In the meantime I did revert to buster and with that everything works fine.
(for reference: https://forums.raspberrypi.com/viewtopic.php?t=257631)
I did find many links from people having problems with switching the display on and off, but no working solution for bullseye.
Bullseye is out now for more than a year, and things didn't seem to have improved.