SvenVD/rpisurv #144: RTSPS Support?

gavzzorz, 2021-12-02:
Hi, Sorry to log this under Issues as its more of a query...

Does RPISurv support RTSPS? we have been using it for years and its been dead reliable but have had to upgrade our CCTV from Unifi Video to Unifi Protect and have now found that Standard RTSP isn't an option anymore and the only option is RTSPS.

I have tried the link by editing the original config and replacing the URL's but it causes RPISurv service to crash on startup.
At first i thought this was a config issue as i have known this to happen before when the config has a typo or a space somewhere but sadly that isn't the case this time :(.

This is the RTSPS address Protect is giving me...
rtsps://***.***.***.***:7441/nNc25veoEF3RZEzD?enableSrtp

Hope thats fine.

Thanks

moonlightjames, 2021-12-09:
We had ran into the same issue with the Unifi Protect, 

But on our system, it offers the RTSPS, and standard un-authenticated RTSP
[image]

We are running a couple Pi's connecting to the RTSP, 
and not been able to get RTSPS to work with several other pi-based viewers,

gavzzorz, 2021-12-09:
Found that you can change the port to to 7447 and remove the end then works fine still on the latest version of protect.

Original 
rtsps://...:7441/nNc25veoEF3RZEzD?enableSrtp

Changed to...
rtsp://...:7447/nNc25veoEF3RZEzD

SvenVD, 2021-12-10:
It seems vlc has issues with rtsps https://community.ui.com/questions/Help-streaming-via-RTSPS-from-UniFi-Protect-NVR/43db6df6-27e4-49bd-baf9-d96652f89047

closed 2022-01-02 by SvenVD