SvenVD/rpisurv #146: Ability to add widgets or browser tabs?

ajquick, 2021-12-25:
Long story short, I want to have my video feeds only taking up 1/4 of the screen and was wondering if there was any way to add additional things to the screen?

I'd love if I could display a webpage to the side of the cameras and/or display useful information.

Widget ideas:

* Date / Time
* Weather / Temperature
* Browser window (with refresh on interval)
* Display config only at certain times of day

I will probably create dynamic images that update on a schedule if that's the only way.

SvenVD, 2021-12-26:
If you can convert your information to an image and host that on a webserver ( or local image), point Rpisurv to it, then Rpisurv will fetch and keep it fresh. If  change information on the image, then Rpisurv will refresh by default every 19 seconds

ajquick, 2021-12-26:
Tutorial for adding Google's Weather to your screen. This will run every hour and relies upon this project: https://github.com/maaaaz/webscreenshot

```
sudo pip install webscreenshot
sudo apt-get update
sudo apt-get install xvfb
sudo apt-get install phantomjs
sudo crontab -e
```

Add the following code to your Crontab:

`0 * * * * webscreenshot -z "weather.png" --no-error-file --crop "175,60,665,350" https://www.google.com/search?q=YOUR+CITY+weather >/dev/null 2>&1`

Replace YOUR+CITY with whatever your city and state is.

Add to your display.yml file:

```
- url: "file:///home/pi/weather.png"   
  imageurl: true
```

SvenVD, 2021-12-26:
Thats really nice, thanks for sharing!

ajquick, 2021-12-27:
Another neat thing I just got working.

The ability to control screen rotation using the TV remote over HDMI (since I am using a TV, not a PC monitor).

Need to install cec-client by using the following: https://github.com/Pulse-Eight/libcec

Libcec, requires installing a compiler and compiling the scripts on your Pi, but once it's done cec-client becomes available.

After that you need to run cec-client at boot and feed the messages into a script as follows:

`cec-clien | /home/pi/cec.sh`

The file cec.sh has the following code:

```
#!/bin/bash
while read oneline
do
   keyline=$(echo $oneline | grep " key ")
   if [ -n "$keyline" ]; then
      strkey=$(grep -oP '(?<=sed: ).*?(?= \()' <<< "$keyline")
      strstat=$(grep -oP '(?<=key ).*?(?=:)' <<< "$keyline")
      strpressed=$(echo $strstat | grep "pressed")

      if [ -n "$strpressed" ]; then
         case "$strkey" in
            "left")
                sudo python /home/pi/firstscreen.py
                ;;
            "right")
                sudo python /home/pi/nextscreen.py
                ;;
         esac
      fi
   fi
done
```

There are also two Python files (because I couldn't get the Python to execute directly in the Bash file. This will require installing pip install keyboard (saw it on the API repo).

firstscreen.py:

```
import keyboard

keyboard.press_and_release('F1')
```

nextscreen.py:

```
import keyboard

keyboard.press_and_release('n')
```

Boom. The left button on my remote goes to screen 1 (F1). The right button goes to the next screen.

You can also mess around with other buttons and features such as setting up to start cycling, down to stop cycling... and any number keys can correspond to which screen you want. Ideally it would also check to see if the TV is on and if it isn't, then Rpisurv would shut down until the TV resumes showing the security cameras.

SvenVD, 2021-12-30:
Nice stuff, On the forum we also had this https://www.tapatalk.com/groups/rpisurv/show-weather-forcast-t165.html#p838

closed 2022-02-27 by SvenVD