SvenVD/rpisurv #147: Updated to the latest pi os and now can't connect

geeklingo, 2022-01-06:
I just updated my Pi to the latest OS, and now i can't seem to connect to my 4 streams. Was working fine before on older Pi OS.

Also did a GIT pull and install to ensure that i am running the latest version of rpisurv. Is there a log location i can check for rpisurv?

geeklingo, 2022-01-07:
Actually, the pi update seems to have broken my VLC

```
pi@raspberrypi:/etc/rpisurv $ vlc rtsp://admin:pwd@192.168.1.59:554/cam/realmonitor?channel=1&subtype=1
[1] 30456
pi@raspberrypi:/etc/rpisurv $ VLC media player 3.0.12 Vetinari (revision 3.0.12-1-0-gd147bb5e7e)
[014515d8] main interface error: no suitable interface module
[013e2b58] main libvlc error: interface "globalhotkeys,none" initialization failed
[013e2b58] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[014515d8] skins2 interface error: cannot initialize OSFactory
[014515d8] [cli] lua interface: Listening on host "*console".
VLC media player 3.0.12 Vetinari
Command Line Interface initialized. Type `help' for help.
> Created new TCP socket 13 for connection
[61600a58] live555 demux error: Failed to connect with rtsp://192.168.1.59:554/cam/realmonitor?channel=1
[61602d00] satip stream error: Failed to setup RTSP session
[61800668] main input error: Your input can't be opened
[61800668] main input error: VLC is unable to open the MRL 'rtsp://admin:pwd@192.168.1.59:554/cam/realmonitor?channel=1'. Check the log for details.
```

SvenVD, 2022-01-09:
thanks  for reporting

seems duplicate see https://github.com/SvenVD/rpisurv/issues/141#issuecomment-981159241

closed 2022-01-09 by SvenVD