SvenVD/rpisurv #153: Wont stream RTSP Failed to connect to the D-Bus session

icepicknz, 2022-03-11:
Hi,

I just installed 2022-01-28-raspios-buster-armhf on a RPi4, followed the 
gpu_mem=512 to your /boot/config.txt
framebuffer_width=1920' and 'framebuffer_height=1080

The demo boots fine and plays the video of the mkv and says connecting in a few other boxes.

I commented out all the streams and added mine:

    - camera_streams:
        - url: "rtsp://192.168.44.4:7447/zR4CkwuhXKsMKu2a"
#        - url: "rtsp://192.168.44.4:7447/WEPB3Jn6yz8b8sEE"
#        - url: "rtsp://192.168.44.4:7447/v15hh7bd4qAfxZnv"
#        - url: "rtsp://192.168.44.4:7447/3sjnpi31GTGL0EwF"
#        - url: "rtsp://192.168.44.4:7447/2VOOkDfwzJpXvYBy"
#        - url: "rtsp://192.168.44.4:7447/oBlKwRyQFMywVRgC"
#        - url: "rtsp://192.168.44.4:7447/jHtj9GYW9F1sFBFH"
      rtsp_over_tcp:  true
      duration: 20
      nr_of_columns: 2

Trying single stream with RTSP over TCP true or false just shows connecting to stream on the screen.

/var/log/syslog shows the following.
```
Mar 11 01:00:48 office-left-tv rpisurv[464]: [01706090] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 11 01:00:48 office-left-tv rpisurv[464]: [0173f850] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 11 01:00:48 office-left-tv rpisurv[464]: [0173f850] main interface error: no suitable interface module
Mar 11 01:00:48 office-left-tv rpisurv[464]: [01688b58] main libvlc error: interface "dbus,none" initialization failed
Mar 11 01:00:48 office-left-tv rpisurv[464]: [0172a630] main interface error: no suitable interface module
Mar 11 01:00:48 office-left-tv rpisurv[464]: [01688b58] main libvlc error: interface "globalhotkeys,none" initialization failed
Mar 11 01:00:48 office-left-tv rpisurv[464]: [0172a630] dummy interface: using the dummy interface module...
Mar 11 01:00:48 office-left-tv rpisurv[464]: Created new TCP socket 4 for connection
Mar 11 01:00:48 office-left-tv rpisurv[464]: [736136b0] mmal_codec decoder: VCSM init succeeded: Legacy
Mar 11 01:00:49 office-left-tv rpisurv[464]: [736136b0] main decoder error: buffer deadlock prevented

```
I have tried changing GL Driver in Advanced settings to all 3, and gone back to legacy now as G3 wouldn't work at all
-G1 Legacy        Original non-GL desktop driver
-G2 GL (Fake KMS) OpenGL desktop driver with fake KMS
-G3 GL (Full KMS) OpenGL desktop driver with full KMS

icepicknz, 2022-03-11:
So, I thought I may try suggested versions from a while back, so installed a fresh 2021-05-07-raspios-buster-armhf

Same issue
```
Mar 11 03:33:26 raspberrypi systemd[1]: rpisurv.service: Succeeded.
Mar 11 03:33:26 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Mar 11 03:33:26 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Mar 11 03:33:27 raspberrypi rpisurv[1434]: pygame 1.9.4.post1
Mar 11 03:33:27 raspberrypi rpisurv[1434]: Hello from the pygame community. https://www.pygame.org/contribute.html
Mar 11 03:33:27 raspberrypi rpisurv[1434]: 2022/03/11 03:33:27 - l_default - INFO - Starting rpisurv 3.0.0
Mar 11 03:33:27 raspberrypi rpisurv[1434]: 2022/03/11 03:33:27 - l_default - INFO - Auto detected display SAM-SAMSUNG at HDMI 0 with display number 2 1920 x 1080
Mar 11 03:33:27 raspberrypi rpisurv[1434]: 2022/03/11 03:33:27 - l_default - INFO - Unique id of this installation is 8e255237f343b3c7ce914e08ce22ece49873b696ac5ac4f2766a3be1406cff77
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [01666630] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [01694300] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [01694300] main interface error: no suitable interface module
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [015edb58] main libvlc error: interface "dbus,none" initialization failed
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [01690040] main interface error: no suitable interface module
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [015edb58] main libvlc error: interface "globalhotkeys,none" initialization failed
Mar 11 03:33:31 raspberrypi rpisurv[1434]: [01690040] dummy interface: using the dummy interface module...
Mar 11 03:33:31 raspberrypi rpisurv[1434]: Created new TCP socket 3 for connection
Mar 11 03:33:32 raspberrypi rpisurv[1434]: 2022/03/11 03:33:32 - l_default - INFO - Sending stats is disabled, not sending stats
Mar 11 03:33:32 raspberrypi rpisurv[1434]: [75114958] mmal_codec decoder: VCSM init succeeded: Legacy
Mar 11 03:33:32 raspberrypi rpisurv[1434]: [75114958] main decoder error: buffer deadlock prevented
Mar 11 03:33:37 raspberrypi rpisurv[1434]: [75114958] main decoder error: buffer deadlock prevented
```

VLC is at least trying to run. I can stream rtsp://192.168.44.4:7447/zR4CkwuhXKsMKu2a just fine from my MAC VLC
```
root@raspberrypi:~# ps -auxwww | grep vlc
root      1447  2.0  3.3 124732 16936 ?        Ssl  03:33   0:02 /usr/bin/vlc -I dummy --aspect-ratio=1920:1080 --vout mmal_vout --network-caching 500 --no-video-title-show --mmal-display=hdmi-1 --input-timeshift-granularity=0 --repeat --mmal-vout-transparent --mmal-vout-window 1920x1080+0+0 --mmal-layer 2000000000 --no-audio rtsp://192.168.44.4:7447/zR4CkwuhXKsMKu2a

```

icepicknz, 2022-03-11:
Ok this is interesting, I decided to try stream same quality camera off my local home NVR and it works, then tried medium and high quality, they all worked. So its only the remote stuff thats not working.

I suspect there is some buffer setting that needs to be used for remote RTSP??

SvenVD, 2022-03-14:
You could try playing with network_caching_ms config option

sebburon, 2023-01-16:
I am facing the same issue. I also tried the network_caching_ms without any luck.
Any solutions to that would be appreciated.

Charlsburg, 2023-10-28:
> Ok this is interesting, I decided to try stream same quality camera off my local home NVR and it works, then tried medium and high quality, they all worked. So its only the remote stuff thats not working.
> 
> I suspect there is some buffer setting that needs to be used for remote RTSP??

I've been battling this issue also, and found I was able to pull a PC RTSP stream from my PC on my home network, but as soon as I plug it into the network on site (which is far less complex, same subnet, no firewalls), I get this same buffer deadlock error.

Really is doing my head in and I've tried so many different iterations but nothing seems to stop triggering the deadlock.  Did you manage to find any solutions yet?

dlasher, 2024-02-06:
Trying to convert from v2 to v3, and seeing this often on the pi3/bullseye/vlc install I've done. video streams all reset, and logs show
[code]
Feb 06 04:18:48 pi-camera-den rpisurv[655]: [73a39bf0] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:18:49 pi-camera-den rpisurv[655]: error: XDG_RUNTIME_DIR not set in the environment.
Feb 06 04:18:49 pi-camera-den rpisurv[655]: error: XDG_RUNTIME_DIR not set in the environment.
Feb 06 04:19:57 pi-camera-den rpisurv[655]: Created new TCP socket 4 for connection
Feb 06 04:19:57 pi-camera-den rpisurv[655]: [73a3a1a8] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:19:57 pi-camera-den rpisurv[657]: Created new TCP socket 6 for connection
Feb 06 04:19:57 pi-camera-den rpisurv[657]: [7390d378] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:19:57 pi-camera-den rpisurv[650]: Created new TCP socket 6 for connection
Feb 06 04:19:57 pi-camera-den rpisurv[650]: [72c0a8b8] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:19:57 pi-camera-den rpisurv[648]: Created new TCP socket 6 for connection
Feb 06 04:19:57 pi-camera-den rpisurv[648]: [73a0d378] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:19:57 pi-camera-den rpisurv[655]: [73a3a1a8] main decoder error: buffer deadlock prevented
Feb 06 04:19:57 pi-camera-den rpisurv[650]: [72c0a8b8] mmal_codec decoder: VCSM init succeeded: CMA
Feb 06 04:19:57 pi-camera-den rpisurv[648]: [73a0d378] main decoder error: buffer deadlock prevented
Feb 06 04:19:57 pi-camera-den rpisurv[657]: [7390d378] main decoder error: buffer deadlock prevented
Feb 06 04:19:57 pi-camera-den rpisurv[650]: [72c0a8b8] main decoder error: buffer deadlock prevented
Feb 06 04:19:58 pi-camera-den rpisurv[657]: [7390d378] mmal_codec decoder: VCSM init succeeded: CMA
[/code]

For now I'm going back to my buster/2.0 install. the video stays on screen for hours with zero blips. :(

Charlsburg, 2024-02-06:
I wonder if its as simple as needing to manually set the environment within the script, given its running in headless mode it doesnt have a desktop environment running.
Something like the following only you should setup your user ID (i.e. pi:pi)

`import os

if 'XDG_RUNTIME_DIR' not in os.environ:
    user_id = os.getuid()  # Get the current user's ID
    os.environ['XDG_RUNTIME_DIR'] = f'/run/user/{user_id}'

    # Print the set directory to confirm
    print(f"XDG_RUNTIME_DIR set to: {os.environ['XDG_RUNTIME_DIR']}")`

dlasher, 2024-02-07:
Per camera buffering seems to make it better, but having to run a 3000-5000ms "buffer" on each camera dramatically reduces the "realtime" effect, especially when it's things like a front door. They can be walking away before you hear the doorbell. 

v2/omxplayer - no such issues. Going back for now.

SvenVD, 2024-02-10:
I did not do any tests with sdl, not sure if it used hardware acceleration or that it can be use to provide specific coordinates