SvenVD/rpisurv #154: Screen Tearing on high resolution rtsp feeds

jared-bloomer, 2022-04-08:
I am using Reolink RLC-410-5MP cameras. The low resolution feeds work fine, but if I try to use High resolution feeds the image tearing is unviewable. 

Looking in `/var/log/syslog` I see the following when it tries to play a low resolution stream (rtsp://username:password@192.168.1.202:554/h264Preview_01_sub)

```
Apr  8 10:53:59 displaycameras rpisurv[5188]: 2022/04/08 10:53:59 - core.worker - INFO - This stream screen_manager_0_screen6_cam_stream1 has been stopped
Apr  8 10:53:59 displaycameras systemd[1]: Starting Cleanup of Temporary Directories...
Apr  8 10:53:59 displaycameras systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Apr  8 10:53:59 displaycameras systemd[1]: Started Cleanup of Temporary Directories.
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00be66f8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00be62f8] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00be62f8] main interface error: no suitable interface module
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00b65b58] main libvlc error: interface "dbus,none" initialization failed
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00bfce38] main interface error: no suitable interface module
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00b65b58] main libvlc error: interface "globalhotkeys,none" initialization failed
Apr  8 10:53:59 displaycameras rpisurv[5188]: [00bfce38] dummy interface: using the dummy interface module...
Apr  8 10:53:59 displaycameras rpisurv[5188]: Created new TCP socket 3 for connection
Apr  8 10:53:59 displaycameras rpisurv[5188]: [b360e4a8] mmal_codec decoder: VCSM init succeeded: CMA
Apr  8 10:54:01 displaycameras rpisurv[5188]: MultiFramedRTPSource::doGetNextFrame1(): The total received frame size exceeds the client's buffer size (250000).  680301 bytes of trailing data will be dropped!
Apr  8 10:54:01 displaycameras rpisurv[5188]: [b360e4a8] main decoder error: buffer deadlock prevented
Apr  8 10:54:01 displaycameras CRON[8365]: (root) CMD (/usr/bin/displaycameras repair)
Apr  8 10:54:03 displaycameras rpisurv[5188]: MultiFramedRTPSource::doGetNextFrame1(): The total received frame size exceeds the client's buffer size (500000).  148121 bytes of trailing data will be dropped!
Apr  8 10:54:05 displaycameras rpisurv[5188]: [b360e4a8] mmal_codec decoder: VCSM init succeeded: CMA
Apr  8 10:54:06 displaycameras rpisurv[5188]: error: XDG_RUNTIME_DIR not set in the environment.
Apr  8 10:54:06 displaycameras rpisurv[5188]: error: XDG_RUNTIME_DIR not set in the environment.
```

However, when attempting to play a high resolution stream (rtsp://username:password@192.168.1.203:554/h264Preview_01_main) I see

```
Apr  8 10:54:19 displaycameras rpisurv[5188]: 2022/04/08 10:54:19 - core.worker - INFO - This stream screen_manager_0_screen1_cam_stream1 has been stopped
Apr  8 10:54:19 displaycameras rpisurv[5188]: [010ce400] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Apr  8 10:54:19 displaycameras rpisurv[5188]: [010d5130] dbus interface error: Failed to connect to the D-Bus session daemon: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Apr  8 10:54:19 displaycameras rpisurv[5188]: [010d5130] main interface error: no suitable interface module
Apr  8 10:54:19 displaycameras rpisurv[5188]: [01050b58] main libvlc error: interface "dbus,none" initialization failed
Apr  8 10:54:19 displaycameras rpisurv[5188]: [010e7d90] main interface error: no suitable interface module
Apr  8 10:54:19 displaycameras rpisurv[5188]: [01050b58] main libvlc error: interface "globalhotkeys,none" initialization failed
Apr  8 10:54:19 displaycameras rpisurv[5188]: [010e7d90] dummy interface: using the dummy interface module...
Apr  8 10:54:19 displaycameras rpisurv[5188]: Created new TCP socket 3 for connection
Apr  8 10:54:19 displaycameras rpisurv[5188]: [b350e438] mmal_codec decoder: VCSM init succeeded: CMA
Apr  8 10:54:22 displaycameras rpisurv[5188]: [b350e438] mmal_codec decoder: VCSM init succeeded: CMA
Apr  8 10:54:23 displaycameras rpisurv[5188]: [b350e438] main decoder error: buffer deadlock prevented
Apr  8 10:54:29 displaycameras rpisurv[5188]: error: XDG_RUNTIME_DIR not set in the environment.
Apr  8 10:54:29 displaycameras rpisurv[5188]: error: XDG_RUNTIME_DIR not set in the environment.
```

Any idea if there is currently a fix for this? Am I just missing something?

SvenVD, 2022-04-08:
Try increasing https://github.com/SvenVD/rpisurv/blob/b60cb27f3b32af48f21fbca4e5292ed2f874038e/surveillance/conf/display1.yml#L98
Or try enabling
https://github.com/SvenVD/rpisurv/blob/b60cb27f3b32af48f21fbca4e5292ed2f874038e/surveillance/conf/display1.yml#L95

jared-bloomer, 2022-04-08:
So enabling `rstp_over_tcp` did not work. Already tried that. Doing network caching of 500ms didn't help either. Guess I could try a higher value. I have not tried `--rtsp-frame-buffer-size=500000` yet, but I will definitely try and comment the results when I get a chance. Thanks for the guidance.

jared-bloomer, 2022-04-08:
No joy.

This is what the stream looks like on a rpi4
[image]

jared-bloomer, 2022-04-08:
This is the config for that camera

```
    - camera_streams:
        - url: "rtsp://username:password@192.168.1.203:554/h264Preview_01_main" #FrontYard
      network_caching_ms: 800
      rtsp_over_tcp: true
      freeform_advanced_vlc_options: "--rtsp-frame-buffer-size=500000"
```

SvenVD, 2022-04-08:
Then I am out of options, I suggest to contact the VLC developers

ChrisH861, 2022-04-08:
I've seen occasionally seen similar on some 4&8MP cameras, but never worked out exactly what causes it.
Are you using a 16:9 resolution on the camera? (e.g 2560x1440), also maybe try a higher network cache.

jared-bloomer, 2022-04-08:
Yes I am using 16:9 ratio at a resolution of 2304x1296 @10fps with a max bitrate of 6144kbps

jared-bloomer, 2022-06-22:
None on my end. It'd still borked

Sent from my Verizon, Samsung Galaxy smartphone

JGVB, 2022-07-02:
I have the same problem. I need to use the lower resolution rtsp stream offered by unifi