SvenVD/rpisurv #158: Raspberry PI OS Lite (Legacy) Buster release date 2022-04-04 VLC errors

MelvinRC, 2022-06-21:
Fresh installed Raspberry PI OS Lite (Legacy) Buster release date 2022-04-04 via Raspberry Pi Imager v1.7.2 :

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Steps after logging in:

	1. sudo apt-get install git
	2. git clone https://github.com/SvenVD/rpisurv 
	3. cd rpisurv
	4. git checkout v3_latest
	5. sudo ./install.sh

And now the installation fails with the following error:

Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)
Aborting installation, upgrade to latest vlc player with mmal support

VLC info:

VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
VLC version 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Compiled by buildd on bm-wb-01 (Jun 18 2022 13:59:48)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public Licence;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.

Yesterday did exactly the same and work fine, now tried three times over and over and still no luck, what do I miss?

anazz20, 2022-06-22:
This is my issue as well it seems. https://www.tapatalk.com/groups/rpisurv/fine-for-months-vlc-update-yesterday-t191.html
Upgraded VLC now stuck with 8 screens just saying connecting.

Kuuchuu, 2022-06-22:
@anazz20 @MelvinRC 
I was able to fix this issue for myself with the following steps:

First comment out the abort in the install.sh script so it installs
Change:
```
if ! is_vlc_mmal_present;then
    echo "Your version of vlc does not have the needed mmal options. Rpisurv needs those"
    echo "Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)"
    echo "Aborting installation, upgrade to latest vlc player with mmal support"
    exit 2
fi
```
To:
```
if ! is_vlc_mmal_present;then
    echo "Your version of vlc does not have the needed mmal options. Rpisurv needs those"
    echo "Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72)"
#    echo "Aborting installation, upgrade to latest vlc player with mmal support"
#    exit 2
fi

```
Run the installer,
Then stop rpisurv
```
sudo systemctl stop rpisurv.service
```
Then remove vlc, ensure dependencies are removed, and install an older version of vlc and its dependencies
```
sudo apt remove vlc; sudo apt autoremove; sudo apt install xdg-utils vlc-data=3.0.12\* libvlc-bin=3.0.12\* vlc-l10n=3.0.12\* vlc-plugin-notify=3.0.12\* vlc-plugin-samba=3.0.12\* vlc-plugin-skins2=3.0.12\* vlc-plugin-video-splitter=3.0.12\* vlc-plugin-visualization=3.0.12\* vlc-bin=3.0.12\* vlc-plugin-base=3.0.12\* vlc-plugin-qt=3.0.12\* vlc-plugin-video-output=3.0.12\* vlc=3.0.12\*
```
Finally, reboot (or I guess just start the rpisurv.service)

Kuuchuu, 2022-06-22:
Alternatively, you could just edit the install.sh script removing vlc from the apt-get install line, then add the line above that installs the older, working version of vlc, but I didn't do it this way so I'm unsure if you'd run into any issues.
Change this:
```
sudo apt-get install vlc rsync sed coreutils fbset ffmpeg openssl procps python3-pygame python3-yaml python3-openssl python3 libraspberrypi-bin -y
```
To this:
```
sudo apt install xdg-utils vlc-data=3.0.12\* libvlc-bin=3.0.12\* vlc-l10n=3.0.12\* vlc-plugin-notify=3.0.12\* vlc-plugin-samba=3.0.12\* vlc-plugin-skins2=3.0.12\* vlc-plugin-video-splitter=3.0.12\* vlc-plugin-visualization=3.0.12\* vlc-bin=3.0.12\* vlc-plugin-base=3.0.12\* vlc-plugin-qt=3.0.12\* vlc-plugin-video-output=3.0.12\* vlc=3.0.12\* -y
sudo apt-get install rsync sed coreutils fbset ffmpeg openssl procps python3-pygame python3-yaml python3-openssl python3 libraspberrypi-bin -y
```

SvenVD, 2022-06-22:
> 

Thanks for the pull request, I merged it

closed 2022-06-22 by SvenVD

anazz20, 2022-06-23:
Back up and running. Thank you Kuuchuu!

MelvinRC, 2022-06-23:
Thanks to both works for me as well, tried a new install without any issues :-)