SvenVD/rpisurv #161: Fix for #158 no longer works

Kuuchuu, 2022-07-09:
Issue #158 is back. v3.0.12 is no longer in the repos.

sebguilbaud, 2022-07-15:
3.0.12 VLC debs are still available here : http://archive.raspbian.com/raspbian/pool/main/v/vlc/vlc_3.0.12-0%2Bdeb9u1_armhf.deb

Kuuchuu, 2022-07-21:
Just pushed a fix to [my own repo](https://github.com/Kuuchuu/rpisurv), but I still need to test it. I'll do so in a few hours after a bit of sleep. If it works I'll submit a PR

TheKoola, 2022-07-21:
FWIW, the RPI-Distro build of VLC for Bullseye has been [updated](https://github.com/RPi-Distro/vlc/issues/66) to include MMAL.  Version 3.0.17.4-0+deb10u1+rpt1 appears to be working:
```
$ sudo apt list --installed vlc*
Listing... Done
vlc-bin/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-data/oldstable,now 3.0.17.4-0+deb10u1+rpt1 all [installed]
vlc-l10n/oldstable,now 3.0.17.4-0+deb10u1+rpt1 all [installed]
vlc-plugin-base/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-notify/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-qt/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-samba/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-skins2/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-video-output/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-video-splitter/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc-plugin-visualization/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
vlc/oldstable,now 3.0.17.4-0+deb10u1+rpt1 armhf [installed]
```
```
$ vlc --version
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
VLC version 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Compiled by serge on arm-build.pitowers.org (Jun 22 2022 10:02:16)
Compiler: gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```
```
$ /usr/bin/vlc -H | grep -- '--mmal-layer'
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
      --mmal-layer <integer>     VideoCore layer where the video is displayed.
```
This version is working fine for me.  Are there other compelling reasons to go back to vlc 3.0.12?

Kuuchuu, 2022-07-21:
@TheKoola Thanks for letting me know, I was under the impression they stopped supporting mmal all together.

I just submitted 2 pull requests to change vlc back to the latest version in the repo, one for master and one for the v3 branch

SvenVD, 2022-07-25:
> @TheKoola Thanks for letting me know, I was under the impression they stopped supporting mmal all together.
> 
> I just submitted 2 pull requests to change vlc back to the latest version in the repo, one for master and one for the v3 branch

merged

closed 2022-07-25 by SvenVD