SvenVD/rpisurv #165: problen after start

javierc28, 2022-08-07:
hi im new on this i follow the steps but i can make it work i have a raspberry pi 4 with a 7' touch screem, thanks

pi@raspberrypi:~ $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2022-08-07 16:14:22 EDT; 32s ago
  Process: 705 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 705 (code=exited, status=1/FAILURE)

SvenVD, 2022-08-10:
what is in the logs? /var/log/daemon.log?

javierc28, 2022-08-15:
Aug 15 17:45:32 raspberrypi rpisurv[1093]:   File "surveillance.py", line 188, in <module>
Aug 15 17:45:32 raspberrypi rpisurv[1093]:     screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
Aug 15 17:45:32 raspberrypi rpisurv[1093]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 20, in __init__
Aug 15 17:45:32 raspberrypi rpisurv[1093]:     self._init_drawinstance()
Aug 15 17:45:32 raspberrypi rpisurv[1093]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 214, in _init_drawinstance
Aug 15 17:45:32 raspberrypi rpisurv[1093]:     self.drawinstance = Draw([int(self.display["resolution"]["width"]), int(self.display["resolution"]["height"])], self.disable_pygame, self.name)
Aug 15 17:45:32 raspberrypi rpisurv[1093]:   File "/usr/local/bin/rpisurv/core/util/draw.py", line 52, in __init__
Aug 15 17:45:32 raspberrypi rpisurv[1093]:     self.surface = pygame.display.set_mode((self.resolution_width, self.resolution_height))
Aug 15 17:45:32 raspberrypi rpisurv[1093]: pygame.error: No video mode large enough for 1920x1080
Aug 15 17:45:32 raspberrypi systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Aug 15 17:45:32 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Aug 15 17:45:44 raspberrypi systemd[1]: Started Session 4 of user pi.
Aug 15 17:45:52 raspberrypi systemd[1]: rpisurv.service: Service RestartSec=20s expired, scheduling restart.
Aug 15 17:45:52 raspberrypi systemd[1]: rpisurv.service: Scheduled restart job, restart counter is at 24.
Aug 15 17:45:52 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Aug 15 17:45:52 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 15 17:45:53 raspberrypi rpisurv[1120]: pygame 1.9.4.post1
Aug 15 17:45:53 raspberrypi rpisurv[1120]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug 15 17:45:53 raspberrypi rpisurv[1120]: 2022/08/15 17:45:53 - l_default - INFO - Starting rpisurv 3.0.0
Aug 15 17:45:54 raspberrypi rpisurv[1120]: 2022/08/15 17:45:54 - l_default - ERROR - Could not autodetect displays, use values from fallback config
Aug 15 17:45:54 raspberrypi rpisurv[1120]: 2022/08/15 17:45:54 - l_default - ERROR - Using config display fallback_device_0 at HDMI 0 with display number 2 1920 x 1080
Aug 15 17:45:54 raspberrypi rpisurv[1120]: 2022/08/15 17:45:54 - l_default - INFO - Unique id of this installation is 7a3a2da9ee3174303b59480b47b8b605f7a28a8107da0a9a280a45b68c463e94
Aug 15 17:45:55 raspberrypi rpisurv[1120]: Traceback (most recent call last):
Aug 15 17:45:55 raspberrypi rpisurv[1120]:   File "surveillance.py", line 188, in <module>
Aug 15 17:45:55 raspberrypi rpisurv[1120]:     screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
Aug 15 17:45:55 raspberrypi rpisurv[1120]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 20, in __init__
Aug 15 17:45:55 raspberrypi rpisurv[1120]:     self._init_drawinstance()
Aug 15 17:45:55 raspberrypi rpisurv[1120]:   File "/usr/local/bin/rpisurv/core/ScreenManager.py", line 214, in _init_drawinstance
Aug 15 17:45:55 raspberrypi rpisurv[1120]:     self.drawinstance = Draw([int(self.display["resolution"]["width"]), int(self.display["resolution"]["height"])], self.disable_pygame, self.name)
Aug 15 17:45:55 raspberrypi rpisurv[1120]:   File "/usr/local/bin/rpisurv/core/util/draw.py", line 52, in __init__
Aug 15 17:45:55 raspberrypi rpisurv[1120]:     self.surface = pygame.display.set_mode((self.resolution_width, self.resolution_height))
Aug 15 17:45:55 raspberrypi rpisurv[1120]: pygame.error: No video mode large enough for 1920x1080
Aug 15 17:45:55 raspberrypi systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Aug 15 17:45:55 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Aug 15 17:46:15 raspberrypi systemd[1]: rpisurv.service: Service RestartSec=20s expired, scheduling restart.
Aug 15 17:46:15 raspberrypi systemd[1]: rpisurv.service: Scheduled restart job, restart counter is at 25.
Aug 15 17:46:15 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Aug 15 17:46:15 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 15 17:46:17 raspberrypi rpisurv[1141]: pygame 1.9.4.post1
Aug 15 17:46:17 raspberrypi rpisurv[1141]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug 15 17:46:17 raspberrypi rpisurv[1141]: 2022/08/15 17:46:17 - l_default - INFO - Starting rpisurv 3.0.0
Aug 15 17:46:17 raspberrypi rpisurv[1141]: 2022/08/15 17:46:17 - l_default - ERROR - Could not autodetect displays, use values from fallback config
Aug 15 17:46:17 raspberrypi rpisurv[1141]: 2022/08/15 17:46:17 - l_default - ERROR - Using config display fallback_device_0 at HDMI 0 with display number 2 1920 x 1080
Aug 15 17:46:17 raspberrypi rpisurv[1141]: 2022/08/15 17:46:17 - l_default - INFO - Unique id of this installation is 7a3a2da9ee3174303b59480b47b8b605f7a28a8107da0a9a280a45b68c463e94
Aug 15 17:46:18 raspberrypi rpisurv[1141]: Traceback (most recent call last):
Aug 15 17:46:18 raspberrypi rpisurv[1141]:   File "surveillance.py", line 188, in <module>
Aug 15 17:46:18 raspberrypi rpisurv[1141]:     screen_manager=ScreenManager(f'screen_manager_{count}', display, enable_opportunistic_caching_next_screen, disable_pygame)
[... cut ...]
Aug 15 17:46:41 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Aug 15 17:47:01 raspberrypi systemd[1]: rpisurv.service: Service RestartSec=20s expired, scheduling restart.
Aug 15 17:47:01 raspberrypi systemd[1]: rpisurv.service: Scheduled restart job, restart counter is at 27.
Aug 15 17:47:01 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Aug 15 17:47:01 raspberrypi systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 15 17:47:03 raspberrypi rpisurv[1162]: pygame 1.9.4.post1

javierc28, 2022-08-15:
hi this is the cinfig file screen on 800x400, thanks

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
framebuffer_width=800
framebuffer_height=400

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

gpu_mem=512
lcd_rotate=2
lcd_rotate=2

SvenVD, 2022-08-27:
Try manually configuring 800x400 in 
https://github.com/SvenVD/rpisurv/blob/master/surveillance/conf/general.yml

javierc28, 2022-08-28:
you say from line 29? and where i have to copy that? thanks

SvenVD, 2022-08-28:
> you say from line 29? and where i have to copy that? thanks

check /etc/rpisurv/general.yml

javierc28, 2022-08-31:
hi, yes works, the only thing is after like 10min freeze, i set memory to 512, what alse i can do? thanks

SvenVD, 2022-09-03:
Something in /usr/local/bin/rpisurv/logs/main.log?

javierc28, 2022-09-03:
[image]

SvenVD, 2022-09-03:
your gpu is too busy downscaling your incoming rstp streams to 800x400 use lower resolution rtsp streams

javierc28, 2022-09-04:
Thanks I will try to do it, if I use one can per screen that can be better?