SvenVD/rpisurv #166: Fresh Install Service issue?

GremlinCCS, 2022-08-13:
Good Day

I just installed a fresh rpisurv on my pi loaded with Pi OS Legacy

When I boot up my Pi it loads into a black screen and stays on the black screen. If I ssh into the Pi and "systemctl restart rpisurv" the camera image do come on for a while and then it goes back to the connecting screen. restarting the service again starts the rtsp stream again for a while and goes back to the Connecting screen.

The service are active when booted, "systemctl status rpisurv" produceses

 rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2022-08-13 17:00:29 SAST; 5s ago
 Main PID: 1491 (rpisurv)
    Tasks: 4 (limit: 1023)
   CGroup: /system.slice/rpisurv.service
           ├─1491 /bin/bash /usr/bin/rpisurv
           └─1492 python3 surveillance.py

Aug 13 17:00:29 JWCMonitor systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 13 17:00:30 JWCMonitor rpisurv[1491]: pygame 1.9.4.post1
Aug 13 17:00:30 JWCMonitor rpisurv[1491]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug 13 17:00:30 JWCMonitor rpisurv[1491]: 2022/08/13 17:00:30 - l_default - INFO - Starting rpisurv 3.0.0
Aug 13 17:00:31 JWCMonitor rpisurv[1491]: 2022/08/13 17:00:31 - l_default - INFO - Auto detected display GDH-CHHWJT a
Aug 13 17:00:31 JWCMonitor rpisurv[1491]: 2022/08/13 17:00:31 - l_default - INFO - Unique id of this installation is

restart and it works for a while

when it goes to the connecting screen the status shows the same.

I have redone the install 3 times hopping something just messed up but with no luck getting the same problems.

/boot/config.txt have gou_mem=512
/etc/rpisurv/display1.yml
    - camera_streams:
        - url: "rtsp://192.168.100.123/live2.sdp"
          rtsp_over_tcp:  false

I have tried TCP true but to the same en. tried network_caching_ms: 500, probe_timeout: 3 creates a black screen only until I remove it.

omxplayer --video_fifo 1 --video_queue 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 rtsp://192.168.100.123/live2.sdp works 100% with no problems

/var/log/daemon.log fresh after a startup, service restart, connecting screen and an restart again after the startup sequence.

Aug 13 17:29:33 JWCMonitor systemd[1]: Started User Manager for UID 1000.
Aug 13 17:29:33 JWCMonitor systemd[1]: Started Session 1 of user pi.
Aug 13 17:29:34 JWCMonitor rpisurv[478]: /usr/bin/rpisurv: line 3:   483 Segmentation fault      python3 surveillanc$Aug 13 17:29:34 JWCMonitor systemd[1]: rpisurv.service: Main process exited, code=exited, status=139/n/a
Aug 13 17:29:34 JWCMonitor systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Aug 13 17:29:35 JWCMonitor dhcpcd[461]: eth0: probing address 192.168.100.109/24
Aug 13 17:29:37 JWCMonitor systemd[1]: systemd-rfkill.service: Succeeded.
Aug 13 17:29:37 JWCMonitor bthelper[540]: Changing power off succeeded
Aug 13 17:29:38 JWCMonitor bthelper[540]: Changing power on succeeded
Aug 13 17:29:41 JWCMonitor dhcpcd[461]: eth0: leased 192.168.100.109 for 86400 seconds
Aug 13 17:29:41 JWCMonitor dhcpcd[461]: eth0: adding route to 192.168.100.0/24
Aug 13 17:29:41 JWCMonitor dhcpcd[461]: eth0: adding default route via 192.168.100.2
Aug 13 17:29:41 JWCMonitor avahi-daemon[401]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.1$Aug 13 17:29:41 JWCMonitor avahi-daemon[401]: New relevant interface eth0.IPv4 for mDNS.
Aug 13 17:29:41 JWCMonitor avahi-daemon[401]: Registering new address record for 192.168.100.109 on eth0.IPv4.
Aug 13 17:29:45 JWCMonitor dhcpcd[461]: eth0: no IPv6 Routers available
Aug 13 17:29:54 JWCMonitor systemd[1]: rpisurv.service: Service RestartSec=20s expired, scheduling restart.
Aug 13 17:29:54 JWCMonitor systemd[1]: rpisurv.service: Scheduled restart job, restart counter is at 1.
Aug 13 17:29:54 JWCMonitor systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Aug 13 17:29:54 JWCMonitor systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Aug 13 17:29:55 JWCMonitor rpisurv[642]: pygame 1.9.4.post1
Aug 13 17:29:55 JWCMonitor rpisurv[642]: Hello from the pygame community. https://www.pygame.org/contribute.html
Aug 13 17:29:55 JWCMonitor rpisurv[642]: 2022/08/13 17:29:55 - l_default - INFO - Starting rpisurv 3.0.0
Aug 13 17:29:55 JWCMonitor rpisurv[642]: 2022/08/13 17:29:55 - l_default - INFO - Auto detected display GDH-CHHWJT a$Aug 13 17:29:55 JWCMonitor rpisurv[642]: 2022/08/13 17:29:55 - l_default - INFO - Unique id of this installation is $Aug 13 17:29:56 JWCMonitor systemd[1]: systemd-fsckd.service: Succeeded.
Aug 13 17:49:01 JWCMonitor systemd[1]: systemd-hostnamed.service: Succeeded.
Aug 13 17:49:36 JWCMonitor systemd[1]: Started Session 3 of user pi.
Aug 13 17:49:41 JWCMonitor systemd[1]: Stopping Rpisurv Raspberry Pi Surveillance...
Aug 13 17:49:41 JWCMonitor systemd[1]: rpisurv.service: Main process exited, code=killed, status=15/TERM
Aug 13 17:49:41 JWCMonitor rpisurv[642]: 2022/08/13 17:49:41 - l_default - INFO - Sending stats is disabled, not sen$Aug 13 17:49:41 JWCMonitor rpisurv[642]: [00d814a8] vlcpulse audio output error: PulseAudio server connection failur$Aug 13 17:49:41 JWCMonitor rpisurv[642]: [00dc0bd8] dbus interface error: Failed to connect to the D-Bus session dae$Aug 13 17:49:41 JWCMonitor rpisurv[642]: [00dc0bd8] main interface error: no suitable interface module
Aug 13 17:49:41 JWCMonitor rpisurv[642]: [00d1db58] main libvlc error: interface "dbus,none" initialization failed
[... 24 lines cut ...]
Aug 13 17:51:37 JWCMonitor rpisurv[1357]: [75119d40] mmal_codec decoder: VCSM init succeeded: Legacy
Aug 13 17:51:37 JWCMonitor rpisurv[1357]: [75119d40] mmal_codec decoder: VCSM init succeeded: Legacy
Aug 13 17:51:37 JWCMonitor rpisurv[1357]: error: XDG_RUNTIME_DIR not set in the environment.
Aug 13 17:51:37 JWCMonitor rpisurv[1357]: error: XDG_RUNTIME_DIR not set in the environment.

/usr/local/bin/rpisurv/logs/main.log

2022/08/13 17:29:32 - l_default - INFO - Starting rpisurv 3.0.0
2022/08/13 17:29:32 - l_default - INFO - Auto detected display GDH-CHHWJT at HDMI 0 with display number 2 1280 x 720
2022/08/13 17:29:32 - l_default - INFO - Unique id of this installation is 9cf36ce09763b8e01dfe4042dd5c54e6fdf821885547ece529f69061b09de12e
2022/08/13 17:29:55 - l_default - INFO - Starting rpisurv 3.0.0
2022/08/13 17:29:55 - l_default - INFO - Auto detected display GDH-CHHWJT at HDMI 0 with display number 2 1280 x 720
2022/08/13 17:29:55 - l_default - INFO - Unique id of this installation is f973cda2dc13eea9b00a2ba61b1b667d726d3e3e6e74e6737460660b1c212cde
2022/08/13 17:49:41 - l_default - INFO - Sending stats is disabled, not sending stats
2022/08/13 17:51:12 - l_default - INFO - Starting rpisurv 3.0.0
2022/08/13 17:51:12 - l_default - INFO - Auto detected display GDH-CHHWJT at HDMI 0 with display number 2 1280 x 720
2022/08/13 17:51:12 - l_default - INFO - Unique id of this installation is f973cda2dc13eea9b00a2ba61b1b667d726d3e3e6e74e6737460660b1c212cde
2022/08/13 17:51:37 - l_default - INFO - Sending stats is disabled, not sending stats
2022/08/13 17:53:08 - l_default - INFO - Starting rpisurv 3.0.0
2022/08/13 17:53:08 - l_default - INFO - Auto detected display GDH-CHHWJT at HDMI 0 with display number 2 1280 x 720
2022/08/13 17:53:08 - l_default - INFO - Unique id of this installation is f973cda2dc13eea9b00a2ba61b1b667d726d3e3e6e74e6737460660b1c212cde

Camera used are a Vivotek CC8370-HV using stream two set to 640x480 15fps 512kbps constant bit rate H.264

any ideas would be nice. I'm not sure what else to try to make it work.

SvenVD, 2022-09-15:
could you try running the stream with the cvlc command, It seems it can not handle your camera stream. Did you increase gpu_mem?

SvenVD, 2023-11-30:
stale issue

closed 2023-11-30 by SvenVD