SvenVD/rpisurv #169: No  stream shown

winkelu, 2022-10-03:
Hello 
I get the following error

pygame, used to display image, is not compatible with dispmanx layers caching mechanism

installed version 3 
on buster lite 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

/usr/include/python3.7m/pygame
/usr/lib/python3/dist-packages/pygame

Demo is ok

any thoughts?
thanks in advance

SvenVD, 2022-10-07:
more info is needed, as in what is the problem you are experiencing. The error by itself should not stop rpisurv. Are you using image streams? Multiple HDMI?

winkelu, 2022-10-08:
Even with only the Demo I get this error but demo is working
I have  only one Monitor attached
Same with buster full

SvenVD, 2022-10-09:
What is the problem you are experiencing?

closed 2023-11-30 by SvenVD