SvenVD/rpisurv #170: vlc memory leak

si458, 2022-10-31:
Hi All,
im not sure what to do about this but im having an issue where after its running for a few hours, its starts memory leaking
so much so its crashing my pi hard withint 24hours
its a Raspberry Pi 2 B Model,
i installed the raspbian buster full os version,
setup version 3 and it runs fine for a few hours then suddenly crashes,
i then activated snmp to watch the pi remotely, and i can see the memory usage just starts to climb after awhile and then every 5mins coming from the vlc processes?
the streams are from dahua DVR, sub stream (352x288), the display is a DVI monitor (used hdmi to dvi cable) resolution 1024x768
only 4 streams
any suggestions?

si458, 2022-11-02:
few extra logs from debug, might be helpful not sure
```
Nov  2 10:43:08 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:08 - l_default - DEBUG - free malloc gpu mem value is 9M
Nov  2 10:43:08 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:08 - l_default - DEBUG - total available malloc gpu mem value is 12M
Nov  2 10:43:08 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:08 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:9M outputmem_bytes:9437184.0
Nov  2 10:43:08 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:08 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:12M outputmem_bytes:12582912.0
Nov  2 10:43:09 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:09 - l_default - DEBUG - free reloc gpu mem value is 466M
Nov  2 10:43:09 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:09 - l_default - DEBUG - total available reloc gpu mem value is 492M
Nov  2 10:43:09 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:09 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:466M outputmem_bytes:488636416.0
Nov  2 10:43:09 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:09 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
Nov  2 10:43:09 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:09 - l_default - DEBUG - Screen: screen_manager_0_screen1 active_run_time: 18 / 30
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - free malloc gpu mem value is 9M
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - total available malloc gpu mem value is 12M
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:9M outputmem_bytes:9437184.0
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:12M outputmem_bytes:12582912.0
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - free reloc gpu mem value is 466M
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - total available reloc gpu mem value is 492M
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:466M outputmem_bytes:488636416.0
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
Nov  2 10:43:10 fthcctvpi rpisurv[7007]: 2022/11/02 10:43:10 - l_default - DEBUG - Screen: screen_manager_0_screen1 active_run_time: 19 / 30
```

SvenVD, 2022-11-02:
Describe "so much so its crashing my pi hard withint 24hours". What does happen? Kernel panic?

si458, 2022-11-02:
> Describe "so much so its crashing my pi hard withint 24hours". What does happen? Kernel panic?

It's causing the pi to lock up, and the video streams to freeze

It doesn't seem to kernel panic? 

Just the memory usage climbs then the swap starts to climb too til they are both at 100% then the pi stops responding, 

But i have noticed if I leave it alone for say 5 hours it does eventually kill the vlc process itself and sorta return to normal

Also the ssh is still accessible, but take about 10 mins to login remotely and then another 10 mins to type the command `systemctl restart rpisurv` and 10 more mins to run the command

The only fix I have found at the moment is to run a cron every 3 hours and restart rpisurv to avoid the memory leak

SvenVD, 2022-11-02:
normally oomkiller should kick in do you see something in the syslogs? Other then that what does top say before it freezes? Does this also occur with one stream?

si458, 2022-11-03:
@SvenVD im checking now for you, i will edit this post with logs as i find stuff
but i dont see anything weird and the oomkiller only seems to kick in after about 5 hours when the menmorys at 100% and swaps at 100% too

```
Nov  3 01:33:23 fthcctvpi rpisurv[21485]: [73800668] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 500 ms)
Nov  3 01:33:23 fthcctvpi rpisurv[21485]: [7350e998] main decoder error: Timestamp conversion failed for 1667439201485001: no reference clock
Nov  3 01:33:23 fthcctvpi rpisurv[21485]: [7350e998] main decoder error: Could not convert timestamp 0 for MMAL decoder
Nov  3 01:33:24 fthcctvpi rpisurv[21485]: [739006b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 500 ms)
Nov  3 01:33:24 fthcctvpi rpisurv[21485]: [7350e998] main decoder error: buffer deadlock prevented
Nov  3 01:33:24 fthcctvpi rpisurv[21485]: [7350ea88] main decoder error: buffer deadlock prevented
Nov  3 01:34:23 fthcctvpi rpisurv[21485]: [739006b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 513 ms)
Nov  3 01:34:23 fthcctvpi rpisurv[21485]: [7350ea88] main decoder error: Timestamp conversion failed for 1667439261854001: no reference clock
Nov  3 01:34:23 fthcctvpi rpisurv[21485]: [7350ea88] main decoder error: Could not convert timestamp 0 for MMAL decoder
Nov  3 01:34:23 fthcctvpi rpisurv[21485]: [73800668] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 507 ms)
Nov  3 01:34:24 fthcctvpi rpisurv[21485]: [7350ea88] main decoder error: buffer deadlock prevented
Nov  3 01:34:24 fthcctvpi rpisurv[21485]: [7350e998] main decoder error: buffer deadlock prevented
```

si458, 2022-11-03:
@SvenVD ok added logs for you, hope they help!
the only thing i can think of is after checking the logs is the AUDIO,
i did yesterday morning ater i restarted the pi about 10amish change the audio setting from HDMI to 3.5 because the HDMI has an adapter to DVI so the is no audio,
but i can see from the logs again, after the oomkiller happened, rpisurv restarted vlc with the 'no-audio' option?
```
Nov  2 07:16:24 fthcctvpi rpisurv[396]: 2022/11/02 07:16:23 - core.worker - DEBUG - Starting stream screen_manager_0_screen1_cam_stream1 with commandline ['/usr/bin/cvlc', '--aspect-ratio=512:384', '--vout', 'mmal_vout', '--network-caching', '500', '--no-video-title-show', '--mmal-display=hdmi-1', '--input-timeshift-granularity=0', '--repeat', '--mmal-vout-transparent', '--mmal-vout-window', '512x384+0+0', '--mmal-layer', '2000000000', '--no-audio', 'rtsp://USER:PASSWORD!@192.168.52.123:554/cam/realmonitor?channel=3&subtype=1']
Nov  2 07:16:24 fthcctvpi rpisurv[396]: 2022/11/02 07:16:24 - l_default - DEBUG - screen_manager_0: update_active_screen screen_manager_0_screen1
Nov  2 07:16:24 fthcctvpi rpisurv[396]: 2022/11/02 07:16:24 - l_default - DEBUG - Screen: Start polling connectivity for the camera_streams part of screen: screen_manager_0_screen1
```

si458, 2022-11-05:
@SvenVD right so it still ran outta memory again even with memory switched to the 3.5m jack not hdmi,
i also tried uninstall vlc 3.0.17 and installing an old version 3.0.12 from a repo i found but it was missing the mmla patch,
now i cant get vlc to install again with mmla patch from the raspbian repo,
so i need to format the card again when im next on site, groan...

SvenVD, 2022-11-05:
It seems to be a problem with you sdcard
https://github.com/raspberrypi/linux/issues/2810

Memory is complex, Linux always tries to use all available memory for caching. You should measure immediate freeable memory which is your true available memory.

si458, 2022-11-05:
> It seems to be a problem with you sdcard
> https://github.com/raspberrypi/linux/issues/2810
> 
> Memory is complex, Linux always tries to use all available memory for caching so it will fill over time. You should measure immediate freeable memory which is your true available memory.

Hmmm OK I do think the micro sd card is about 4 years old so maybe a new one I can try?

SvenVD, 2022-11-05:
I would try a another sdcard to see if that solves the issue

si458, 2022-11-06:
> I would try a another sdcard to see if that solves the issue

ive bought a class 10 sd card brand new (all i could get my hand on this minute) so driving to the site today to swap it out, which OS should i use? buster or bullseye? i did have bullseye working no problem after changing it to the fake graphics but i no your support buster only?

si458, 2022-11-06:
ok fresh install on a brand new class 10 micro sd card and im seeing this error?
`Nov  6 12:33:01 fthcctvpi rpisurv[371]: 2022/11/06 12:33:01 - l_default - ERROR - Skipping calculating free malloc memory because of TimeoutExpired(['/usr/bin/vcgencmd', 'get_mem', 'malloc'], 1)`

si458, 2022-11-07:
@SvenVD right ok so after 12 hours, the memory started climbing again!
but weirdly enough its only a single VLC instance that has this memory leak? (im running 4 cameras)
so ive changed the camera thats seems to be causing this memory leak to a different one to see if it happens again?
(the camera that has this issue does indeed have a feed issue to the DVR unit itself, it flickers on and off and wavey lines, i think the BNC cable is broke, the the RTRSP stream is to the DVR itself not the camera as the camera is BNC based)

SvenVD, 2022-11-07:
> 

I only tested things on buster and buster is still supported by raspberry pi

SvenVD, 2022-11-07:
> ok fresh install on a brand new class 10 micro sd card and im seeing this error? `Nov 6 12:33:01 fthcctvpi rpisurv[371]: 2022/11/06 12:33:01 - l_default - ERROR - Skipping calculating free malloc memory because of TimeoutExpired(['/usr/bin/vcgencmd', 'get_mem', 'malloc'], 1)`

This seems to mean you gpu is overloaded for some reason

SvenVD, 2022-11-07:
> @SvenVD right ok so after 12 hours, the memory started climbing again! but weirdly enough its only a single VLC instance that has this memory leak? (im running 4 cameras) so ive changed the rtsp stream thats seems to be causing this memory leak to a different camera to see if it happens again? (the camera that has this issue does indeed have a feed issue to the DVR unit itself, it flickers on and off and wavey lines, i think the BNC cable is broke, but the RTRSP stream is to the DVR itself and not the camera as the camera is BNC based)

Does it still crash with the new sdcard?

si458, 2022-11-07:
> Does it still crash with the new sdcard

yes it does and i have the GPU set to 512mb

SvenVD, 2022-11-07:
Do you still get the events_freezable mmc_rescan and mmc errors? Other than that

the streams are from dahua DVR, sub stream (352x288), the display is a DVI monitor (used hdmi to dvi cable) resolution 1024x768

Could you try to make each stream 512x384, so that the gpu does not need to scale

si458, 2022-11-07:
> Do you still get the events_freezable mmc_rescan and mmc errors? Other than that
> 
> the streams are from dahua DVR, sub stream (352x288), the display is a DVI monitor (used hdmi to dvi cable) resolution 1024x768
> 
> Could you try to make each stream 512x384, so that the gpu does not need to scale

i will try changing the stream quality tomorrow (if it crashes again) as i have changed the camera from one camera to another to see if its maybe that perticular camera thats causing the issue as every time i check is appears to only a single VLC instance thats causing the memory leak and not all 4 of them?

also my monitor is actually 1280x1024
`2022/11/07 11:49:37 - l_default - INFO - Auto detected display ENC-L461 at HDMI 0 with display number 2 1280 x 1024`

SvenVD, 2022-11-07:
640x512 then

si458, 2022-11-09:
@SvenVD, ok so it lasted longer this time (about 24 hours instead of 12hours)
i removed the camera i throught was causing it to crash BUT instead it was a different camera that crashed it this time
BUT this time the OOM kicked in and killed the VLC and it restarted as expected
(so the OOM problem of crashing the PI must of been a bad memory card)

i cant change the quality of the resolution on the stream from the DVR unit sadly (its a BNC unit not IP based)
i only have D1: 720 X 480, CIF: 352 X 240 or 960H 976 X 582 to pick from sadly
the monitor is locked to 1280x1024 or lower (like 800x600)

could i set the GPU memory to 640 or 768?
```
pi@fthcctvpi:/var/log $ free -mh
              total        used        free      shared  buff/cache   available
Mem:          491Mi       115Mi       238Mi       5.0Mi       137Mi       320Mi
Swap:          99Mi        29Mi        70Mi
```

SvenVD, 2022-11-09:
maybe set the GPU mem lower so the OS processes have more memory left?

si458, 2022-11-10:
> maybe set the GPU mem lower so the OS processes have more memory left?

so i lowered the gpu memory but its still the same issue, 1 camera crashed and oom killed it then rpisurv detected the rtsp wasnt responding so killed all vlc's and restarted them, so im totally at a loss now, unless the rpi 2 b+ just isnt powerful enough 👎

Vattar, 2022-12-10:
@si458  I believe we are having the same problem, however, I have no idea how I can solve it, nor where the problem is.

I've tried to reinstall using other versions of RaspiOS, as well as various combinations of system settings and rpisurv itself, but to no avail.

I can't say with 100% certainty, but I suspect this problem started when I updated vlc from version 3.0.16 to 3.0.17. I tried to downgrade to the old version of vlc but was unsuccessful.

si458, 2023-07-29:
@SvenVD im still having this issue with VLC doing a memory leak,
i think i may have identified the problem,
it might be down to me connecting the pi to a DVR unit and the BNC cable being faulty for the camera on a certain channel
because the VLC im having the memory run away issue with is actually a faulty camera(the cable is broken) and i cant replace it,
so sometimes the image will just show a BLACK picture with a timestamp on it, BUt the monitor on the Pi shows the timestamp has frozen. 
BUT if i connect with VLC from my laptop with the same RTSP url, the timestamp is indeed moving just a black image
so VLC doesnt like the fact the camera goes on and off due to a faulty BNC cable
so the checker thats built into rpisurv to check is the stream is working, isnt working correctly to restart the VLC

SvenVD, 2023-07-30:
If VLC does not crash ( or is killed) and the remote device is still responding then Rpisurv will not attempt to restart vlc. Using earlyoom or a cronjob to restart seems to be a good enough solution. Otherwise you can perhaps report the issue to the VLC developers.

SvenVD, 2023-11-30:
stale issue

closed 2023-11-30 by SvenVD