SvenVD/rpisurv #171: I can't install rpisurv

Vattar, 2022-12-06:
I can not complete the installation of rpisurv, any solution?

Output after try to install...

Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support

Vlc version:

pi@raspberrypi:~$ vlc --version

VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)

Vattar, 2022-12-08:
Solved!

sudo apt remove vlc* libvlc*
sudo apt autoremove
sudo apt autoclean
sudo apt update
sudo apt install libvlc5 vlc

sudo ./install (rpisurv)

Vattar, 2022-12-09:
Hello, @tboyce021 

Your tutorial and it also works, however, but after a few hours of use all streams crashed, with the following output.

systemctl status rpisurv ->

```
Dec 08 20:22:06 raspberrypi rpisurv[642]: 2022/12/08 20:22:06 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 08 21:31:00 raspberrypi rpisurv[642]: 2022/12/08 21:31:00 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 08 22:39:52 raspberrypi rpisurv[642]: 2022/12/08 22:39:52 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 08 23:48:44 raspberrypi rpisurv[642]: 2022/12/08 23:48:44 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 00:57:36 raspberrypi rpisurv[642]: 2022/12/09 00:57:36 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 02:06:28 raspberrypi rpisurv[642]: 2022/12/09 02:06:28 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 03:15:21 raspberrypi rpisurv[642]: 2022/12/09 03:15:21 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 04:23:02 raspberrypi rpisurv[642]: 2022/12/09 04:23:02 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 05:29:38 raspberrypi rpisurv[642]: 2022/12/09 05:29:38 - l_default - INFO - Sending stats is disabled, not sending stats
Dec 09 06:36:16 raspberrypi rpisurv[642]: 2022/12/09 06:36:16 - l_default - INFO - Sending stats is disabled, not sending stats
```

After restarting the rpisurv the streams work again, but with other error messages. Any idea about these it ?

```
Dec 09 06:40:40 raspberrypi rpisurv[1237]: [b36105f0] main decoder error: buffer deadlock prevented
Dec 09 06:40:40 raspberrypi rpisurv[1237]: [b3510608] main decoder error: buffer deadlock prevented
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
Dec 09 06:40:40 raspberrypi rpisurv[1237]: error: XDG_RUNTIME_DIR not set in the environment.
```

si458, 2022-12-09:
@Vattar im had the exact same issue as you where the streams where crashing after so long, i ended up putting a `systemctl restart rpisurv` in my cronjob every 6 hours but mine was a memory leak issue it seemed? #170

gh3597, 2023-04-16:
> Create `/etc/apt/preferences.d/vlc` with the following:
> 
[...]
> 
> If you already have vlc/libvlc installed then you'll need to remove and reinstall them:
> 
[...]
> 
> Then you should be able to install/run rpisurv.

I've tried this many times (with additional cleaning/purging and apt-get as well) and still get the same issue, seems apt is ignoring preferences.d.

````
pi@monitor:~ $ apt policy vlc
vlc:
  Installed: (none)
  Candidate: 3.0.18-0+rpt3+deb11u1
  Version table:
     3.0.18-0+rpt3+deb11u1 600
        500 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages
     3.0.18-0+deb11u1 -1
        500 http://deb.debian.org/debian bullseye/main arm64 Packages
        500 http://security.debian.org/debian-security bullseye-security/main arm64 Packages
````

Vattar, 2023-11-21:
**Again with the same problem!

I've already tried the solution proposed by user tboyce021, but it doesn't work either!

The VLC version seems correct, but I can't install rpisurv.

Any idea ?**

pi@raspberrypi:~/rpisurv $ sudo ./install.sh
Use this installer on your own risk. Make sure this host does not contain important data and is replacable
This installer will disable graphical login on your pi, please revert with the raspi-config command if needed.

The following version will be installed: "3.0.0"

Do you want to continue press <Enter>, <Ctrl-C> to cancel

Hit:1 http://deb.debian.org/debian buster InRelease                                                      
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease                                     
Hit:3 http://deb.debian.org/debian buster-updates InRelease                           
Hit:4 http://archive.raspberrypi.org/debian buster InRelease                          
Reading package lists... Done                                   
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
coreutils is already the newest version (8.30-3).
fbset is already the newest version (2.1-30).
procps is already the newest version (2:3.3.15-2).
python3-pygame is already the newest version (1.9.4.post1+dfsg-3).
python3-openssl is already the newest version (19.0.0-1).
python3 is already the newest version (3.7.3-1).
python3-yaml is already the newest version (3.13-2).
rsync is already the newest version (3.1.3-6).
sed is already the newest version (4.7-1).
openssl is already the newest version (1.1.1n-0+deb10u6).
ffmpeg is already the newest version (7:4.1.11-0+deb10u1+rpt1).
libraspberrypi-bin is already the newest version (2+git20211125~155417+14b90ff-1~buster).
vlc is already the newest version (3.0.17.4-0+deb10u2+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support

pi@raspberrypi:~/rpisurv $ vlc --version
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
VLC version 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Compiled by serge on arm-build.pitowers.org (Dec  7 2022 23:06:22)
Compiler: gcc version 8.3.0 (Debian 8.3.0-6)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public Licence;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.

gh3597, 2023-11-21:
I gave up, stuck with Buster and used the package I was using previously.

Vattar, 2023-11-21:
> I gave up, stuck with Buster and used the package I was using previously.

I'm using Debian Buster, yet rpisurv refuses to install.

This is a new problem for me... ;/

2wenty2wo, 2023-11-28:
Just replaced my Rpi3 with a Rpi4 and I'm facing this same issue. The Rpisurv installation auto updates VLC to a version that isn't compatible.

2wenty2wo, 2023-11-28:
I tried the VLC fix above with Buster and I still can't get rpisurv to install.

Vattar, 2023-11-28:
> Tentei a correção do VLC acima com Buster e ainda não consigo instalar o rpisurv. Executando um Raspberry Pi 4.
> 
> **Editar:** Problema [nº 167](https://github.com/SvenVD/rpisurv/issues/167) - "O sistema operacional Raspberry Pi de 64 bits não parece suportar mmal para VLC." com a resposta: _"Basta usar Raspberry Pi OS Full (Legacy) 32 bits?"_
> 
> Experimentei, instalei o Buster de 32 bits e trabalhei na primeira vez, sem problemas. Também não foi necessário fazer a correção acima. A instalação funcionou conforme o esperado, sem erros.
> 
> Demorei um pouco para encontrar a imagem do Buster, pois ela é muito antiga agora, se ajudar alguém, baixei o Buster aqui: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/

Thanks! The issue was resolved installing the 32-bit version.

closed 2023-11-30 by SvenVD