SvenVD/rpisurv #172: streaming URL doesn't work

turbo2ltr, 2023-02-04:
I have three Ubiquiti cameras streaming wonderfully.   I wanted to add the stream from my 3D printer camera.

If I open the command line and type `vlc http://10.0.0.110:8080/?action=stream` it opens VLC and displays the stream.

But if I add it to the rpisurv config, that camera never connects.

```
     - camera_streams:
         - url: "rtsp://10.0.0.1:7447/XXXXXXXX8B15"
           rtsp_over_tcp:  True
           freeform_advanced_vlc_options: " --rtsp-frame-buffer-size=1000000 "
         - url: "rtsp://10.0.0.1:7447/XXXXXXXXXBm8lY"
           rtsp_over_tcp:  True
           freeform_advanced_vlc_options: " --rtsp-frame-buffer-size=1000000 "
         - url: "rtsp://10.0.0.1:7447/XXXXXXXXXX4T1A"
           rtsp_over_tcp:  True
           freeform_advanced_vlc_options: " --rtsp-frame-buffer-size=1000000 "
         - url: "http://10.0.0.110:8080/?action=stream"
```
I'm not really understanding what the issue is.  As a test I changed the URL to an image on that same server and it displayed it without issue so the connectivity is there.  Any ideas?
Thanks.

SvenVD, 2023-02-04:
can you try what happens if you remove ubiquity cameras and only keep the printer camera?

turbo2ltr, 2023-02-04:
Still nothing.  Just says "Connecting"

turbo2ltr, 2023-02-04:
I found an odd behavior.. When I have all four cameras configured, the 4th camera shows a black screen with a big "connecting..." in the center.
I then turned off the camera and a few seconds later, all four camera views momentarily blanked out, and then came back up, but this time the 4th came up with a white screen that had a R logo on it.
If a camera was down, I always saw the connecting screen so I thought maybe this may be of note.

SvenVD, 2023-02-04:
> 

This is normal behaviour, rpisurv removes dead cameras.

Can you post some debug logs, I suspect something wrong the http probe in combination with your specific camera

SvenVD, 2023-02-04:
Alternatively and maybe easier is to first try putting disable_probing_for_all_streams: to True and restarting, is the problem still there?

turbo2ltr, 2023-02-05:
This was with disable probing set to true.

```
2023/02/04 17:10:40 - l_default - INFO - Starting rpisurv 3.0.0-beta7
2023/02/04 17:10:40 - l_default - DEBUG - Make sure vlc binary can be executed as root
2023/02/04 17:10:40 - l_default - INFO - Auto detected display DEL-DELL_E2414H at HDMI 0 with display number 2 1920 x 1080
2023/02/04 17:10:40 - l_default - INFO - Unique id of this installation is a90c0c4579a07d18bb00e19005c74aec7875288bc848375f2bfc589cd086e270
2023/02/04 17:10:40 - l_default - DEBUG - Start_time is 1675555840.6246085
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0: Looking for config file conf/display1.yml
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0: Initialising screen with config {'camera_streams': [{'url': 'rtsp://10.0.0.1:7447/zXVHvRNTCO4X8B15', 'rtsp_over_tcp': True, 'freeform_advanced_vlc_options': ' --rtsp-frame-buffer-size=1000000 '}, {'url': 'rtsp://10.0.0.1:7447/dWzRjRi1TqXBm8lY', 'rtsp_over_tcp': True, 'freeform_advanced_vlc_options': ' --rtsp-frame-buffer-size=1000000 '}, {'url': 'rtsp://10.0.0.1:7447/YDiaeW5Eiqjo4T1A', 'rtsp_over_tcp': True, 'freeform_advanced_vlc_options': ' --rtsp-frame-buffer-size=1000000 '}, {'url': 'http://10.0.0.110:8080/?action=stream'}], 'disable_probing_for_all_streams': True, 'duration': 60}
2023/02/04 17:10:41 - l_default - DEBUG - Screen: screen_manager_0_screen1 duration from config is: 60
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0 Drawing placeholder with coordinates: 0, 0 and width: 1920 height: 1080 with image images/connecting.png
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0: only one screen configured, do not rotate
2023/02/04 17:10:41 - l_default - DEBUG - Screen: reset_active_timer screen_manager_0_screen1
2023/02/04 17:10:41 - l_default - DEBUG - Screen: screen_manager_0_screen1 start_of_active_time: 1675555841.7439792
2023/02/04 17:10:41 - l_default - DEBUG - MAIN screen_manager_0: bootstrap update_active_screen
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0: update_active_screen screen_manager_0_screen1
2023/02/04 17:10:41 - l_default - DEBUG - Screen: screen_manager_0_screen1 Skipping checking connectable cameras, skip value is: True
2023/02/04 17:10:41 - l_default - DEBUG - Screen: screen_manager_0_screen1 first run
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0: SKIPPING update_connectable_camera_streams, because disable_probing_for_all_streams for this screen screen_manager_0_screen1 was set
2023/02/04 17:10:41 - l_default - DEBUG - Screen screen_manager_0_screen1 needs update/redraw: changes in connectable camera streams detected.( previous: 0 / now: 4 or different connectable streams then before )
2023/02/04 17:10:41 - l_default - DEBUG - Screen: screen_manager_0_screen1 number of fields= 4
2023/02/04 17:10:41 - l_default - DEBUG - Screen: cam stream name =screen_manager_0_screen1_cam_stream1
2023/02/04 17:10:41 - l_default - DEBUG - CameraStream: Start stream screen_manager_0_screen1_cam_stream1 on layer 2000000000
2023/02/04 17:10:41 - l_default - DEBUG - CameraStream: Stop stream screen_manager_0_screen1_cam_stream1
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0 Drawing placeholder with coordinates: 0, 0 and width: 960 height: 540 with image images/connecting.png
2023/02/04 17:10:41 - l_default - DEBUG - Screen: cam stream name =screen_manager_0_screen1_cam_stream2
2023/02/04 17:10:41 - l_default - DEBUG - CameraStream: Start stream screen_manager_0_screen1_cam_stream2 on layer 2000000000
2023/02/04 17:10:41 - l_default - DEBUG - CameraStream: Stop stream screen_manager_0_screen1_cam_stream2
2023/02/04 17:10:41 - l_default - DEBUG - screen_manager_0 Drawing placeholder with coordinates: 960, 0 and width: 960 height: 540 with image images/connecting.png
2023/02/04 17:10:41 - l_default - DEBUG - Screen: cam stream name =screen_manager_0_screen1_cam_stream3
2023/02/04 17:10:41 - l_default - DEBUG - CameraStream: Start stream screen_manager_0_screen1_cam_stream3 on layer 2000000000
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: Stop stream screen_manager_0_screen1_cam_stream3
2023/02/04 17:10:42 - l_default - DEBUG - screen_manager_0 Drawing placeholder with coordinates: 0, 540 and width: 960 height: 540 with image images/connecting.png
2023/02/04 17:10:42 - l_default - DEBUG - Screen: cam stream name =screen_manager_0_screen1_cam_stream4
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: Start stream screen_manager_0_screen1_cam_stream4 on layer 2000000000
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: Stop stream screen_manager_0_screen1_cam_stream4
2023/02/04 17:10:42 - l_default - DEBUG - screen_manager_0 Drawing placeholder with coordinates: 960, 540 and width: 960 height: 540 with image images/connecting.png
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: This stream screen_manager_0_screen1_cam_stream1 is not an imageurl, skip refreshing imageurl
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: This stream screen_manager_0_screen1_cam_stream2 is not an imageurl, skip refreshing imageurl
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: This stream screen_manager_0_screen1_cam_stream3 is not an imageurl, skip refreshing imageurl
2023/02/04 17:10:42 - l_default - DEBUG - CameraStream: This stream screen_manager_0_screen1_cam_stream4 is not an imageurl, skip refreshing imageurl
[... cut ...]
2023/02/04 17:10:42 - l_default - DEBUG - free reloc gpu mem value is 468M
2023/02/04 17:10:42 - l_default - DEBUG - total available reloc gpu mem value is 492M
2023/02/04 17:10:42 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:468M outputmem_bytes:490733568.0
2023/02/04 17:10:42 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:492M outputmem_bytes:515899392.0
2023/02/04 17:10:42 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
```

SvenVD, 2024-11-17:
closing old age

closed 2024-11-17 by SvenVD