SvenVD/rpisurv #174: rpisurv works on Bullseye and RPI 4 except being started by systemctl

russgmoore, 2023-06-18:
I loaded Bullseye on a new microsd card on a new rpi 4 today. I used the full install bullseye. 
I changed the settings in raspi-config to boot to console.
Set the following in /boot/config.txt
dtoverlay=vc4-fkms-v3d 
gpu_mem=512
disable_overscan=1

I can run rpisurv from a command prompt with sudo and it works just fine with my 6 cameras. If I try to start it via systemctl the screen remains black. So, until I can debug what's causing the startup script method to fail, I'm using "screen" to run the process in a background vty.  I need Bullseye because I'm using a Camera Module 3 and that won't run on Buster.

russgmoore, 2023-06-18:
Updates, I set my system to boot to console/autologin. But then had weird hdmi/vidoe problems. After studying my setup further, I determined I had an issue with the "gpu_mem" config. I removed that and rpisurv started fine but hung up after loading. I assumed that was due to running too low on GPU memory. I then went and read more about Bullseye and RPI 4. I have now set gpu_mem to 128 in /boot/config.txt and this appears to run fine with 6 displays online.
gpu=128M
malloc_total=16M
reloc_total=108M
reloc=46M

I'm happy to say its running fine on the latest full Raspbian and a RPI 4 4GB. I can also access the Camera Module 3 so life is great.

russgmoore, 2023-06-18:
should have posted in tapatalk.

UntouchedWagons, 2023-09-20:
I can't install rpisurv on Bullseye:

    $ sudo ./install.sh
    Use this installer on your own risk. Make sure this host does not contain important data and is replacable
    This installer will disable graphical login on your pi, please revert with the raspi-config command if needed.

    The following version will be installed: "3.0.0"

    Do you want to continue press <Enter>, <Ctrl-C> to cancel

    Hit:1 https://deb.debian.org/debian bookworm InRelease
    Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
    Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
    Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
    Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
    Reading package lists... Done
    Building dependency tree... Done
    Reading state information... Done
    All packages are up to date.
    Reading package lists... Done
    Building dependency tree... Done
    Reading state information... Done
    sed is already the newest version (4.9-1).
    sed set to manually installed.
    coreutils is already the newest version (9.1-1).
    coreutils set to manually installed.
    openssl is already the newest version (3.0.9-1).
    openssl set to manually installed.
    procps is already the newest version (2:4.0.2-3).
    libraspberrypi-bin is already the newest version (1:2+git20230322~143557+9d5250f-1).
    Some packages could not be installed. This may mean that you have
    requested an impossible situation or if you are using the unstable
    distribution that some required packages have not yet been created
    or been moved out of Incoming.
    The following information may help to resolve the situation:

    The following packages have unmet dependencies:
    libvlccore9 : Depends: libidn11 (>= 1.13) but it is not installable
    vlc-plugin-base : Depends: libavcodec58 (>= 7:4.2)
                    Depends: libavfilter7 (>= 7:4.0)
                    Depends: libavformat58 (>= 7:4.2) but it is not going to be installed
                    Depends: libdav1d4 (>= 0.1.0) but it is not installable
                    Depends: libflac8 (>= 1.3.0) but it is not installable
                    Depends: libprotobuf-lite23 (>= 3.12.4) but it is not installable
                    Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable
                    Depends: libx264-160 (>= 2:0.160.3011+gitcde9a93) but it is not installable
                    Depends: libx265-192 (>= 3.2) but it is not installable
    vlc-plugin-video-output : Depends: libplacebo72 (>= 0.4.0) but it is not installable
    E: Unable to correct problems, you have held broken packages.
    sed: can't read /usr/bin/vlc: No such file or directory
    Your version of vlc does not have the needed mmal options. Rpisurv needs those
    Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
    Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
    Aborting installation, upgrade to latest vlc player with mmal support

closed 2023-11-30 by SvenVD

UntouchedWagons, 2023-11-30:
How did you get it installed? I tried installing rpisurv on a pi 3 running debian 11 and 12 and it won't install because vlc doesn't support mmal.