SvenVD/rpisurv #176: disable_autorotation: True is ignored after switching camera.

shitfacevk, 2023-08-31:
Setup.
RPI3 with 7" Offical rpi touchscreen.
os: "Raspbian GNU/Linux 10 (buster)"
2 cameras.

I dont want autorotation, so i have disabled that. But when i switch camera by tapping the screen, the autorotation setting is ignored and i start rotatin stream every 30sec. between the two cameras. If i dont tap the screen autorotation is disabled. 

Its in the logs.
Aug 30 19:28:36 raspberrypi rpisurv[17909]: 2023/08/30 19:28:36 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - screen_manager_0 draw: pygame.MOUSEBUTTONUP detected
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - Screen: screen_manager_0_screen1 active_run_time: 11 / 30

cat /etc/rpisurv/display1.yml

essentials:
  disable_autorotation: True

  screens:
    - camera_streams:
          #1
        - url: "rtsp://192.168.1.1:7447/SeNGm5oSKCghDmXn"
          rtsp_over_tcp: true
    - camera_streams:
          #2
        - url: "rtsp://192.168.1.1:7447/H6z2LUI2bW3GU408"
          rtsp_over_tcp: true

advanced:
  fixed_width: 800
  fixed_height: 480
  update_stats: False
  #cache_next_screen: True
  interval_check_status: 19
  #memory_usage_check: True

/var/log/daemon.log:

Aug 30 19:28:36 raspberrypi rpisurv[17909]: 2023/08/30 19:28:36 - l_default - DEBUG - screen_manager_0: disable_autorotation is True, use input keyboard/mouse/touch only to rotate between screens
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - screen_manager_0 draw: pygame.MOUSEBUTTONUP detected
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - screen_manager_0 draw touch/mouse handling: pygame detected display width of 800
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - screen_manager_0 draw touch/mouse handling: firstQuarter 200.0 and lastquarter 600.0
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - screen_manager_0 draw touch/mouse handling: detected touch/mouse in middle
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - MAIN: resume_rotation event detected
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - free malloc gpu mem value is 10M
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - total available malloc gpu mem value is 12M
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:10M outputmem_bytes:10485760.0
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:12M outputmem_bytes:12582912.0
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - free reloc gpu mem value is 179M
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - total available reloc gpu mem value is 236M
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:179M outputmem_bytes:187695104.0
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - convert_gpumem_string_to_bytes: inputmem:236M outputmem_bytes:247463936.0
Aug 30 19:28:37 raspberrypi rpisurv[17909]: 2023/08/30 19:28:37 - l_default - DEBUG - Screen: screen_manager_0_screen1 active_run_time: 11 / 30

SvenVD, 2023-09-02:
It depends where you touch on the screen, you need to touch on the right quarter for next screen only. Touching in the middle two quarters (re)enables autorotation and touching in the first quarter pauses autorotation.

shitfacevk, 2023-09-02:
> It depends where you touch on the screen, you need to touch on the right quarter for next screen only. Touching in the middle two quarters (re)enables autorotation and touching in the first quarter pauses autorotation.

Ofc!

Touchscreen control: The width of the screen is divided in four sections,

Touching on the first section trigger a pause event.
Touching In the two sections in the middle trigger a resume event.
Touching In the last section, a next screen event. Note that a mouse can be used, however mouse cursor is hidden by default.

Thank you, seems to work!

closed 2023-09-02 by shitfacevk