SvenVD/rpisurv #178: RPISurv (2x2) not working well on RasPI 3 Model B+ Bullseye

tartech34, 2023-11-01:
Hello, I am trying to get RPISurv to work on a raspberry pi to output RTSP streams from Digital Watchdog, but I am unable to get the service to start. Below is the output of sudo systemctl status rpisurv.

pi@pi-adm-ath1:~ $ pi@pi-adm-ath1:~ $ systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
     Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2023-10-31 17:35:46 EDT; 4s ago
    Process: 1060 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
   Main PID: 1060 (code=exited, status=1/FAILURE)
        CPU: 1.471s

**Here is the output of main.log:**
2023/10/31 17:36:07 - l_default - INFO - Starting rpisurv 3.0.0
2023/10/31 17:36:07 - l_default - DEBUG - Make sure vlc binary can be executed as root
2023/10/31 17:36:29 - l_default - INFO - Starting rpisurv 3.0.0
2023/10/31 17:36:29 - l_default - DEBUG - Make sure vlc binary can be executed as root

**And the output daemon.log:**

> Oct 31 17:37:11 pi-adm-ath1 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: /usr/local/bin/rpisurv/core/util/setuplogging.py:15: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/l>
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:   logcfg = yaml.load(ymlfile)
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: pygame 1.9.6
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: Hello from the pygame community. https://www.pygame.org/contribute.html
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: 2023/10/31 17:37:13 - l_default - INFO - Starting rpisurv 3.0.0
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: 2023/10/31 17:37:13 - l_default - DEBUG - Make sure vlc binary can be executed as root
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1090]: /usr/bin/tvservice is not supported when using the vc4-kms-v3d driver.
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1090]: Similar features are available with standard linux tools
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1090]: such as kmsprint from kms++-utils
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: Traceback (most recent call last):
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:   File "/usr/local/bin/rpisurv/surveillance.py", line 170, in <module>
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:     displays=parse_tvservice()
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:   File "/usr/local/bin/rpisurv/surveillance.py", line 69, in parse_tvservice
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:     tvserviceresult_l = subprocess.check_output(['/usr/bin/tvservice', '-l'], text=True )
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:   File "/usr/lib/python3.9/subprocess.py", line 424, in check_output
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:     return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:   File "/usr/lib/python3.9/subprocess.py", line 528, in run
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]:     raise CalledProcessError(retcode, process.args,
> Oct 31 17:37:13 pi-adm-ath1 rpisurv[1087]: subprocess.CalledProcessError: Command '['/usr/bin/tvservice', '-l']' returned non-zero exit status 1.
> Oct 31 17:37:13 pi-adm-ath1 systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
> Oct 31 17:37:13 pi-adm-ath1 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
> Oct 31 17:37:13 pi-adm-ath1 systemd[1]: rpisurv.service: Consumed 1.477s CPU time.
> Click to expand...

Thank you in advance.  I've used a script that worked using deprecated omxplayer in the past for Unifi cameras, but am new to rpisurv, looks like an excellent project.

Update:  It seems the closest workaround I can find on Bullseye is to run: sudo raspi-config and set the GL Driver to Legacy under 'Advanced Options'.  However, I am seeing poor performance for the feeds running the standard 2x2 grid and RTSP feeds (640x480).  I previously used omxplayer to run 3x3 grid of the same RTSP feeds using the same Raspberry Pi hardware with little to no issues.  Am I missing something to optimize performance?

SvenVD, 2023-11-19:
Hi, Sorry currently only buster is supported. There are workarounds for bullseye https://www.tapatalk.com/groups/rpisurv/viewtopic.php?p=1113#p1113

closed 2023-11-30 by SvenVD

aniceberg, 2023-12-14:
> Thank you for the response. We’ve settled on Buster. I’m still having difficulty with the smoothness of video playback on a Raspberry Pi 4 (4gb) but I’m wondering if it’s an issue with the new cameras we have. 

Try lowering the stream resolution to the minimum, the RPi4 will choke on multiple streams, even at 720p.