SvenVD/rpisurv #179: Garbled video at higher resolution

turbo2ltr, 2023-11-21:
I've been running three Unifi cameras on a 1080 monitor without issue for years. The streams I use were the "medium" streams (1280x720)

I upgraded to a larger 4K monitor so I went to change the stream to the "high" stream (2688x1512) and I get what you see in the top left.  

[image]

The main darker area looks like it "scrolls" up. Like every time the frame is drawn it's drawn one pixel off.  The green part on the right side of the frame is a repeating pattern of random garbage.

Opening the same stream in VLC works fine.

```
- camera_streams:
         - url: "rtsp://10.0.0.1:7447/xxx"
           rtsp_over_tcp:  True
           freeform_advanced_vlc_options: " --rtsp-frame-buffer-size=1000000"
           force_coordinates: [0, 0, 1920,1080 ]
```

Any suggestions?

turbo2ltr, 2023-11-21:
I changed the doorbell cam from Medium (960x720) to High (1600x1200) without issue.

SvenVD, 2023-11-30:
did you fix it?

turbo2ltr, 2023-11-30:
No I couldn't get it to work.

SvenVD, 2024-11-17:
> I had similar issues and lockups till I set all of my secondary streams to 640x480 and used that secondary stream for rpiserv.

This is the way for Rpisurv and raspberry pi hardware. For your information, https://opensurv.net enables you to use more powerful hardware.

closed 2024-11-17 by SvenVD