SvenVD/rpisurv #180: new install on buster, VLC error in the install script

zeltak, 2023-12-01:
Hi

im getting this error RE VLC on install 

```
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support
```

any clue?

best

z

Vattar, 2023-12-02:
If your system architecture is 64-bit, consider installing the 32-bit version.

Also, consider checking the installed version of VLC with the command.

vlc --version

msman0, 2023-12-10:
Hi, I have the same problem:

OS:

`PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster`

`
vlc --version
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
VLC version 3.0.20 Vetinari (3.0.20-0-g6f0d0ab126b)`

It is already the 32 bit version.
What can we do to install it?
Thank you!

Vattar, 2023-12-10:
> Hi, I have the same problem:
>
> OS:
>
> `PRETTY_NAME="Raspbian GNU/Linux 10 (buster)" NAME="Raspbian GNU/Linux" VERSION_ID="10" VERSION="10 (buster)" VERSION_CODENAME=buster`
>
> ` vlc --version VLC media [...]

The version 3.0.20 is not compatible with rpisurv. Follow this steps....

1- `sudo apt purge vlc-data`

2- `sudo apt purge vlc*`

3- `sudo apt autoremove`

4- Create `/etc/apt/preferences.d/vlc` with the following:

```
Package: vlc*
Pin: origin "archive.raspberrypi.org"
Pin-Priority: 500

Package: libvlc*
Pin: origin "archive.raspberrypi.org"
Pin-Priority: 500

Package: vlc*
Pin: origin *
Pin-priority: -1

Package: libvlc*
Pin: origin *
Pin-priority: -1
```

5- `sudo apt update`

6- `Reinstall rpisurv`

msman0, 2023-12-10:
Great Thank you! It works now.

pagso, 2024-01-04:
> The version 3.0.20 is not compatible with rpisurv. Follow this steps....
>
> 1- `sudo apt purge vlc-data`
>
> 2- `sudo apt purge vlc*`
>
> 3- `sudo apt autoremove`
>
> 4- Create `/etc/apt/preferences.d/vlc` [...]

Hi. im not very good at linux, can you dumb down step 4 for me? ive been trying to figure out how to make these things but i just dont know

Vattar, 2024-01-04:
> Hi. im not very good at linux, can you dumb down step 4 for me? ive been trying to figure out how to make these things but i just dont know

Hello (:

To create the file, just use the "touch" command

`sudo touch /etc/apt/preferences.d/vlc`

Now, open the created file with a text editor

`sudo nano /etc/apt/preferences.d/vlc`

Now just copy the informations from step 4 and paste it into the file using the right mouse click or CTRL + SHIFT + V

To save the file press CTRL + X and then Y to confirm and ENTER to exit.

If everything was done correctly, you should be able to read the information written to the file with the following command:

`cat /etc/apt/preferences.d/vlc`

Vattar, 2024-01-04:
> Hi. im not very good at linux, can you dumb down step 4 for me? ive been trying to figure out how to make these things but i just dont know

Hello (:

To create a file, just use the "nano" command

`sudo nano /etc/apt/preferences.d/vlc`

Copy all informations from step 4 and paste it into the file using the right mouse click or CTRL + SHIFT + V

To save the file press CTRL + X and then Y to confirm and ENTER to exit.

If everything was done correctly, you should be able to read the information written to the file with the following command:

`cat /etc/apt/preferences.d/vlc`

pagso, 2024-01-05:
> Hello (:
>
> To create a file, just use the "nano" command
>
> `sudo nano /etc/apt/preferences.d/vlc`
>
> Copy all informations from step 4 and paste it into the file using the right mouse click or CTRL + [...]

Thank you very much your a life saver

dejan9393, 2024-03-08:
@raymanEUS I had the same issue, for me it was a matter of removing `libvlc-bin`:

```shell
sudo apt purge libvlc-bin
```

wifispray, 2024-03-18:
Still seeing an error. 

Package vlc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'vlc' has no installation candidate
sed: can't read /usr/bin/vlc: No such file or directory
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support