SvenVD/rpisurv #181: More responsive image updates

EnriqueWood, 2023-12-05:
I've been working with rpisurv for a while now and noticed something about how it handles and reacts image changes.
It feels like the system checks for image updates a bit slow, after I change an image it takes a while to update the image shown.

Is this tied to the `interval_check_status` in the `general.yml` file?

Is there a way to check images more often than streams?

I don't want my streams to be refreshing all the time because all of them have to reconnect and that takes time, which leaves me with a black screen for some seconds (around 5 seconds in my setup), however, for local images that is almost instantly (I have an external program changing images in response to events in my house security system).

Thank you for such a great project!

Also, could you give me some pointers to where do I have to look in the code to implement it myself in case you don't have the time/don't want to do it yourself?

Vattar, 2023-12-10:
https://github.com/SvenVD/rpisurv/issues/180#issuecomment-1849007283

SvenVD, 2024-01-03:
yes, interval_check_status  is the refresh timer for images, but is also the refresh timer for each watchdog to wakeup for probing the health of each camera stream. Currently you can not set different values for those two. But try lowering and see if it works.

SvenVD, 2024-11-17:
https://opensurv.net has a faster update cycle for images independent of health check of cameras

SvenVD, 2024-11-17:
https://opensurv.net/ has a faster update cycle for images independent of health check of cameras

closed 2024-11-17 by SvenVD