SvenVD/rpisurv #182: rpisurv v3 now works under Bullseye, my issue is with 2x2 rotating my 8 cameras

Handssolow, 2023-12-07:
To my surprise after having no, no no success with Buster over several hours of work, because never versions of VLC do not have the mmal module in Buster's repository, rpisurv works here under 32bit Bullseye, git clone --depth=1 https://github.com/SvenVD/rpisurv delivers the v3 version download. I found no need to alter any RPi graphics configuration, it was already there.
I have a working 2x2 display of cctv rtsp cameras, here though I have 8 cameras and I am wanting a 2x2 rotating to another 2x2. No success so far, I remain mystified by python' s programming  requirements.
I prevent any of my cctv system devices having access to the WAN, and that includes rpisurv.
My downloaded rpisurv is in a home/pi directory, it isn't in /etc/ even after /install.sh, rpisurv is in usr/locl/bin.

Vattar, 2023-12-10:
Please, check this: https://github.com/SvenVD/rpisurv/issues/180#issuecomment-1849007283

Handssolow, 2023-12-11:
Success, I now have rpisurv v3 running under Bullseye on a RPi 3B v1.2, it is mostly working as I wanted, that's 2x2 rotating x2 from my 8 cameras' rtsp NVR feeds. I use the sub-stream feeds which is acceptable.
What brought success was editing the general.yml file to stop opportunistic caching, ie enable_opportunistic_caching_next_screen: False.
Earlier I got a sense that caching was the issue here, altering the GPU memory allocation did not work.

I suppose the only downside currently is at the 2x2 display changeover, I briefly see a 2x2 matrix of "connecting" messages. Caching seems to rob resources too early that the first 2x2 screen still needs for display.

closed 2024-11-17 by SvenVD