SvenVD/rpisurv #183: Won't launch when connected to other displays

designatedduff, 2023-12-30:
The streams work on the monitor I used for initial setup, but I have connected the pi to 2 other TV's and rpisurv doesn't launch
Raspberrypi 3B+
Buster

  GNU nano 3.2                                                                                        /usr/local/bin/rpisurv/logs/main.log                                                                                                  

2023/12/25 20:08:03 - l_default - INFO - Starting rpisurv 3.0.0
2023/12/25 20:08:03 - l_default - INFO - Auto detected display ACR-G236HL at HDMI 0 with display number 2 1920 x 1080
2023/12/25 20:08:03 - l_default - INFO - Unique id of this installation is 033bc2e40acf0820fd32df002dc5b3874c18c35066e988f886de11cc7b3e5759
2023/12/25 20:08:10 - l_default - INFO - Sending stats is disabled, not sending stats
2023/12/25 20:08:28 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen3 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:08:49 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen4 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:09:19 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen1 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 19:50:05 - l_default - INFO - Starting rpisurv 3.0.0
2023/12/25 19:50:05 - l_default - INFO - Auto detected display ACR-G236HL at HDMI 0 with display number 2 1920 x 1080
2023/12/25 19:50:05 - l_default - INFO - Unique id of this installation is 65284c16d6b5074d64f7d190c59c42696d52af27490907bac3a4b52f52396eb3
2023/12/25 19:50:11 - l_default - INFO - Sending stats is disabled, not sending stats
2023/12/25 19:50:30 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen3 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:10:27 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen4 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:10:58 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen1 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:11:38 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen3 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:11:58 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen4 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:12:29 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen1 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:13:09 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen3 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)
2023/12/25 20:13:30 - l_default - INFO - screen_manager_0: skip caching next screen: screen_manager_0_screen4 (pygame, used to display image, is not compatible with dispmanx layers caching mechanism)

SvenVD, 2023-12-31:
I do not see any relevant error in these logs

SvenVD, 2024-11-17:
stale issue

closed 2024-11-17 by SvenVD