SvenVD/rpisurv #184: 2 streams 1 column fails to work

itsmeduhdoi, 2024-01-12:
I'm trying to have 2 camera streams appear on screen in 1 column, but when they both load, and i have to force tcp to get them to, they don't refresh.  They don't get detected as dead streams, and i can log in to the cameras themselves and see that they are sending a stream to my rpisurv device, but the streams the device shows are a static image of when it first connected.

1 single stream, of either camera, works as expected.

occasionally, using backspace to return to command line, making adjustments in the display1.yml and restarting causes the device to lockup and i have to manually power cycle to get it functioning again.

SvenVD, 2024-01-13:
try lower resolution substreams

itsmeduhdoi, 2024-01-15:
Ok,
Will do.  I have these cameras set with their main stream at h.265, but the sub stream at h.264
some googling has informed me that these cameras don't correctly pass their substreams over rtsp...all url's connect only to the main stream.

itsmeduhdoi, 2024-01-17:
> try lower resolution substreams

got ahold of the manufacturer support group, the URL for these cameras to get to the substream had to be ip:port/multicast/video2  any other version of a url would work but only pull the main stream.

1 column will work now, did i just miss that this can't handle high quality camera feeds somewhere?  thank you

closed 2024-01-17 by itsmeduhdoi

SvenVD, 2024-01-18:
it depends on the hardware of the raspberry pi, it has a limit on how much downscalen from a highr resolution to fit within the window you want to show your stream. Generally try to avoid downscaling.