SvenVD/rpisurv #186: nr_of_columns: 3 not working

abhi237086, 2024-02-16:
when i set the number of columns  to 3 then  rpisurv is not loading, Just working on columns i.e Default 2

essentials:
  #Rpisurv will rotate over the following screens in the order they are configured here, if you only define one screen no rotation will happen

  #If you enable this option no auto rotation will happen, but you will still be able to select a screen by pressing and holding a Function Key (F1-F12) or numpad keys
  #Defaults to False
  #disable_autorotation: False

  screens:
#      #######################################################################################################
#      # Below config is included for demo and test purposes
#      # After installation you can start Rpisurv and the demo should be started
#      #######################################################################################################
    - camera_streams:
        - url: "rtmp://192.168.1.188/live/test1"
        - url: "rtmp://192.168.1.188/live/test2"
        - url: "rtmp://192.168.1.188/live/test3"
        - url: "rtmp://192.168.1.188/live/test4"
        - url: "rtmp://192.168.1.188/live/test5"
        - url: "rtmp://192.168.1.188/live/test6"
        - url: "rtmp://192.168.1.188/live/test7"
        - url: "rtmp://192.168.1.188/live/test8"
        - url: "rtmp://192.168.1.188/live/test9"
    nr_of_columns: 3
     duration: 20
     
     
     Model           : Raspberry Pi 4 Model B Rev 1.2
     OS : Buster

abhi237086, 2024-02-17:
If i Upgrade it to 8 GB then it will work for 9 Stream.

akoryuerek, 2024-02-17:
Unlikely, you need more cpu and/or gpu power. Ram doesn't decode streams.

SvenVD, 2024-11-17:
FYI, https://opensurv.net with the correct hardware can enable you 20 streams or even more. Closing this issue for now

closed 2024-11-17 by SvenVD