SvenVD/rpisurv #187: Cameras Images Freeze after about 24h

SemoTech, 2024-03-02:
Hello, 1st I want to say I am so grateful for all the work that went into this project.

I do have a small issue with RPISurv camera feeds freezing while running on a 4GB Pi 4B. This issue seems to have started fairly recently in the last month. 

I am running a 9 camera tile screen, and about once a day the streams freeze and require me to ssh in, 
I then issue "sudo systemctl restart rpisurv" to restart RPISurv and everything is back to normal. 

My config options for /usr/local/bin/rpisurv/conf/display1.yml are pretty standard and include:

      nr_of_columns: 3
      probe_timeout: 20
      disable_autorotation: True
      duration: 30        
      interval_check_status: 300

I did not used to have the "interval_check_status" option but read about it in another issue so added it.  Hope I added it correctly.

Any idea how to fix the freezing cameras issue?

Thank you.

SemoTech, 2024-03-02:
As an intermediary solution I added a new crontab, as follows:

#Reboot the RPISurv Daemon at 6AM each day
0 6 * * 1 /bin/systemctl restart rpisurv

But hope there is a better way...

SvenVD, 2024-03-06:
9 camera seems like pushing to limit of the gpu, Would it still freeze with less streams?

SemoTech, 2024-03-06:
Thanks for getting back to me @SvenVD.

Well, I have 2 identical Pi's, both 4B w/ 4GB RAM running same RpiSurv on both w/same 9 cameras in low-resolution RTSP streams.  Both are NOT overheating. 
One has no issues, the other the issues I explained.  Also seems like the issues one the one unit started recently and even with the restart cron I find some cameras (squares) frozen while others are live.  Very strange!

SemoTech, 2024-03-17:
I was also looking at the newer Pi 5 with 4GB RAM ([here](https://www.canakit.com/raspberry-pi-5-4gb.html)), and wanted to know if this would be better suitable for RPISurv running 9 RTSP streams (maybe even 12) given it seems to have a better GPU than the Pi4?

What are your thoughts?

SvenVD, 2024-03-20:
> Thanks for getting back to me @SvenVD.
> 
> Well, I have 2 identical Pi's, both 4B w/ 2GB RAM running same RpiSurv on both w/same 9 cameras in low-resolution RTSP streams. Both are NOT overheating. One has no issues, the other the issues I explained. Also seems like the issues one the one unit started recently and even with the restart cron I find some cameras (squares) frozen while others are live. Very strange!
> 
> I just did a top to see the processes and this is the screenshot: ![image](https://private-user-images.githubusercontent.com/1452100/310684181-398ec162-08f8-4639-97b7-95fca6b69256.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NzU2NTQsIm5iZiI6MTcxMDk3NTM1NCwicGF0aCI6Ii8xNDUyMTAwLzMxMDY4NDE4MS0zOThlYzE2Mi0wOGY4LTQ2MzktOTdiNy05NWZjYTZiNjkyNTYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjBUMjI1NTU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjkxMTdhNGEyZTE5YjUzNTA3M2E2MTU5YTgyMDhhN2ViODhlNTBhNWZjNjE1ZGQ4Y2YxYWZlYjc0MmJlYzlkYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.hZLZEj7xG9J5VoKZjkRbeB_51TVIn3P0fDH5DoElMZo)
> 
> This the the top result while several "tiles" of the 9 cameras are frozen: ![image](https://private-user-images.githubusercontent.com/1452100/313480197-fcf5ec1c-8da8-4192-97c0-88dcc2f51733.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NzU2NTQsIm5iZiI6MTcxMDk3NTM1NCwicGF0aCI6Ii8xNDUyMTAwLzMxMzQ4MDE5Ny1mY2Y1ZWMxYy04ZGE4LTQxOTItOTdjMC04OGRjYzJmNTE3MzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjBUMjI1NTU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTRlMjU5YTcxNmJiODU2MDRhNjdjZTkyM2UzNTZiMTFmYWRiZjljMzQ1Njc4OTRhZjE4ZTcwMzg0NzVlN2RkNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9DgGEoual_CDgh26RldQpC_ydwrIEY5Vw02_KdcgKlE)
> 
> Could it be an issue with VLC?

You can not see gpu bandwidth issue of raspberry pi in top. Bandwidth usage can fluctuate even depends on motion in the camera. I still would not rule out  the option of hitting the gpu bandwidth. You can easily test this by only configuring 4 streams or so and see if the issue persist.

SvenVD, 2024-03-20:
> @SvenVD I was also looking at the newer Pi 5 with 4GB RAM ([here](https://www.canakit.com/raspberry-pi-5-4gb.html)), and wanted to know if this would be better suitable for RPISurv running 9 RTSP streams (maybe even 12) given it seems to have a better GPU than the Pi4?
> 
> Pi 5 supports supports H.265 and comes with VideoCore VII at 1.1 GHz replacing VideoCore VI at 800 MHz in the Pi 4
> 
> What are your thoughts?

I have not tested the pi5 yet so I can not make any claims about the gpu

SemoTech, 2024-03-20:
> I have not tested the pi5 yet so I can not make any claims about the gpu

Understood.  I will try to reduce the cameras count from 9 to maybe 6 and see, but the crazy thing is my other identical Pi4 2GB has same 9 cameras and no issues with same RTSP streams freezing! 

I will also see about getting a Pi5 4GB and testing on it as well.  I will post an update once I do. 

Thank you.

SemoTech, 2024-03-23:
Hi @SvenVD on my problem Pi4B 2GB I have reduced the number of cameras from 9 to 6 and same issue.  I also noticed this morning that most of the cameras were frozen so I SSH'd in to issue "sudo systemctl restart rpisurv" but as soon as SSH connected, the cameras refreshed themselves.  Almost as if  the network stack was hung and the SSH connection woke it up...

SemoTech, 2024-04-04:
@SvenVD  I think I have solved it.  

Turns out some of the streams were higher resolution (1280x720) so reducing them all to just 640x360 each has allowed the Pi4B 2GB to handle them and not freeze up some of them randomly.
  
At this time, 9 RTSP streams in a tile configuration of 3 x 3 with each camera stream at 640x360 resolution has been solid again.

closed 2024-04-04 by SemoTech