SvenVD/rpisurv #189: Use Raspberry Pi OS Lite (Debian v12 bookworm) 32Bit or 64Bit on Pi 5 for a RpiSurv fresh install?

SemoTech, 2024-03-21:
Hello @SvenVD,  

As I posted about before and you are aware, I've been having some camera freezing issues with RpiSurv on a Pi4B 2GB that could be related to the GPU getting overloaded or possibly failing.  As such I just ordered a Pi5 4GB and plan to do a clean OS install on it ([Raspberry Pi OS Lite Debian 12 - bookworm](https://www.raspberrypi.com/software/operating-systems/)) but need advice as to what build to install.  Is RpiSurv compatible with both 32 Bit and 64 Bit "Raspberry Pi OS Lite"?  Which bit flavor do you recommend I use?

Also, any tweaks for RpiSurv to get the most out of the new Pi5 GPU and run 9 cameras smoothly again?

Thank you.

SemoTech, 2024-03-21:
Well I tried both the 32Bit and 64Bit versions of ([Raspberry Pi OS Lite Debian 12 - bookworm](https://www.raspberrypi.com/software/operating-systems/)), the only versions that are supported on the Pi5 and unfortunately neither works! 

As you install with "sudo ./install.sh" and after a lot of files get downloaded and installed, the installer exists with an error!
```

Processing triggers for libvlc-bin:armhf (1:3.0.20-0+rpt5+deb12u1) ...
Processing triggers for libgdk-pixbuf-2.0-0:armhf (2.42.10+dfsg-1+b1) ...
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support
pi@pi5:~/rpisurv $ vlc --version
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
VLC version 3.0.20 Vetinari (3.0.20-0-g6f0d0ab126b)
Compiled by Raspbian buildd for bookworm on Raspbian armhf buildd (Feb 20 2024 15:31:25)
Compiler: gcc version 12.2.0 (Raspbian 12.2.0-14+rpi1)
This program comes with NO WARRANTY, to the extent permitted by law.
You may redistribute it under the terms of the GNU General Public Licence;
see the file named COPYING for details.
Written by the VideoLAN team; see the AUTHORS file.
pi@pi5:~/rpisurv $ 
```
The included version of VLC with both distributions does indeed support "mmal" so RpiSurv needs to be updated to work with the newer versions of VLC...

Anyone else have and other suggestions and advises if 64 Bit is better than 32Bit for RpiSurv?

SvenVD, 2024-03-22:
As I own no pi5 nor bookworm, could you paste output of vlc --help it should contain mmal options

SemoTech, 2024-03-22:
Thanks @SvenVD here you go:

```
pi@pi5:~ $ vlc --list | grep mmal
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
  mmal_xsplitter         MMAL x11 splitter for Raspberry Pi
pi@pi5:~
```
While still on bookworm 32Bit, I even used the "fix" from [HERE](https://forums.raspberrypi.com/viewtopic.php?t=343972#p2061074) and: 

1. Created a new file /etc/apt/preferences.d/vlc containing:
```
Package: vlc*
Pin: origin archive.raspberrypi.org
Pin-Priority: 600

Package: libvlc*
Pin: origin archive.raspberrypi.org
Pin-Priority: 600
```
2. Forced load of the archive versions:
```
sudo apt -y remove vlc* libvlc*
sudo dpkg --configure -a                          # Make sure that all the packages are properly configured
sudo apt-get clean                                # Clean up the old archived packages
sudo apt-get update --allow-releaseinfo-change    # Update the package list
sudo apt-get -y dist-upgrade 
sudo apt-get -y install vlc                       # May have been removed earlier
sudo apt-get -y install python3-vlc               # May also have been removed
sudo apt -y remove vlc-data                       # Old dependencies seem to hang around
sudo apt-get -y install vlc-plugin-base           # And re-install
```
Than again issued:
```
pi@pi5:~ $ cd rpisurv/
pi@pi5:~/rpisurv $ sudo ./install.sh 
```
and got:
```
Your version of vlc does not have the needed mmal options. Rpisurv needs those
Minimum tested vlc version for Rpisurv is (VLC media player 3.0.11 Vetinari (revision 3.0.11-0-gdc0c5ced72),
Maximum tested vlc version for Rpisurv is (VLC media player 3.0.17.4 Vetinari (3.0.13-8-g41878ff4f2)
Aborting installation, upgrade to latest vlc player with mmal support
pi@pi5:~/rpisurv $ 
```
Any other suggestions?

akoryuerek, 2024-03-22:
Uninstall current vlc and the try this.

sudo apt install vlc=3.0.17.4

SemoTech, 2024-03-22:
> sudo apt install vlc=3.0.17.4

Thanks @akoryuerek but no dice on bookworm 32bit, see below:
```
pi@pi5:~ $ sudo apt install vlc=3.0.17.4
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package vlc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version '3.0.17.4' for 'vlc' was not found
pi@pi5:~ $ 
```

SvenVD, 2024-03-24:
Seems we are also bumping into https://github.com/RPi-Distro/vlc/issues/80#issuecomment-1817884925. As I understand drm replaces mmal, but lacks the windows coordinates functionality for multiple instances

SemoTech, 2024-03-24:
> Seems we are also running into [RPi-Distro/vlc#80 (comment)](https://github.com/RPi-Distro/vlc/issues/80#issuecomment-1817884925). Or can you still have the mmal on 32 bit on bookworm?

That's crazy!  Any alternatives to VLC we could use on bookworm v12 and the Pi5?

jdarwood007, 2024-03-28:
> Seems we are also running into [RPi-Distro/vlc#80 (comment)](https://github.com/RPi-Distro/vlc/issues/80#issuecomment-1817884925). Or can you still have the mmal on 32 bit on bookworm?

I don't have a 5, but a 4 and it can install bookworm.  Same issue occurs on 32 bit.  Bullseye I couldn't get to work either.

dlasher, 2024-05-21:
Would love to see RPISURV using something like SDL2 (https://wiki.libsdl.org/SDL2/FrontPage) or SDL3 (https://wiki.libsdl.org/SDL3/FrontPage)  instead as video output --- well supported, and used by quite a few projects.

UntouchedWagons, 2024-07-08:
Any updates on this? I've got two pi 3Bs still running Debian 10 because I've been unable to install rpisurv on any debian version newer.

SvenVD, 2024-07-08:
I am in the middle of a proof of concept trying something out for the next version of rpisurv. However to finish it will take some time as my time is very limited.

dlasher, 2024-07-08:
Appreciate the update - hoping for something to give me a reason to update the old RPISURV2 installs.

SemoTech, 2024-07-08:
Nice!  We will wait patiently @SvenVD...
But can you give us a hint?   
Does it have anything to do with [SDL2 ](https://wiki.libsdl.org/SDL2/FrontPage)or [SDL3](https://wiki.libsdl.org/SDL3/FrontPage) for better compatibility with Pi5 running v12 Bookworm?
Thanks.

SvenVD, 2024-07-08:
I am now focusing on x86 hardware like https://www.asus.com/displays-desktops/mini-pcs/pn-series/asus-expertcenter-pn42/
Reasons:

- More powerful, more simultaneous streams possible
- More stability long term vs raspberry pi changing and breaking stuff every major release. Like the case is now for VLC ( which is a blocker)
- More generic
- Still fanless, vesa mountable and low power draw
- Other x86 hardware will work too in most cases, more options
Downside is that it is generally more expensive hardware.

SemoTech, 2024-07-09:
Interesting to hear for sure. 
Sadly much larger footprint and more expensive hardware. 

I would have thought the improved GPU and CPU in the Pi5 would have helped on the performance side as would moving away from VLC.

dlasher, 2024-07-09:
> * More powerful, more simultaneous streams possible

Honestly we don't need more than 9 streams per screen (for 99% of setups) and a PI will handle that.

> * More stability long term vs raspberry pi changing and breaking stuff every major release. Like the case is now for VLC rpi version ( which is a blocker)

But you can solve that be shifting away from VLC (As has been suggested above with SDL). 

On the bright side, the focus on x86 and intel GPU accel means the older NUC will be usable, but those are still $120-150'ish, which is a FAR cry from a POE powered PI. :(

SvenVD, 2024-07-09:
There is probably a good chance that it will work on x86 as well as pi 5, but we will see.

SvenVD, 2024-11-03:
Update I now released a complete new version (4) that works on following hardware https://github.com/SvenVD/rpisurv/tree/v4_latest?tab=readme-ov-file#verified-hardware-list see https://github.com/SvenVD/rpisurv/tree/v4_latest

UntouchedWagons, 2024-11-03:
I'll see if I can test V4 on a Pi 3B running Debian 12 tomorrow.

SvenVD, 2024-11-03:
> I'll see if I can test V4 on a Pi 3B running Debian 12 tomorrow.

The new version relies on Xorg and mpv player. The installer also expects ubuntu packages.

Versions of the Raspberry Pi other then 5 need special hardware decoding in the player, so for those better keep using the older Rpisurv versions, unless you can find and enable older rpi hardware decoding with mpv options  for each stream manually in https://github.com/SvenVD/rpisurv/blob/d1da3c2275dd36e2118cc0b2899faf60d1866f99/surveillance/etc/monitor1.yml#L95

For the Raspberry Pi 5 your best chance would be to use Ubuntu and make X work like is described here https://askubuntu.com/questions/1501903/x11-session-fails-to-start-on-arm64-ubuntu-23-10-on-raspberry-pi-5. Also the freeform_advanced_mpv_options can be used to enable specialized Raspberry Pi 5 options if needed. Happy to hear the results, thanks for testing.

UntouchedWagons, 2024-11-07:
Okay I installed Ubuntu 24.04 on my Pi 3B and got rpisurv V4 installed but I just get a black screen. If I tap the touchscreen I get a cursor but that's it.

SvenVD, 2024-11-07:
> Versions of the Raspberry Pi other than 5 need special hardware decoding in the player, so for those better keep using the older Rpisurv versions, unless you can find and enable older rpi hardware decoding with mpv options for each stream manually in

Versions of the Raspberry Pi other than 5 and even for [5](https://github.com/SvenVD/rpisurv/issues/194) need special hardware decoding in the player, so for those better keep using the older Rpisurv versions, unless you can find and enable older rpi hardware decoding with mpv options for each stream manually in

UntouchedWagons, 2024-11-07:
Yeah that problem with that is I'm stuck on Debian 10, if I try to update the packages everything just breaks and I have to restore a img backup.

SvenVD, 2024-11-10:
Announcement: I forked Rpisurv into https://opensurv.net/ to better reflect the case that the project is not focused only on Raspberry Pi hardware anymore.

dlasher, 2025-02-24:
@SvenVD 

What's the recommended path for PI3/PI4 hardware to run opensurv? 
* vlc doesn't appear to have mmal support
* pi3/pi4 have less CPU to play with

SvenVD, 2025-03-08:
> [@SvenVD](https://github.com/SvenVD)
> 
> What's the recommended path for PI3/PI4 hardware to run opensurv?
> 
> * vlc doesn't appear to have mmal support
> * pi3/pi4 have less CPU to play with

In short, Raspberry pi hardware is very specific. Rpisurv was specifically designed to make use of the graphical capabilities of a Raspberry Pi out of the box, while the downside was that it could ONLY run on a Raspberry Pi. 
OpenSurv works on all kinds of different [hardware](https://github.com/OpenSurv/OpenSurv/discussions?discussions_q=is%3Aopen+label%3ACompatible) and it can work on a Raspberry Pi, although with (sometimes significantly) lesser performance as Rpisurv could do. OpenSurv however can make maximum use of for example integrated Intel GPU/CPU and others.

Check following discussions:
- https://github.com/OpenSurv/OpenSurv/discussions/22
- https://github.com/OpenSurv/OpenSurv/discussions/4
- https://github.com/OpenSurv/OpenSurv/discussions/7