SvenVD/rpisurv #190: Question:  Any chance for UniFi Protect Integration?

SemoTech, 2024-04-04:
Hi @SvenVD, hope this finds you well.

I was wondering if aside from manually specifying RTSP streams, for those of us that use the Ubiquiti UniFi Protect cameras, there could maybe be a way to have Rpisurv actually connect to the Protect a UniFi  NVR (UDR, UDM-Pro, UDM-SE, UNVR/UNVR-Pro) and allow selection of a "Live View" cameras configurations dashboard, in the same way that the UniFi Protect [ViewPort](https://store.ui.com/us/en/collections/unifi-camera-security-special-viewport/products/ufp-viewport) device does? 

We could create a Protect User (Read-Only) and use those credentials in RpiSurv to automatically pull the RTSP streams for the cameras we want based on the pre-created dashboard ("Live View") in Protect.

This will also alleviate the issue where the UniFi Protect Cameras RTSP stream URL's change if a camera is reset. 

Thank you.

SvenVD, 2024-04-11:
I am happy to accept any pull requests for this feature

SemoTech, 2024-04-11:
> I am happy to accept any pull requests for this feature

Fingers crossed that someone who can code (that's not me :-) ) will do a pull. But I am happy to be a beta tester....

aniceberg, 2024-04-26:
There is a Home Assistant component you may be able to more easily integrate: https://github.com/home-assistant/core/tree/dev/homeassistant/components/unifiprotect

SemoTech, 2024-04-26:
> There is a Home Assistant component you may be able to more easily integrate: https://github.com/home-assistant/core/tree/dev/homeassistant/components/unifiprotect

Thanks @aniceberg I already have that setup in my Home Assistant, but the idea here was for RpiSurv to eliminate the need to manually configure RTSP URL's (which sometimes change in UniFi Protect) via the command line, and read them straight from the UniFi protect controller layout profile, via a login to the UniFi system.

SemoTech, 2024-04-28:
@SvenVD alternatively, what about having an option in RpiSurv to alternatively display an existing NDI Stream (i.e a stream of already tiled security cameras in a single image) being broadcasted on the LAN?  One could specify the NDI source name (in a file) and then RpiSurv could just display the stream.  There should be no need for a license if using NDI | HX or NDI | HX2 or NDI | HX3 instead of full NDI. I even found an Open Source NDI Library [HERE](https://www.reddit.com/r/VIDEOENGINEERING/comments/hbifrh/open_source_ndi_library/)
Thoughts?

schmoizz, 2024-06-19:
What exactly is the need here?
I have a bunch of Unifi Cameras hanging around and simply used the rtsp Streams the Protect App is serving per Camera. Works for a year now. No Problems whatsoever.

gabiX87, 2024-10-21:
What you want is complicated and doesn't make sense. Protect source code is closed anyway. The speed Protect is going it would break it all the time.  I have read someone made a WEB UI for rpisurv , you might want to look into that.

Or just buy their **Viewport** device and save yourself all the headaches.

SemoTech, 2024-10-22:
Thanks for the reply @gabiX87.  
I already have the Protect Viewport device and am able to see all the cameras on my AppleTV, and that works great.  

I just thought having an NDI viewer option in RpiSurv so it can all run on a Pi would be amazing, and it's and Open Source library, [HERE](https://www.reddit.com/r/VIDEOENGINEERING/comments/hbifrh/open_source_ndi_library/)

aniceberg, 2024-10-22:
> Thanks for the reply @gabiX87. I already have the Protect Viewport device and am able to see all the cameras on my AppleTV, and that works great.

Why do you need the Viewport when you can just use the Protect app on your Apple TV?
> 
> I just thought having an NDI viewer option in RpiSurv so it can all run on a Pi would be amazing, and it's and Open Source library, [HERE](https://www.reddit.com/r/VIDEOENGINEERING/comments/hbifrh/open_source_ndi_library/)

SemoTech, 2024-10-22:
@aniceberg The Protect App on ATV works fine but does not allow ALL my cameras to be seen one one screen.  The NDI stream I have created from the Protect Viewport does!   Also a lot more practical to decode the NDI stream with RpiSurv on an "embedded" Pi than to use an Apple TV.