SvenVD/rpisurv #193: 2K monitor 2688x1512  issue

gabiX87, 2024-10-28:
I have a 2K monitor with native **2560x1440** resolution. Using _Raspberry PI 4 model B 2GB_ , _Buster 32bit LITE_.

The problem is **rpisurv** can't display a solo/single rtsp stream in a 2688x1512 resolution. Which is a RTSP from a Unifi Camera in HIGH resolution. MEDIUM (1280x720) and LOW (640x360) streams do work.

I tried to edit/change general.yml fallbacks to 2560x1440 , that had no effect.

Any ideas ?

gabiX87, 2024-10-28:
[](https://github.com/raspberrypi/firmware/issues/1385)

`sudo nano /boot/config.txt`

and add this...

dtoverlay=vc4-fkms-v3d
gpu_mem=256
hdmi_group=2
hdmi_mode=87
hdmi_cvt=2560 1440 60 3 0 0 1
max_framebuffer_width=2560
max_framebuffer_height=1440
hdmi_pixel_freq_limit=400000000
hdmi_group:1=2
hdmi_mode:1=87
hdmi_cvt:1=2560 1440 60 3 0 0 1
hdmi_pixel_freq_limit:1=400000000

gabiX87, 2024-10-28:
I had to change ENCODING from Standard to Enhanced. Whatever Enhanced means in this case , it works now.

[image]

gabiX87, 2024-10-29:
I am out of ideas how to fix this ...

[image]

SvenVD, 2024-10-31:
I am not sure the raspberry pi4 can even handle such higher resolution streams. You could try with vlc and different vlc options first on the raspberry pi

gabiX87, 2024-10-31:
> I am not sure the raspberry pi4 can even handle such higher resolution streams. You could try with vlc and different vlc options first on the raspberry pi

It works with Enhanced encoding (H265) but not with standard encoding (H264) the RTSP stream resolution is the same 2688x1512.
 
Raspberry Pi 4 has 4K (3840 x 2160) support  for two monitors , it can handle it ....

gabiX87, 2024-11-02:
I started over to try older version fo VLC to play the HIGH Quqliaty RTSP stream (h264) in 2688x1512.

[](https://downloads.raspberrypi.com/raspios_oldstable_armhf/images/)
**2021-12-02-raspios-buster-armhf.img from** 

 - VLC version included is 3.0.12 and HIGH quality stream works 

I think the issue is , the VLC version 3.0.17.

71403, 2024-11-05:
This also worked with a Hanwha 4mp camera streaming at 2560 x 1440, also had to use H.265 stream.

Pi 4, 4gb, Buster 32-bit full version