SvenVD/rpisurv #197: 4 channel window camera off-line

wibyo, 2025-09-27:
I would like to still have 4 channel window frame even if I’m having an offline camera and not going to have automatically reconfiguration of the window goes to 2 channel view instead of 4 because i have 3 cameras and one is unstable right now and make triggered the views of others while trying to reconnect

[image]

SvenVD, 2025-09-27:
You can set fixed coordinates https://gist.github.com/SvenVD/5c82acbb96dfb697e5e2d2420ab73ad9

SvenVD, 2025-09-27:
Depends on  your monitor resolution

wibyo, 2025-09-27:
1080

SvenVD, 2025-09-28:
essentials:
  screens:
  - camera_streams:
    - url: "https://images.rpisurv.net/example1.png"
      imageurl: true
      force_coordinates: [0, 0, 960, 540]
    - url: "https://images.rpisurv.net/example2.png"
      imageurl: true
      force_coordinates: [960, 0, 1920, 540]
    - url: "https://images.rpisurv.net/example3.png"
      imageurl: true
      force_coordinates: [0, 540, 960, 1080]
    - url: "https://images.rpisurv.net/example4.png"
      imageurl: true
      force_coordinates: [960, 540, 1920, 1080]